evcc-io/evcc #20766: Falsche Leistungsberechnung bei Ladesessions in Verbindung mit evcc neustart

userwithoutpassword, 2025-04-21:
### Describe the bug

Musste heute den evcc mal paarmal neustarten dabei habe ich folgende Entdeckung gemacht in den Ladesessions, das Fahrzeug blieb mit der Wallbox im übrigen verbunden (Easee)

[image]

[image]

Die durchschnittlich errechnete Ladeleistung beim zweiten Screenshot stimmt nicht. Ich gehe mal davon aus dass die selbe geladene Energie bei beiden Einträgen zugrunde legt.

### Steps to reproduce

siehe oben

### Configuration details

```yaml
siehe oben
```

### Log details

```text
n/a
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.203.1

andig, 2025-04-22:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen. 

/cc @GrimmiMeloni

closed 2025-04-22 by andig

userwithoutpassword, 2025-04-22:
> [@userwithoutpassword](https://github.com/userwithoutpassword) likely due to a known limitation of the charger. Would need a log with easee on trace level to confirm.

Wie komme ich an ein Log aus der Vergangenheit was schon Stunden oder ein Tag zurück liegt?