evcc-io/evcc #20768: Add support for secrets via environment

drpump, 2025-04-22:
**Is your feature request related to a problem? Please describe.**
I prefer not to save secrets or passwords in files: too many security implications. The configuration tool writes the credentials required to access my car's cloud API into the `evcc.yaml` file. I also prefer to keep configuration under version control, which makes this even more of a security problem if a cloud-based repository like GitHub or bitbucket is used. 

There is currently no documented method for passing these credentials into the application through a more secure mechanism. 

**Describe the solution you'd like**
Best practice is to use either the environment or (better still) some form of secrets management to supply secrets to the application. Since secrets management is not especially standardized, support for environment variables to set per-vehicle passwords seems like a reasonable solution. 

**Describe alternatives you've considered**
I'm currently setting the password temporarily, applying the configuration, then removing the password before committing and pushing. I have also configured a private version repository to minimize the risks. Neither is ideal. 

**Additional context**
I'm deploying EVCC into a home Kubernetes cluster. I've published the Kubernetes manifests in GitHub at https://github.com/drpump/evcc-k8s, in case you're interested. The EVCC config in my repository is a direct copy of your `evcc.dist.yaml` file. If you add support for environment variables, I will update my repository to include a sample manifest and configuration to use a Kubernetes secret to hold my cloud API password.

andig, 2025-04-22:
We won't add new functionality for the yaml-based configuration at this time. Yaml is being deprecated and we'll move to db-only. If there is a minimally-invasive option of still adding this, happy to take a PR, but without promise that yaml will be kept alive indefinitely.

closed 2025-04-22 by andig

vankampenp, 2025-07-05:
It stays a consideration to have a means to share the configuration with other users without secrets revealed.

andig, 2025-07-05:
> It stays a consideration to have a means to share the configuration with other users without secrets revealed.

…and of course `evcc config` does not expose secrets.