evcc-io/evcc #20771: grünstromindex Error 429 oo Many Requests - again (new issue)

silent-pc, 2025-04-22:
### Describe the bug

planner: tariff not available: cannot create tariff type 'grünstromindex': unexpected status: 429 (Too Many Requests) 

after update from v0.209.0 to v0.203.1

grünstromindex fail again.
Log-Requests to grünstromindex are every 30 seconds (more as 1000, so blocked).
with v0.209.0 whas whis fine.
what changed between?

i actualy only use the config file evcc.yaml

any hints?

### Steps to reproduce

only update from v0.209.0 to v0.203.1 
apt update
apt dist-upgrade

### Configuration details

```yaml
on a raspberry-pi 400 
ubuntu server (minimized) 24.02.2 installation

evcc.yaml:

# open evcc at http://evcc.local:7070

# LASTMANAGEMENT:
circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hauptstromkreis # Anzeige in der UI (später)
    maxCurrent: 20         # 16A (optional)
    maxPower: 13080        # 11040 = 11,040kW (optional)
    meter: grid1           # optional

interval: 30s # control cycle interval

vehicles:
  - name: guest_vehicle
    type: template # generic
    template: offline
    title: 'Gast Fahrzeug'
    phases: 3      # Standardmäßig 3-phasiges Laden
    mincurrent: 6  # minimale Lade-Stromstärke in Ampere  ( 6 A * 1 Phase  * 230 V =  1,380 kW)
    maxCurrent: 20 # maximaler Lade-Stromstärke in Ampere (20 A * 3 Phasen * 230 V = 13,800 kW)

  - name: ora_03
    title: 'ORA 03 300 Pro'
    type: template # generic
    template: offline
    capacity: 47.8 # Brutto - Akkukapazität in kWh (45.4 Netto)
    phases: 3      # Anzahl der Phasen
    mincurrent: 6  # minimale Lade-Stromstärke in Ampere  ( 6 A * 1 Phase  * 230 V =  1,380 kW)
    maxCurrent: 16 # maximaler Lade-Stromstärke in Ampere (16 A * 3 Phasen * 230 V = 11,040 kW)
    icon: car

#   Geht bei unbekannten 'offline' Fahrzeugen nicht - oder hier ungültig (bei vehicles).
#    range: 230     # Reichweite in km
#    maxSoc: 100    # 100 %
#    powerLimit: 11 # Maximaler LadeLeistungswert in (kW)

tariffs:
  co2:
    type: grünstromindex # Grünstromindex
    zip: '07768'  # PLZ

[... 159 lines cut ...]
[site  ] DEBUG 2025/04/22 10:39:52 grid power: 1W
[site  ] DEBUG 2025/04/22 10:39:52 battery 1 soc: 76%
[site  ] DEBUG 2025/04/22 10:39:52 grid currents: [2.1 -1.1 -1.3]A
[site  ] DEBUG 2025/04/22 10:39:52 site power: -913W
[lp-1  ] DEBUG 2025/04/22 10:39:52 charge total import: 284.970kWh
[lp-1  ] DEBUG 2025/04/22 10:39:52 charger status: A
[site  ] DEBUG 2025/04/22 10:39:52 solar forecast: accumulated 1.537kWh, produced 0.000kWh, scale 0.000
...
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.1

andig, 2025-04-22:
> Log-Requests to grünstromindex are every 30 seconds

Can you please share the Log showing this?

closed 2025-04-22 by andig

andig, 2025-04-22:
It just repeats the error but does not make additional requests.

silent-pc, 2025-04-22:
I just wanted to create logfiles with the new v0.203.1 and realized that everything works now.
The exact cause is not clear to me.
I have definitely restarted the server on which evcc is running and I would actually rule out browser caches (although that would also be possible). The evcc system hangs behind a normal vdsl-router which has no other large firewall or similar.
Are the requests to grünstromindex marked somehow during the request? 
I had started an old version for a short time before for test reasons. Did this possibly block my access?
I suspect that was the reason.
That the blockade lasted a little longer.

It seems to be working now and I would like to thank you for your work.
The thread can be marked as solved if I don't get back to you in the next 1-2 days.

Translated with DeepL.com (free version)

sicheste, 2025-04-26:
Maybe not fully related, but in the response they say that without token it is deprecated:
```
{"err":true,"message":"invalid zipcode","input":{},"provisioning":{"warning":"Deprecation of usage without token! Get your token at https://console.corrently.io/ .","source":"CloudFlare-IP","license":"CC BY-NC-SA 4.0 ( https://creativecommons.org/licenses/by-nc-sa/4.0/legalcode.en )","tier":"authenticated","info":"Anonmous access"}}
```
(result when fetching from https://api.corrently.io/v2.0/gsi/prediction, sure, you can also give a zip code (e.g. https://api.corrently.io/v2.0/gsi/prediction?zip=10178), then you also get this warning)

When calling the address in the browser I get a reply, but with this deprecation warning. But in evcc it told me that it is always error 503 (service not available). I have added for now the free token, it is working fine for now. But the free token is only valid for two months. I don't want to change the token every 2 months...

Maybe the doc should be updated, that the token is no longer optional.

andig, 2025-04-26:
We are using a token…

sicheste, 2025-04-28:
What do you mean with we? The development team or evcc as software? In the config the token is optional. I have not checked the code, whether the software is fetching a token which is used later.