evcc-io/evcc #20773: E3DC verliert Verbindung

creatronics-de, 2025-04-22:
### Describe the bug

Seit dem Update auf die neue 0.2xx Version verliert evcc alle paar Tage die Verbindung zu meiner E3DC. 
Außer dem Update wurde die Konfiguration auf DB umgestellt. 
Nachdem die Verbindung verloren wurde, muss ich evcc neu starten und alles funktioniert wieder. 
Zuvor (0.1xx) hatte ich diese Probleme nie. 
Ich werde hier die Logs posten, sobald ein erneuter Verbindungsverlust passiert.

### Steps to reproduce

1. update to 0.2xxx
2. change to db config
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  cache: error

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.32 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

# unique installation id
plant: *****

interval: 10s # control cycle interval

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: evcc
  user: *****
  password: *****

meters:
- type: template
  template: e3dc-rscp 
  usage: grid  
  host: 192.168.78.170  
  port: 5033
  user: *****
  password: *****
  key: 13471621
  name: grid1
- type: template
  template: e3dc-rscp 
  usage: pv  
  host: 192.168.78.170  
  port: 5033
  user: *****
  password: *****
  key: 13471621
  name: pv2
- type: template
  template: e3dc-rscp 
  usage: battery
  host: 192.168.78.170  
  port: 5033
  user: *****
  password: *****
  key: ******
  name: battery3
  capacity: 18.6

site:
  title: Dahoam
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

chargers:
- name: e3dc
  type: template
  template: phoenix-ev-eth 
  id: 255  
  host: 192.168.79.171  
  port: 502  
  modbus: tcpip  

vehicles:
#  - name: ora
#    type: template 
#    template: offline 
#    title: Ora Funky Cat #Will be displayed in the user interface (optional) 
#    capacity: 45 # Battery capacity in kWh (optional) 

  - name: ora
    type: custom 
    title: Ora Funky Cat #Will be displayed in the user interface (optional) 
    capacity: 45 # Battery capacity in kWh (optional) 
    soc:
      source: script
      cmd: /bin/cat /tmp/ora_soc

  - name: bmw
    type: template
    template: bmw
    title: Creatronics
    user: *****
    password: *****
    vin: *****
    capacity: 66.5

loadpoints:
  - title: Garage
    charger: e3dc
    #vehicle: ora
    mode: pv # charge mode (off, now, minpv, pv)

modbusproxy:
  - port: 502
    uri: 192.168.79.171:502
    readonly: true

Version: `0.203.1`
```

### Log details

```text
Apr 10 02:25:39 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:39 battery 1 soc: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:39 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:39 pv 1 power: dial tcp 192.168.78.170:5033: i/o timeout
Apr 10 02:25:39 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:39 grid power: dial tcp 192.168.78.170:5033: i/o timeout
Apr 10 02:25:40 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:40 battery 1 power: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:40 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:40 battery 1 soc: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:40 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:40 pv 1 power: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:40 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:40 grid power: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:44 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:44 pv 1 power: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:45 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:45 battery 1 power: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:45 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:45 battery 1 soc: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:45 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:45 grid power: dial tcp 192.168.78.170:5033: connect: connection refused
Apr 10 02:25:49 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:49 battery 1 power: authentication error: EOF
Apr 10 02:25:49 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:49 battery 1 soc: authentication error: EOF
Apr 10 02:25:50 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:50 pv 1 power: authentication error: EOF
Apr 10 02:25:50 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:50 grid power: authentication error: EOF
Apr 10 02:25:54 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:54 pv 1 power: authentication error: EOF
Apr 10 02:25:55 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:55 battery 1 power: authentication error: EOF
Apr 10 02:25:55 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:55 battery 1 soc: authentication error: EOF
Apr 10 02:25:55 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:55 grid power: authentication error: EOF
Apr 10 02:25:59 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:59 battery 1 power: authentication error: EOF
Apr 10 02:25:59 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:25:59 battery 1 soc: authentication error: EOF
Apr 10 02:26:00 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:00 pv 1 power: authentication error: EOF
Apr 10 02:26:00 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:00 grid power: authentication error: EOF
Apr 10 02:26:04 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:04 battery 1 power: authentication error: EOF
Apr 10 02:26:04 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:04 battery 1 soc: authentication error: EOF
Apr 10 02:26:05 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:05 pv 1 power: authentication error: EOF
Apr 10 02:26:05 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:05 grid power: authentication error: EOF
Apr 10 02:26:09 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:09 battery 1 power: authentication error: EOF
Apr 10 02:26:09 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:09 battery 1 soc: authentication error: EOF
Apr 10 02:26:10 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:10 pv 1 power: authentication error: EOF
Apr 10 02:26:10 raspberrypi evcc[112101]: [site  ] ERROR 2025/04/10 02:26:10 grid power: authentication error: EOF
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

andig, 2025-04-22:
>connect: connection refused

Sagt leider nichts darüber warum die Verbindung verloren geht. Du kannst mal `trace` probieren, vllt. gibts dann mehr Erkenntenisse. Bis dahin mache ich erstmal zu.

closed 2025-04-22 by andig

nylocx, 2025-05-15:
I have the same issue I currently run evcc as an AddOn in HomeAssistant, just restarting evcc resolves the "authentication" issue with e3dc. Is there a way to get trace logging in the AddOn version or do I need to move it out of HA?

creatronics-de, 2025-05-17:
Hi @nylocx, 
andig added some changes in #21249 
Based on the code, chances are good this won't be an issue anymore. Just update to the latest release.

 @andig Thank you!

nylocx, 2025-05-18:
Thanks a lot for the flollow up, with the new version I have not seen the issue again.