evcc-io/evcc #20784: Vorhersagen: Alternative tariff templates (fall back options)

An-Wi, 2025-04-22:
**Is your feature request related to a problem? Please describe.**
Sowohl **_gruenstromindex_** als auch **_api.akkudoktor.net_** scheinen momentan keine "stabile Verfügbarkeiten" aufzuweisen.
Das nervt beim Neustart von evcc (wenn der tariff-Teil dann gar nicht verwendet wird).
Und auch im laufenden Betrieb, wenn plötzlich gar keine Vorhersagedaten mehr angezeigt werden (ist mir bei CO2 noch nie aufgefallen, aber jetzt aktuell bei Solar).

**Describe the solution you'd like**
Wenn man weitere "tariff templates" hinterlegen könnte, die dann als Alternative herangezogen werden, wenn "Primary"-tariff nicht verfügbar ist bzw. einen Fehler wirft, wäre das genial.
Diese müssten dann natürlich als "Secondary" gekennzeichnet werden, damit die bereits implentierte Aufsummieren-Funktion bei "PV Vorhersagen" nicht tangiert wird.

andig, 2025-04-23:
Wie soll das im UI aussehen? Was spricht dagegen, von vornherein einen verlässlichen Service zu nutzen?

naltatis, 2025-04-23:
> Was spricht dagegen, von vornherein einen verlässlichen Service zu nutzen?

Naja, ich verstehe wo der Wunsch herkommt. So viele "stabile" kostenfreie CO2 Quellen haben wir aktuell leider nicht.

Bin aber auch nicht davon überzeugt, dass eine Fallback Logik auf Seiten von evcc hier die richtige Lösung ist. Das erzeugt nur mehr Komplexität. Dann müssen wir uns um Themen wie "Healthcheck" kümmern.

An-Wi, 2025-04-23:
> Wie soll das im UI aussehen? 

Im Bestfall sieht man im UI keinen Unterschied, weil eigentlich egal woher - Hauptsache es sind Daten da, mit denen man arbeiten kann.

<hr>

Mal einen ganz anderen Ansatz gedacht:

Warum muss sich eigentlich der Nutzer über den jeweiligen Service Gedanken machen und entscheiden, welchen Dienst für ihn am besten funktionieren könnte? Ich teste gefühlt seit Wochen diverse Vorhersagemodelle und finde keine dauerhaft stabile Lösung.

Optimal aus Anwendersicht wäre, wenn nur die lokalen Parameter (PLZ, lat, lon, dec, kwp, az, ac, efficiency, horizon) bspw. unterhalb eines neuen Punkt „locals“ (in site) hinterlegt werden müssten und evcc dann intern die jeweiligen Dienste abfragt ggf. auch mehrstufig bei verschiedenen Anbieter und daraus dann die Vorhersagedaten bereitstellt.

Klingt vielleicht im ersten Moment nach viel Umstrukturierungsaufwand, aber wenn ich die evcc roadmap richtig im Kopf habe, dann soll auf Grundlage der Vorhersagen die Ladesteuerung irgendwann mal dahingehend optimiert werden und dazu sind zuverlässige Prognosedaten aus stabilen Quellen unabdingbar…

andig, 2025-04-24:
> Warum

...weil die Dienste nicht gleichwertig und einfach austauschbar sind. Ich verstehe den Request, aktuell aber wontfix.

closed 2025-04-24 by andig

An-Wi, 2025-04-30:
Bei "nicht gleichwertig" stimme ich dir teilweise zu. Allerdings nervt(e) mich die Unzuverlässigkeit von Grünstromindex in den letzten Woche so extrem, dass ich auf "Green-Grid-Compass" umgestiegen bin und damit keine PLZ-genauen CO2-Werte haben.
**Denn**: Lieber "unscharfe" Werte als _gar keine_! (so war zumindest mein Gedankengang, der letztendlich auch zu diesem feature request geführt hat)

andig, 2025-04-30:
Leider hab ich vom GSI auf mehrere Anfragen keine Antwort erhalten.