evcc-io/evcc #20785: Jeder PV Anlage einen eigenen Namen geben können wie bei Autos auch mit "title:"

typxxi, 2025-04-22:
AKTUELL: "PV-Anlage #1" bis "PV-Anlage #n" 
[image]

Wäre schön den Anlagen einen sprechenderen Namen zu geben wie OST oder WEST oder SÜD oder BALKON. 
Da ich das für meinen Nachbarn einrichten und betreuen darf, ist mir das nach einer Reihe Anläufen natürlich klar, was sich zwischen #1 und #2 verbirgt, weil ich ja die Sensoren in HA definiert habe, um zu testen, ob und wie das geht, was mir Volker empfohlen hatte. Das ging dann auch , nur verbergen sich hinter "PV #2" in Wirklichkeit bei ihm noch 3 weitere Anlagen, wo er mit den einfachen Nummern nix anfangen kann, weil die bei ihm nicht von 1 bis 4 nummeriert sind, sondern OST, WEST und dann SÜD 1 und SÜD 2.

Daher wäre wie beim Fahrzeug wünschenswert eine Art 
title: "PV SÜD 1"

Ich habe das bereits unter **0.203.2 ausprobiert mit "title:" wie beim Auto und bekam prompt die Warnung**, dass evcc wegen "unbekannten key title:" nicht starten konnte.

ALTERNATIVEN: keine
PV #1  ist nun mal kein Name.  Die PV Anlagen lassen sich heute nach teils weit über 2 Jahren nicht einfach mal umnennen auf #1 bis #4 von der Cloud über Homeassistant bis sonst wohin

Ist keine TOP Priorität und kein Beinbruch, hat jetzt aber schon zu mehreren "Kommunikationsmissverständnissen" geführt, weil es keine Durchgängigkeit der "Benennung" gibt. 
Beim Auto haben wir ja auch diese Titel Eigenschaft.

DANKE

StefanSchoof, 2025-04-23:
>  use meter title for ui-configured devices (yaml entries get #.. fallback)

https://github.com/evcc-io/evcc/pull/20389

naltatis, 2025-04-23:
Ja, wie @StefanSchoof schreibt. Du kannst sie benennen. Das geht aber nur, wenn du die Konfiguration dieser Geräte über die Config UI 🧪 vornimmst. Es gibt technische Gründe wieso wir den Metern in der Yaml nicht einfach ein Titel-Feld geben können.

closed 2025-04-23 by naltatis