evcc-io/evcc #20788: SEHBEHINDERTE: AKKULADESTAND & AUTOSTART in der Grafik "kontraststärker / prägnanter" darstellen statt der aktuellen grün auf grün "Tarnung"

typxxi, 2025-04-22:
Der "petrol grüne" BALKEN auf grüner BATTERIEFLÄCHE ist zumindest ein großes Problem für Sehbehinderte.
Zugleich wäre es an der Stelle auch wünschenswert, so eine ART DREIECK als ANZEIGER auf beiden Seiten zu haben, bis zu welchem SOC des Hausakku das Autoladen automatisch starten wird. 

Die aktuelle "schwarze" Linienstückchen sind etwas unauffällig und wirken bedeutungslos. Keine Ahnung, ob solche Dreiecke ein Problem wären in der Darstellung im Vergleich zu diesen Linienenden, die etwas zu kurz, klein oder wenig prägnant "verkannt" werden im Vergleich zu den Farben. 

[image]

eine kontrastreiche LADESTANDS ANZEIGE wäre gut, also weiß zum Beispiel wie hier zu sehen 
und "anfängerhafte Paint Pixel Dreiecke" 

[image]

Diese Dreiecke gerne auch in weiß und damit in Kontrast zum Hintergrund, wobei dann wohl die SOC Anzeige in schwarz besser zu erkennen ist.

[image]

KEINE ALTERNATIVEN - JE NACH SEHBEHINDERUNG mag es aber ggf. noch welche geben.

Ich habe beide Anzeigen dem Betroffenen in einem 1:1 Screenshot gezeigt und Alternative 2 gefiel ihm besser.

DANKE SCHÖN

naltatis, 2025-04-23:
Hi @typxxi, danke für den Input. Ich habe die Elemente in https://github.com/evcc-io/evcc/pull/20793 überarbeitet. Schau dir gerne die Screenshots an. Ich bin bei den runden Indikatoren geblieben um hier nicht aus dem generellen Style unserer Anwendung auszubrechen. Habe sie aber Höher und Breiter gemacht.

StefanSchoof, 2025-04-23:
Playwright hat auch die Möglichkeit von Accessibility testing. Findet nicht alles, aber ein Kontrast Check ist dabei 

https://playwright.dev/docs/accessibility-testing

closed 2025-04-23 by andig