evcc-io/evcc #20789: Nach Update auf 0.203.2  wird Shelly PM Mini nicht mehr erkannt

MucheteWO, 2025-04-22:
### Describe the bug

Hallo. 

Leider wird nach dem Update mein Shelly PM Mini nicht mehr erkannt. Dieser misst und zeigt mir die Werte vom Balkonkraftwerk an.

Gruß und Danke!

### Steps to reproduce

1. Update auf 0.203.2 Fehler vorhanden
2.  Nach Downgrade wieder OK

### Configuration details

```yaml
- type: template
  template: shelly-1pm
  usage: pv
  name: pv3
  host: 192.168.178.184
  channel: 0
```

### Log details

```text
[site ] ERROR 2025/04/23 00:15:46 2025-04-23 00:15:42.041052201 +0200 CEST m=+160.892244229 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.056226398 +0200 CEST m=+160.907418315 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.095496171 +0200 CEST m=+160.946688106 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.124199185 +0200 CEST m=+160.975391101 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.207041225 +0200 CEST m=+161.058233142 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.306293123 +0200 CEST m=+161.157485040 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.434531811 +0200 CEST m=+161.285723746 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:42.560788732 +0200 CEST m=+161.411980667 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:43.010349136 +0200 CEST m=+161.861541053 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:43.633924431 +0200 CEST m=+162.485116366 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:44.48645501 +0200 CEST m=+163.337646946 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:45.166266509 +0200 CEST m=+164.017458444 !! 0ms unknown shelly model: SNPM 2025-04-23 00:15:46.251656171 +0200 CEST m=+165.102848106 !! 0ms unknown shelly model: SNPM
[site ] ERROR 2025/04/23 00:15:46 pv 3 power: unknown shelly model: SNPM
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.2

hwmaier, 2025-04-22:
Das gleiche gilt auch fuer das *Shelly PRO EM* das ich habe, das geht leider auch nicht mehr:
```
[lp-1  ] ERROR 2025/04/23 09:17:11 charger enabled: unknown shelly model: SPEM
```
Ich nutze den Digitalausgang um mein Heisswasser als Last ueber evcc zu steuern:
```
chargers:
  - name: water_heater
    type: template
    template: shelly
    host: 192.168.0.123
    channel: 0
    standbypower: -3600
    integrateddevice: true # No vehicle assignment
    heating: true # No charging plan for example
    icon: waterheater
```

hwmaier, 2025-04-22:
Nachtrag: Ich habe test-halber `SPEM` zu der `switchModels` Liste der Switch.GetStatus endpoints in `gen2go` hinzugefuegt:
```
	switchModels := []string{"SNSW", "SNPL", "SPSW", "S3SW", "S4SW", "SPEM", "S3EM"}
```
Dann funktioniert mein Shelly wieder.
Ich denke die energy meter mit Relais Ausgang sollten in der `switchModels` Liste enthalten sein.

andig, 2025-04-23:
/cc @thierolm

andig, 2025-04-23:
@hwmaier magst Du direkt einen PR machen?

hwmaier, 2025-04-23:
@andig Ja klar, mach ich. Ich have eh schon eine lokale Umgebung wo ich evcc komilieren und dann auch mit meiner Anlage testen kann. Damit ist das schnell gemacht.

andig, 2025-04-23:
@thierolm vor https://github.com/evcc-io/evcc/pull/20570 gab es keinen Modellfilter. Brauchen wir den denn unbedingt? Das Issue hier zeigt, dass das eher Fehler- und pflegeanfällig sein wird, diesen immer nachzuziehen?

thierolm, 2025-04-23:
@hwmaier bitte ergänze auch deine konkrete Modelnummer in den Kommentaren der Switch check Funktion.

andig, 2025-04-23:
Wäre es mal möglich, eine API Response für the Shelly Gen2 Switch, EM1 und EM Geräte zum Vergleich zu sehen? Status und Data?

thierolm, 2025-04-23:
@andig yes, it can help:
Ich versuch mich mal die Tage an einem entsprechenden improvement PR ...
```json
{"methods":["Switch.ResetCounters","Switch.SetConfig","Switch.GetConfig","Switch.GetStatus","Switch.Toggle","Switch.Set","Input.CheckExpression","Input.SetConfig","Input.GetConfig","Input.GetStatus","Schedule.List","Schedule.DeleteAll","Schedule.Delete","Schedule.Update","Schedule.Create","Webhook.ListSupported","Webhook.List","Webhook.DeleteAll","Webhook.Delete","Webhook.Update","Webhook.Create","Script.Stop","Script.Start","Script.Eval","Script.GetCode","Script.PutCode","Script.SetConfig","Script.GetConfig","Script.GetStatus","Script.List","Script.Delete","Script.Create","WS.SetConfig","WS.GetConfig","WS.GetStatus","Mqtt.SetConfig","Mqtt.GetConfig","Mqtt.GetStatus","Cloud.SetConfig","Cloud.GetConfig","Cloud.GetStatus","BLE.SetConfig","BLE.GetConfig","BLE.GetStatus","BLE.CloudRelay.ListInfos","BLE.CloudRelay.List","Wifi.Scan","Wifi.ListAPClients","Wifi.SetConfig","Wifi.GetConfig","Wifi.GetStatus","Sys.SetConfig","Sys.GetConfig","Sys.GetStatus","KVS.Delete","KVS.Set","KVS.GetMany","KVS.Get","KVS.List","Shelly.Identify","HTTP.Request","HTTP.POST","HTTP.GET","Shelly.ListMethods","Shelly.PutTLSClientKey","Shelly.PutTLSClientCert","Shelly.PutUserCA","Shelly.Reboot","Shelly.SetAuth","Shelly.Update","Shelly.CheckForUpdate","Shelly.DetectLocation","Shelly.ListTimezones","Shelly.GetComponents","Shelly.GetStatus","Shelly.FactoryReset","Shelly.ResetWiFiConfig","Shelly.GetConfig","Shelly.GetDeviceInfo","OTA.Abort","OTA.Write","OTA.Data","OTA.Start","OTA.Revert","OTA.Commit","OTA.Update","Sys.SetDebug"]}
```

andig, 2025-04-23:
Nice- dann müssen wir doch nur schauen, obs die Switch/ EM1/ EM Funktionen gibt.

thierolm, 2025-04-23:
jepp, ich hab allerdings keinen EM, kann daher die non Switch Gegenprobe nicht machen.
Kann auch nicht checken, ob bei den 3 phasigen 3 EMs, die verfügbaren EM Endpoints abhängig von der profile Einstellung (monophase/triphase) sind ...

andig, 2025-04-23:
Ich habs mal auf dumm versucht- braucht aber einen Test.

> Kann auch nicht checken, ob bei den 3 phasigen 3 EMs, die verfügbaren EM Endpoints abhängig von der profile Einstellung (monophase/triphase) sind ...

Ich auch nicht. Ggf. muss Profile wieder rein.

hwmaier, 2025-04-23:
Hier die ListMethods von meinem Shelly ProEM:
```
{"methods":["EM1Data.GetNetEnergies","EM1Data.SetConfig","EM1Data.GetConfig","EM1Data.ResetCounters","EM1Data.DeleteAllData","EM1Data.GetData","EM1Data.GetRecords","EM1Data.GetStatus","EM1.GetCTTypes","EM1.RevertToFactoryCalibration","EM1.CalibrateFrom","EM1.SetConfig","EM1.GetConfig","EM1.GetStatus","Switch.ResetCounters","Switch.SetConfig","Switch.GetConfig","Switch.GetStatus","Switch.Toggle","Switch.Set","Button.SetConfig","Button.GetConfig","Button.GetStatus","Button.Trigger","Text.SetConfig","Text.GetConfig","Text.GetStatus","Text.Set","Number.SetConfig","Number.GetConfig","Number.GetStatus","Number.Set","Group.SetConfig","Group.GetConfig","Group.GetStatus","Group.Set","Enum.SetConfig","Enum.GetConfig","Enum.GetStatus","Enum.Set","Boolean.SetConfig","Boolean.GetConfig","Boolean.GetStatus","Boolean.Set","Virtual.Delete","Virtual.Add","Schedule.List","Schedule.DeleteAll","Schedule.Delete","Schedule.Update","Schedule.Create","BTHomeSensor.SetConfig","BTHomeSensor.GetConfig","BTHomeSensor.GetStatus","BTHomeDevice.UpdateFirmware","BTHomeDevice.GetKnownObjects","BTHomeDevice.SetConfig","BTHomeDevice.GetConfig","BTHomeDevice.GetStatus","BTHome.GetObjectInfos","BTHome.DeleteSensor","BTHome.AddSensor","BTHome.DeleteDevice","BTHome.AddDevice","BTHome.StartDeviceDiscovery","BTHome.SetConfig","BTHome.GetConfig","BTHome.GetStatus","Webhook.ListSupported","Webhook.List","Webhook.DeleteAll","Webhook.Delete","Webhook.Update","Webhook.Create","Modbus.SetConfig","Modbus.GetConfig","Modbus.GetStatus","Script.Stop","Script.Start","Script.Eval","Script.GetCode","Script.PutCode","Script.SetConfig","Script.GetConfig","Script.GetStatus","Script.List","Script.Delete","Script.Create","WS.SetConfig","WS.GetConfig","WS.GetStatus","Mqtt.SetConfig","Mqtt.GetConfig","Mqtt.GetStatus","Cloud.SetConfig","Cloud.GetConfig","Cloud.GetStatus","BLE.SetConfig","BLE.GetConfig","BLE.GetStatus","BLE.CloudRelay.ListInfos","BLE.CloudRelay.List","Eth.ListClients","Eth.SetConfig","Eth.GetConfig","Eth.GetStatus","Wifi.Scan","Wifi.ListAPClients","Wifi.SetConfig","Wifi.GetConfig","Wifi.GetStatus","Sys.SetConfig","Sys.GetConfig","Sys.GetStatus","KVS.Delete","KVS.Set","KVS.GetMany","KVS.Get","KVS.List","Shelly.Identify","HTTP.Request","HTTP.POST","HTTP.GET","Shelly.ListMethods","Shelly.PutTLSClientKey","Shelly.PutTLSClientCert","Shelly.PutUserCA","Shelly.Reboot","Shelly.SetAuth","Shelly.Update","Shelly.CheckForUpdate","Shelly.DetectLocation","Shelly.ListTimezones","Shelly.GetComponents","Shelly.GetStatus","Shelly.FactoryReset","Shelly.ResetWiFiConfig","Shelly.GetConfig","Shelly.GetDeviceInfo","OTA.Abort","OTA.Write","OTA.Data","OTA.Start","OTA.Revert","OTA.Commit","OTA.Update","Sys.SetDebug"]}
```

andig, 2025-04-23:
Wunderbar- ich sehe EM1Data aber kein EMData. Das ist also schonmal disjunkt. Probier gerne mal den PR.

closed 2025-04-23 by andig

longtom99, 2025-04-23:
Braucht ihr was für einen Shelly Uni Plus?
unknown shelly model: SNSN

thierolm, 2025-04-23:
@longtom99 Teste ihn mal mit dem nightly und du kannst uns mal ein TraceLog schicken. Müsste funktionieren.

Was uns fehlt ist ein 3 EM im Triphase Profil.

longtom99, 2025-04-23:
Mach ich und werde es testen!
Habe leider keinen 3EM, nur Pro 3EM.

thierolm, 2025-04-23:
Pro 3 EM ist auch gut! :-) Kannst du uns für diesen im Triphase Profil mal das Ergebnis von http://xxx.xxx.xxx.xxx/rpc/Shelly.ListMethods?id=0 zur Verfügung stellen?

longtom99, 2025-04-24:
> [@longtom99](https://github.com/longtom99) Teste ihn mal mit dem nightly und du kannst uns mal ein TraceLog schicken. Müsste funktionieren.
> 
> Was uns fehlt ist ein 3 EM im Triphase Profil.

Shelly Uni Plus funktioniert leider nicht.

[evcc-20250424-062152-trace.log](https://github.com/user-attachments/files/19884339/evcc-20250424-062152-trace.log)

hwmaier, 2025-04-24:
Gerade getestet mit [bacf49f](https://github.com/evcc-io/evcc/commit/bacf49fd12473fe56bbf86c70cfbb3e9eb978f5c)  und leider geht auch mein Shelly ProEM nicht damit. Es gibt nun ein error 500, allerdings in einem anderen Bereich, naemlich wenn das EMPro als Meter angesprochen wird. Der Relaisausgang (switch) scheint zu funktionieren.

```
[lp-1  ] ERROR 2025/04/24 15:04:09 charge power: unexpected status: 500 (Internal Server Error)
[lp-1  ] ERROR 2025/04/24 15:04:13 charge currents: unexpected status: 500 (Internal Server Error)
```

Das ist die meter definition die mit 203.1 funktioniert hat:

```
meters:
  - name: shelly_hotwater
    usage: charge
    type: template
    template: shelly-1pm
    host: 10.0.0.209
    channel: 1
```

hwmaier, 2025-04-24:
Hier ein trace Auszug:
```
[shelly] TRACE 2025/04/24 15:26:50 POST http://10.0.0.209/rpc/Switch.GetStatus?id=1
[shelly] TRACE 2025/04/24 15:26:50 {"id":1,"on":false,"src":"evcc","method":"Switch.GetStatus?id=1"}
--
{"code":-105,"message":"Argument 'id', value 1 not found!"}
```
Das Shelly ProEM hat 2 channels fuer energy meterering aber nur ein channel fuer den Relais Ausgang (switch). Die neue Version von evcc [bacf49f](https://github.com/evcc-io/evcc/commit/bacf49fd12473fe56bbf86c70cfbb3e9eb978f5c) scheint da Probleme zu haben, wenn ein Shelly gleichzeitig auf einem Kanal als Meter und auf dem anderen Kanal als Switch verwendet wird.

```
chargers:
  - name: water_heater
    type: template
    template: shelly
    host: 10.0.0.209
    channel: 0
    standbypower: -3400
    integrateddevice: true # No vehicle assignment
    heating: true # No charging plan for example
    icon: waterheater
```

Jetzt kann man natuerlich sagen, das ist ein Sonderfall der nicht unterstuetzt wird. Fuer bestehende Anlagen unschoen weil die Verdrahtung im Schaltschrank geaendert werden muss. Ich kann damit aber leben.

andig, 2025-04-24:
> Die neue Version von evcc [bacf49f](https://github.com/evcc-io/evcc/commit/bacf49fd12473fe56bbf86c70cfbb3e9eb978f5c) scheint da Probleme zu haben, wenn ein Shelly gleichzeitig auf einem Kanal als Meter und auf dem anderen Kanal als Switch verwendet wird.

Das sollte eigentlich nichts ändern. Wenn das ein Problem ist dann hätte ich es schon mit https://github.com/evcc-io/evcc/pull/20570 bzw. [21b76d5](https://github.com/evcc-io/evcc/commit/21b76d5627d63afef17754ef7d3f7b79b9c55ad2) erwartet und nicht erst jetzt. Kannst Du das bestätigen?

Wenn ich es richtig verstehe ist das Problem hier, dass wir immer "switch" auto-erkennen, aber stattdessen einen anderen Messkanal und zwar ohne Switch verwenden wollen. Kannst Du bitte neues Issue mit Config auf machen? 

@thierolm das scheint mit ein Problem aus #20570 zu sein da dort _immer_ auf Auto-Erkennung gesetzt wird.

thierolm, 2025-04-24:
@longtom99 in deinem Tracelog sind nicht alle Shelly ausgaben. Bitte mal mit `./evcc -l trace meter` erstellen.

VolkerK62, 2025-04-24:
@thierolm  geht es hier nur um Gen2 Geräte?
Mit einigen Gen1 Geräten könnte ich aushelfen (Shelly1, PlugS, Plug, EM, 3EM)

hwmaier, 2025-04-25:
> Gerade getestet mit [bacf49f](https://github.com/evcc-io/evcc/commit/bacf49fd12473fe56bbf86c70cfbb3e9eb978f5c) und leider geht auch mein Shelly ProEM nicht damit. Es gibt nun ein error 500, [...]

Dieses Problem ist nun in c35a62c behoben. Gute Arbeit!

Doppellhelix, 2025-04-25:
Hi,
ich habe auch einen Shelly PM Mini Gen 3 im Einsatz.
Nach Einsatz auf Version 202.3 ist evcc nicht mehr gestartet. Ich musste alles was mit dem Shelly zu tun hat in der ymal auskommentieren, damit evcc wieder "hoch" kam.
Oder hat sich in der yaml was geändert, was ich in der Doku noch nicht gefunden habe?

12retord, 2025-04-25:
Bei mir wird der Shelly PM Mini auch nach update auf 202.3 weiterhin nicht erkannt.

andig, 2025-04-25:
Das wundert mich nicht: https://github.com/evcc-io/evcc/compare/0.203.3...master

12retord, 2025-04-27:
Auch mit der neuen Version 203.4 bekomme ich den PM Mini nicht zum laufen. Muss in der Konfiguration noch etwas geändert werden?

thierolm, 2025-04-27:
@12retord kannst du bitte dazu ein trace log schicken? `evcc -l trace meter`

12retord, 2025-04-27:
Bei mir läuft das in Home Assistant, ich weiß leider nicht, wie ich den Befehl dort ausführen kann, kannst du mit dem unten stehenden Log was anfangen?

[site ] DEBUG 2025/04/27 19:45:42 pv power: 494W
[site ] DEBUG 2025/04/27 19:45:42 site power: 934W
[site ] DEBUG 2025/04/27 19:45:42 solar forecast: accumulated 0.010kWh, produced 0.103kWh, scale 10.256
[site ] DEBUG 2025/04/27 19:46:07 ----
[site ] DEBUG 2025/04/27 19:46:07 pv 3 power: 34W
[site ] DEBUG 2025/04/27 19:46:08 grid power: 0W
[site ] DEBUG 2025/04/27 19:46:08 battery 1 power: 820W
[site ] DEBUG 2025/04/27 19:46:08 pv 1 power: 458W
[site ] DEBUG 2025/04/27 19:46:08 battery 1 soc: 95%
[site ] ERROR 2025/04/27 19:46:11 2025-04-27 19:46:07.902471538 +0200 CEST m=+358.510086828 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:07.917252627 +0200 CEST m=+358.524867935 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:07.95209056 +0200 CEST m=+358.559705961 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:07.976534705 +0200 CEST m=+358.584150124 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:08.063806182 +0200 CEST m=+358.671421565 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:08.115990842 +0200 CEST m=+358.723606391 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:08.271256389 +0200 CEST m=+358.878871698 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:08.532402571 +0200 CEST m=+359.140017954 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:08.924509117 +0200 CEST m=+359.532124407 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:09.648982135 +0200 CEST m=+360.256597629 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:10.217873405 +0200 CEST m=+360.825488787 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:11.195107048 +0200 CEST m=+361.802722468 !! 0ms unknown shelly model: S3PM
[site ] ERROR 2025/04/27 19:46:11 pv 2 power: unknown shelly model: S3PM
[site ] DEBUG 2025/04/27 19:46:11 pv power: 492W
[site ] DEBUG 2025/04/27 19:46:11 site power: 920W
[site ] DEBUG 2025/04/27 19:46:11 solar forecast: accumulated 0.011kWh, produced 0.105kWh, scale 9.701
[site ] DEBUG 2025/04/27 19:46:37 ----
[site ] DEBUG 2025/04/27 19:46:42 pv 3 power: 34W
[site ] DEBUG 2025/04/27 19:46:42 grid power: 0W
[site ] DEBUG 2025/04/27 19:46:43 battery 1 power: 825W
[site ] DEBUG 2025/04/27 19:46:43 battery 1 soc: 95%
[site ] DEBUG 2025/04/27 19:46:43 pv 1 power: 409W
[site ] ERROR 2025/04/27 19:46:46 2025-04-27 19:46:42.866487582 +0200 CEST m=+393.474102854 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:42.894486591 +0200 CEST m=+393.502101881 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:42.922804762 +0200 CEST m=+393.530420182 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:42.972405334 +0200 CEST m=+393.580020624 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:43.053121249 +0200 CEST m=+393.660736632 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:43.143358553 +0200 CEST m=+393.750973917 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:43.271538713 +0200 CEST m=+393.879154114 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:43.537238592 +0200 CEST m=+394.144854011 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:44.038294793 +0200 CEST m=+394.645910194 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:44.701432092 +0200 CEST m=+395.309047585 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:45.412910347 +0200 CEST m=+396.020525859 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:46.004427383 +0200 CEST m=+396.612042932 !! 0ms unknown shelly model: S3PM 2025-04-27 19:46:46.967781004 +0200 CEST m=+397.575396535 !! 0ms unknown shelly model: S3PM
[site ] ERROR 2025/04/27 19:46:46 pv 2 power: unknown shelly model: S3PM
[site ] DEBUG 2025/04/27 19:46:46 pv power: 443W
[site ] DEBUG 2025/04/27 19:46:46 site power: 925W
[site ] DEBUG 2025/04/27 19:46:46 solar forecast: accumulated 0.012kWh, produced 0.105kWh, scale 8.936

longtom99, 2025-04-27:
In der yaml kannst du das auch einstellen:
log: debug
levels:
  meter: trace

12retord, 2025-04-27:
Hab ich gemacht, auf den ersten Blick spuckt er mir aber die gleichen Werte aus. Ich weiß aber auch nicht, welche Daten genau gebraucht werden. Da bräuchte ich ein paar Infos mehr. Keine Ahnung, ob das folgende einen Mehrwert hat:

[site ] ERROR 2025/04/27 21:38:14 2025-04-27 21:37:14.13495533 +0200 CEST m=+485.445896717 !! 60001ms outdated
[site ] ERROR 2025/04/27 21:38:14 pv 3 power: outdated
[site ] DEBUG 2025/04/27 21:38:14 pv power: 0W
[site ] DEBUG 2025/04/27 21:38:14 site power: 969W
[shelly] TRACE 2025/04/27 21:38:14 GET http://192.168.178.144/status
[httpd ] TRACE 2025/04/27 21:38:14 GET /api/system/log
[shelly] TRACE 2025/04/27 21:38:14 {"wifi_sta":{"connected":true,"ssid":"staytuned","ip":"192.168.178.144","rssi":-68},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"21:38","unixtime":1745782693,"serial":93,"has_update":false,"mac":"C45BBE6B67A3","cfg_changed_cnt":1,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"http"}],"emeters":[{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":240.20,"is_valid":true,"total":3363133.1,"total_returned":8686.9},{"power":-3.68,"reactive":0.00,"pf":0.00,"voltage":240.20,"is_valid":true,"total":2323217.8,"total_returned":110534.8}],"update":{"status":"idle","has_update":false,"new_version":"20230913-114150/v1.14.0-gcb84623","old_version":"20230913-114150/v1.14.0-gcb84623","beta_version":"20231107-164916/v1.14.1-rc1-g0617c15"},"ram_total":51064,"ram_free":35016,"fs_size":233681,"fs_free":157377,"uptime":4922033}
[shelly] TRACE 2025/04/27 21:38:14 GET http://192.168.178.144/status
[shelly] TRACE 2025/04/27 21:38:14 {"wifi_sta":{"connected":true,"ssid":"staytuned","ip":"192.168.178.144","rssi":-68},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"21:38","unixtime":1745782693,"serial":93,"has_update":false,"mac":"C45BBE6B67A3","cfg_changed_cnt":1,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"http"}],"emeters":[{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":240.20,"is_valid":true,"total":3363133.1,"total_returned":8686.9},{"power":-3.68,"reactive":0.00,"pf":0.00,"voltage":240.20,"is_valid":true,"total":2323217.8,"total_returned":110534.8}],"update":{"status":"idle","has_update":false,"new_version":"20230913-114150/v1.14.0-gcb84623","old_version":"20230913-114150/v1.14.0-gcb84623","beta_version":"20231107-164916/v1.14.1-rc1-g0617c15"},"ram_total":51064,"ram_free":34836,"fs_size":233681,"fs_free":157377,"uptime":4922033}
[shelly] TRACE 2025/04/27 21:38:14 GET http://192.168.178.144/relay/0
[shelly] TRACE 2025/04/27 21:38:14 {"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"http"}
[site ] DEBUG 2025/04/27 21:38:14 ----
[my_grid] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b e1 00 02 23 d9
[my_grid] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[my_grid] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b e3 00 02 82 19
[my_grid] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[site ] DEBUG 2025/04/27 21:38:14 grid power: 0W
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b bd 00 02 e3 cb
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[my_grid] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b fd 00 02 e2 1f
[my_grid] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 98 a0 90 3c
[site ] DEBUG 2025/04/27 21:38:14 battery 1 power: 889W
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b c1 00 02 22 13
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b c5 00 02 63 d2
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: send 01 04 0b c9 00 02 a3 d1
[my_pv ] TRACE 2025/04/27 21:38:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[site ] DEBUG 2025/04/27 21:38:14 pv 1 power: 0W
[site ] DEBUG 2025/04/27 21:38:15 battery 1 soc: 69%
[my_pv ] TRACE 2025/04/27 21:38:15 modbus: send 01 04 0b ed 00 02 e3 da
[my_pv ] TRACE 2025/04/27 21:38:15 modbus: recv 01 04 04 00 01 39 2a 39 cb
[site ] ERROR 2025/04/27 21:38:18 2025-04-27 21:38:14.362717125 +0200 CEST m=+545.673658586 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:14.383473046 +0200 CEST m=+545.694414526 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:14.420293769 +0200 CEST m=+545.731235304 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:14.462971043 +0200 CEST m=+545.773912523 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:14.554513716 +0200 CEST m=+545.865455270 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:14.692272777 +0200 CEST m=+546.003214182 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:14.779058145 +0200 CEST m=+546.089999644 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:15.121272522 +0200 CEST m=+546.432214039 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:15.522543805 +0200 CEST m=+546.833485303 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:16.185371404 +0200 CEST m=+547.496312791 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:16.81398657 +0200 CEST m=+548.124928216 !! 0ms unknown shelly model: S3PM 2025-04-27 21:38:18.332325795 +0200 CEST m=+549.643267274 !! 0ms unknown shelly model: S3PM
[site ] ERROR 2025/04/27 21:38:18 pv 2 power: unknown shelly model: S3PM
[httpd ] TRACE 2025/04/27 21:38:19 GET /api/system/log

thierolm, 2025-04-28:
@12retord, hast du auch einen Shelly der ersten genration im Einsatz. das Log oben zeigt nur Ausgaben/endpoints von einem Shelly gen1 ...

12retord, 2025-04-28:
Ja, ich habe noch einen Shelly EM laufen, der die Energie einer Wärmepumpe misst. Der PM Mini soll eigentlich die Energie eines BKW messen

thierolm, 2025-04-28:
für den PM Mini siehe #20919 von heute morgen