evcc-io/evcc #20790: Loadpoint Prioritäten lassen sich nicht mehr auf 0 setzen

profex1337, 2025-04-23:
### Describe the bug

Ich hatte Prioritäten für meine Loadpoints eingestellt (über die Web Oberfläche). Nun kann ich sie nicht mehr auf default (0) setzen. 
Er speichert die Eingabe nach einem Neustart von EVCC nicht

### Steps to reproduce

1. Prioritäten bei Loadpoints setzen
2. EVCC Neustarten
3. Prioritäten wieder auf 0 setzen
4. EVCC Neustarten

Prios sind dann wieder auf dem vorher eingestellten Wert
...

### Configuration details

```yaml
X
```

### Log details

```text
X
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-23:
@naltatis da muss das Feld in der Struktur ein Pointer werden?

naltatis, 2025-04-23:
Hm, eigentlich nicht. Priority können wir ja eigentlich immer gesetzt haben mit 0 als startwert. Ich schau da mal rein.

andig, 2025-04-23:
Das Problem hier dürfte sein, dass 0 als zero value nix überschreibt. Aber ich rate...

profex1337, 2025-04-23:
> Das Problem hier dürfte sein, dass 0 als zero value nix überschreibt. Aber ich rate...

Das vermute ich auch!

naltatis, 2025-04-23:
Ja, @diddip21 hat in seinem PR etwas ähnliches beobachtet: https://github.com/evcc-io/evcc/pull/20499#pullrequestreview-2755930982

closed 2025-04-23 by andig