evcc-io/evcc #20791: 0.203.2 Easee stopped working

dm82m, 2025-04-23:
### Describe the bug

With update to 0.203.2 evcc is not anymore starting up. Rollback to 0.203.1 and it is up working again.

### Steps to reproduce

1. Have 0.203.1 installed
2. Update to 0.203.2
3. Get the error described

### Configuration details

```yaml
- name: chargerDirk
    type: template
    template: easee
    user: mail
    password: pass
    charger: EHxxx
    authorize: true
    timeout: 20s
```

### Log details

```text
[main  ] FATAL 2025/04/23 07:39:01 charger [chargerUte] cannot create charger 'chargerUte': cannot create charger type 'template': cannot create charger type 'easee': i/o timeout
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.2

GrimmiMeloni, 2025-04-23:
Just upgraded to 0.203.2. Cannot confirm this issue.
Likely a local network or ISP issue.
Happy to have a second look if you provide a full trace log, @dm82m

dm82m, 2025-04-23:
reproducible -> updated to 0.203.2 and the issue is back, debug log with trace on evcc attached:

```
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:36:00 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 0.008999999612569809
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:24 +0000 UTC) SMART_CHARGING false
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) CABLE_LOCKED false
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:25 +0000 UTC) REASON_FOR_NO_CURRENT 50
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:13:21 +0000 UTC) SOFTWARE_RELEASE 330
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) CABLE_RATING -1
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:13:21 +0000 UTC) BACK_PLATE_ID 8067F3AAF2CE04
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:24 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 40
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) CHARGER_OP_MODE 1
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) OUTPUT_CURRENT 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) OUTPUT_PHASE 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:24 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 40
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) ERROR_CODE 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:24 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 40
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:25 +0000 UTC) DERATING_ACTIVE false
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:00:00 +0000 UTC) ENERGY_PER_HOUR 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-22 17:27:10 +0000 UTC) ERROR_STRING {"rcd":{"ok":1,"mask":0}}
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:00:00 +0000 UTC) LIFETIME_ENERGY 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:26:17 +0000 UTC) CELL_RSSI -43
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:26 +0000 UTC) WI_FI_RSSI 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-22 18:30:23 +0000 UTC) LOCAL_RADIO_CHANNEL 2
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:13:21 +0000 UTC) CHARGER_RAT 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:28:30 +0000 UTC) TEMP_MAX 32
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:55 +0000 UTC) LOCAL_NODE_TYPE 1
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:55 +0000 UTC) LOCAL_SHORT_ADDRESS 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:55 +0000 UTC) LOCAL_PARENT_ADDR_OR_NUM_OF_NODES 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:28:30 +0000 UTC) IN_VOLT_T2_T3 249.3350067138672
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:28:30 +0000 UTC) IN_VOLT_T2_T5 238.47300720214844
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-23 09:28:30 +0000 UTC) IN_VOLT_T2_T4 245.43699645996094
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:24 +0000 UTC) ObservationID(233) 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-18 06:55:31.532 +0000 UTC) CONNECTED_TO_CLOUD true
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:21 +0000 UTC) ObservationID(234) 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:12:24 +0000 UTC) ObservationID(219) 0
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-03-12 17:13:22 +0000 UTC) ObservationID(241) ac:67:b2:0d:f6:bc
[easee ] TRACE 2025/04/23 11:39:15 ProductUpdate EH651612: (2025-04-18 06:54:15.949 +0000 UTC) CLOUD_DISCONNECT_REASON SERVER_INITIATED_DISCONNECT
[easee ] TRACE 2025/04/23 11:39:15 {"userName":"***","password":"***"}
--
{"accessToken":"eyJhbGciOiJSUzI1NiIsInR5cCIgOiAiSldUIiwia2lkIiA6ICJlcVJKMDhGMTFBQi14QWhnNnNSbXdxSE5PdzllS09JbmJaSVpISmpoOVc4In0.eyJleHAiOjE3NDU0MDQ3NTUsImlhdCI6MTc0NTQwMTE1NSwianRpIjoiOWMzMzRlNWMtMTE1Zi00MjczLTgxMzMtZDBmNTZkYmI2YTkyIiwiaXNzIjoiaHR0cHM6Ly9hdXRoLmVhc2VlLmNvbS9yZWFsbXMvZWFzZWUiLCJhdWQiOlsiYWNjb3VudCIsImVhc2VlIl0sInN1YiI6ImQ0ZmQ4NmQyLWM0MTItNGZmZi05MjIzLWY4YmZjYTU1OTVhMSIsInR5cCI6IkJlYXJlciIsImF6cCI6ImVhc2VlIiwic2lkIjoiMGI2MDUxNzQtNDAzNy00Y2Q4LTgwN2EtZWQ3OWNjYjQ0M2Y0IiwiYWNyIjoic2ZhIiwic2NvcGUiOiJlbWFpbCIsIkFjY291bnRJZCI6IjQ0NjgxIiwicm9sZSI6WyJVc2VyIl0sImVtYWlsX3ZlcmlmaWVkIjp0cnVlLCJTdXBwb3J0IjoiRmFsc2UiLCJVc2VySWQiOiIzMzM2NiIsImVtYWlsIjoiZGlya0BtYXVjaGVyLW9ubGluZS5kZSJ9.SxSPaDXQ3zHOLayJ9ki3hO6P7RLDnx9m_J_EV8Kjkzi3Olf1YC5qeoxM8iIyWCOVkxTd3mbklYL2Mw0esxov0pFm5IBVAFPlKek_HUaCpOQVKSEEyt5WOgWsnFDj4iFWJMd_ffRXvYrRexWMFd_Xk_uRFdhMvApzDPwD-jgvKHQh2JZMYCcRUaIyY7JFgGbUfbRf33zh10H08LL1VIYSfitGtjqkYRfA2FwbWc832Umn5YqxOxDAnAOK_Q_Lyfm7n3KwnT-EYs8bEt2_JrYJ6zGpTLiAqRSJkJBtv-q1EOGicYzV_KQhQZ6C1fBEiiH5aFx1jNAOutMFnvy2e2eeDA","expiresIn":3600,"accessClaims":[],"tokenType":"Bearer","refreshToken":"eyJhbGciOiJIUzUxMiIsInR5cCIgOiAiSldUIiwia2lkIiA6ICI4OGQ4NjljMy0yMmRmLTRjYzYtODM5My1mNGNiMDRiNjBiYzUifQ.eyJleHAiOjE3NDU0ODc1NTUsImlhdCI6MTc0NTQwMTE1NSwianR [... cut ...]
[easee ] TRACE 2025/04/23 11:45:22 ProductUpdate EH769539: (2025-04-17 00:37:57.292 +0000 UTC) CONNECTED_TO_CLOUD true
[easee ] TRACE 2025/04/23 11:45:22 ProductUpdate EH769539: (2025-04-17 00:36:41.924 +0000 UTC) CLOUD_DISCONNECT_REASON SERVER_INITIATED_DISCONNECT
[main  ] FATAL 2025/04/23 11:45:32 charger [chargerUte] cannot create charger 'chargerUte': cannot create charger type 'template': cannot create charger type 'easee': i/o timeout
[main  ] FATAL 2025/04/23 11:45:32 will attempt restart in: 15m0s

```

dm82m, 2025-04-23:
I guess we are ready to re-open @andig, bug is still there and log is also there.

TomF79, 2025-04-23:
@brmien 

hat wohl auch Probleme: https://github.com/evcc-io/evcc/discussions/20781

GrimmiMeloni, 2025-04-23:
I will have a closer look later today. At first sight this looks like a side effect of the initial state handling we introduced.
So far I am failing to see the difference in setups.
Can you please provide full config, including non obfuscated IDs for the chargers, etc. I simply want to make sure I am assessing the right data, as I don't know who Ute is, and which charger belongs to her. ;)

GrimmiMeloni, 2025-04-23:
Thanks for referring to #20781 - common theme seems to be multiple chargers.

@andig from the top of your head, do the separate Easee instances share any state? TTBOMK that is not the case, but maybe I am mistaken?

dm82m, 2025-04-23:
@GrimmiMeloni will send you original data in slack, don’t want to share it publicly

andig, 2025-04-23:
> don’t want to share it publicly

Btw. this is really nice for sending sensitive information via Github: https://github.com/FiloSottile/age

    curl https://github.com/andig.keys | age -R - example.jpg > example.jpg.age

GrimmiMeloni, 2025-04-23:
> Maybe Mid is the actual device? Log above shows updates for 2 different devices! Looking at the code we don't seem to discriminate between them ⭕

That's at least an interesting theory. I so far assumed our subscription is charger specific.

GrimmiMeloni, 2025-04-23:
I spent more time with a local debug build, and cannot identify the issue.

I will push a PR for the upcoming nightly later. It will add more logging, but I cannot promise it will allow me to find it right away.

Ideally someone affected (with 2 Easee's on a single acocunt) would provide their credentials, so that I can directly see the data. Not sure if @dm82m or @brmien would be up for that. 🤷‍♂  (if so, DM me on Slack, or send via email to grund.heuboden0v@icloud.com )

@andig I suggest to revert https://github.com/evcc-io/evcc/pull/20646 and push a new release as an interim solution.

brmien, 2025-04-23:
@GrimmiMeloni, I sent an email to your address.

GrimmiMeloni, 2025-04-23:
@andig in case you have not pulled the trigger, please hold on the revert. I have a local reproduction. Fix should be there soon.

andig, 2025-04-23:
Sitting still.

GrimmiMeloni, 2025-04-23:
_(First of all kudos to @brmien for providing his login so I could reproduce the problem. Thanks a ton for your trust.)_

I should have spotted it earlier, but guess I was not thorough enough. :( 

For setups of both @brmien and @dm82m I can see that one charger is not sending all the state information.
Specifially both chargers send Product Updates for
* CHARGER_OP_MODE
* LIFETIME_ENERGY

However, only one (I assume it is the primary one, but I don't know the exact setups folks have) also sends:
* SESSION_ENERGY
* TOTAL_POWER

What's interesting is, that both chargers that lack the latter product updates, also send a value of 0 for the LIFETIME_ENERGY. This means either the cloud data for these chargers is incorrect, or these are brand new chargers that were never used so far.

Either way, what it means is that our logic in https://github.com/evcc-io/evcc/pull/20646 breaks for these chargers, as there obviously is no way of knowing which state information to expect and which not.

@andig I suggest we revert #20646, release a new version to get folks back into running state, and then come up with a more relaxed solution for fetching the optional state values.

dm82m, 2025-04-23:
To give insights from my side:
- both chargers are main charges in my account, so no primary secondary setup BUT
- the second charger (Ute) is the one of my mother and has never been used so far

brmien, 2025-04-23:
For my site ‘Schuppen’ is Master, and ‘Garage’ has never been used so far.

PR #20822 (2025-04-24): Easee: revert wait for essential state during boot (#20646)
fix #20791
closed 2025-04-24 by andig