evcc-io/evcc #20794: Sungrow sh10.0 & BYD - battery 1 energy: read faile

eikarramba, 2025-04-23:
### Describe the bug

Hallo, ich bekomme seit einiger Zeit vermehrt Fehlermeldung. Seit wann genau kann ich nicht sagen, da das Laden weiterhin funktioniert.

 battery 1 soc: read failed: write tcp 192.168.188.70:58502->192.168.188.67:502: write: broken pipe

battery 1 soc: read failed: EOF

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Keine Ahnung wo ich die Konfiguration finde.
```

### Log details

```text
[lp-1 ] DEBUG 2025/04/23 09:29:43 charge power: 6835W
[lp-1 ] DEBUG 2025/04/23 09:29:43 charge currents: [10.3 10.2 10.3]A
[site ] DEBUG 2025/04/23 09:29:43 pv 1 power: 8081W
[site ] DEBUG 2025/04/23 09:29:43 grid power: -0W
[site ] DEBUG 2025/04/23 09:29:43 battery 1 power: -578W
[site ] ERROR 2025/04/23 09:29:43 battery 1 energy: read failed: EOF
[site ] DEBUG 2025/04/23 09:29:43 grid currents: [0.371 -0.0902 -0.328]A
[site ] DEBUG 2025/04/23 09:29:43 battery 1 soc: 46%
[site ] DEBUG 2025/04/23 09:29:43 site power: -578W
[lp-1 ] DEBUG 2025/04/23 09:29:43 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:29:43 charge total import: 3054.089kWh
[lp-1 ] DEBUG 2025/04/23 09:29:43 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:29:43 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:29:43 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:29:43 pv charge current: 11.8A = 11A + 0.838A (-578W @ 3p)
[site ] DEBUG 2025/04/23 09:30:13 ----
[lp-1 ] DEBUG 2025/04/23 09:30:13 charge power: 6840W
[lp-1 ] DEBUG 2025/04/23 09:30:13 charge currents: [10.3 10.2 10.3]A
[site ] DEBUG 2025/04/23 09:30:13 pv 1 power: 8060W
[site ] DEBUG 2025/04/23 09:30:13 grid power: -0W
[site ] DEBUG 2025/04/23 09:30:13 battery 1 power: -562W
[site ] ERROR 2025/04/23 09:30:13 grid currents: div[1]: read failed: write tcp 192.168.188.70:53634->192.168.188.67:502: write: broken pipe
[site ] DEBUG 2025/04/23 09:30:13 battery 1 soc: 46%
[site ] DEBUG 2025/04/23 09:30:13 site power: -562W
[lp-1 ] DEBUG 2025/04/23 09:30:13 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:30:13 charge total import: 3054.145kWh
[lp-1 ] DEBUG 2025/04/23 09:30:13 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:30:13 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:30:13 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:30:13 pv charge current: 11.8A = 11A + 0.814A (-562W @ 3p)
[site ] DEBUG 2025/04/23 09:30:43 ----
[lp-1 ] DEBUG 2025/04/23 09:30:43 charge power: 6849W
[lp-1 ] DEBUG 2025/04/23 09:30:43 charge currents: [10.4 10.3 10.3]A
[site ] DEBUG 2025/04/23 09:30:43 pv 1 power: 8011W
[site ] DEBUG 2025/04/23 09:30:43 grid power: -0W
[site ] DEBUG 2025/04/23 09:30:43 battery 1 power: -518W
[site ] ERROR 2025/04/23 09:30:43 battery 1 energy: read failed: write tcp 192.168.188.70:49766->192.168.188.67:502: write: broken pipe
[site ] DEBUG 2025/04/23 09:30:43 grid currents: [0.405 -0.013 -0.332]A
[site ] DEBUG 2025/04/23 09:30:43 battery 1 soc: 46%
[site ] DEBUG 2025/04/23 09:30:43 site power: -518W
[lp-1 ] DEBUG 2025/04/23 09:30:43 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:30:43 charge total import: 3054.202kWh
[lp-1 ] DEBUG 2025/04/23 09:30:43 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:30:43 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:30:43 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:30:43 pv charge current: 11.8A = 11A + 0.751A (-518W @ 3p)
[site ] DEBUG 2025/04/23 09:31:13 ----
[lp-1 ] DEBUG 2025/04/23 09:31:13 charge power: 6840W
[lp-1 ] DEBUG 2025/04/23 09:31:13 charge currents: [10.4 10.3 10.3]A
[site ] DEBUG 2025/04/23 09:31:13 pv 1 power: 8036W
[site ] DEBUG 2025/04/23 09:31:13 grid power: -0W
[site ] DEBUG 2025/04/23 09:31:13 battery 1 power: -548W
[site ] ERROR 2025/04/23 09:31:13 grid currents: div[1]: read failed: write tcp 192.168.188.70:40032->192.168.188.67:502: write: broken pipe
[site ] DEBUG 2025/04/23 09:31:13 battery 1 soc: 46%
[site ] DEBUG 2025/04/23 09:31:13 site power: -548W
[lp-1 ] DEBUG 2025/04/23 09:31:13 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:31:13 charge total import: 3054.260kWh
[lp-1 ] DEBUG 2025/04/23 09:31:13 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:31:13 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:31:13 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:31:13 pv charge current: 11.8A = 11A + 0.794A (-548W @ 3p)
[site ] DEBUG 2025/04/23 09:31:43 ----
[lp-1 ] DEBUG 2025/04/23 09:31:43 charge power: 6854W
[lp-1 ] DEBUG 2025/04/23 09:31:43 charge currents: [10.4 10.3 10.4]A
[site ] DEBUG 2025/04/23 09:31:43 pv 1 power: 8105W
[site ] DEBUG 2025/04/23 09:31:43 grid power: -0W
[site ] DEBUG 2025/04/23 09:31:43 battery 1 power: -613W
[site ] ERROR 2025/04/23 09:31:43 battery 1 soc: read failed: write tcp 192.168.188.70:53542->192.168.188.67:502: write: broken pipe
[site ] DEBUG 2025/04/23 09:31:43 grid currents: [0.409 -0.0392 -0.316]A
[site ] DEBUG 2025/04/23 09:31:43 site power: -613W
[lp-1 ] DEBUG 2025/04/23 09:31:43 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:31:43 charge total import: 3054.315kWh
[lp-1 ] DEBUG 2025/04/23 09:31:43 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:31:43 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:31:43 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:31:43 pv charge current: 11.9A = 11A + 0.888A (-613W @ 3p)
[site ] DEBUG 2025/04/23 09:32:13 ----
[lp-1 ] DEBUG 2025/04/23 09:32:13 charge power: 6852W
[lp-1 ] DEBUG 2025/04/23 09:32:13 charge currents: [10.4 10.3 10.3]A
[site ] DEBUG 2025/04/23 09:32:13 pv 1 power: 8024W
[site ] DEBUG 2025/04/23 09:32:13 grid power: 40W
[site ] DEBUG 2025/04/23 09:32:13 battery 1 power: -570W
[site ] ERROR 2025/04/23 09:32:13 grid currents: div[0]: read failed: write tcp 192.168.188.70:58290->192.168.188.67:502: write: broken pipe
[site ] DEBUG 2025/04/23 09:32:13 battery 1 soc: 46%
[site ] DEBUG 2025/04/23 09:32:13 site power: -530W
[lp-1 ] DEBUG 2025/04/23 09:32:13 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:32:13 charge total import: 3054.373kWh
[lp-1 ] DEBUG 2025/04/23 09:32:13 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:32:13 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:32:13 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:32:13 pv charge current: 11.8A = 11A + 0.768A (-530W @ 3p)
[site ] DEBUG 2025/04/23 09:32:43 ----
[lp-1 ] DEBUG 2025/04/23 09:32:43 charge power: 6854W
[lp-1 ] DEBUG 2025/04/23 09:32:43 charge currents: [10.4 10.3 10.3]A
[site ] DEBUG 2025/04/23 09:32:43 pv 1 power: 7974W
[site ] DEBUG 2025/04/23 09:32:43 grid power: 36W
[site ] DEBUG 2025/04/23 09:32:43 battery 1 power: -492W
[site ] ERROR 2025/04/23 09:32:43 battery 1 energy: read failed: write tcp 192.168.188.70:50656->192.168.188.67:502: write: broken pipe
[site ] DEBUG 2025/04/23 09:32:43 grid currents: [0.463 -0.00869 -0.268]A
[site ] DEBUG 2025/04/23 09:32:43 battery 1 soc: 46%
[site ] DEBUG 2025/04/23 09:32:43 site power: -456W
[lp-1 ] DEBUG 2025/04/23 09:32:43 detected active phases: 3p
[lp-1 ] DEBUG 2025/04/23 09:32:43 charge total import: 3054.430kWh
[lp-1 ] DEBUG 2025/04/23 09:32:43 charger status: C
[lp-1 ] DEBUG 2025/04/23 09:32:43 vehicle soc: 94%
[lp-1 ] DEBUG 2025/04/23 09:32:43 vehicle range: 129km
[lp-1 ] DEBUG 2025/04/23 09:32:43 pv charge current: 11.7A = 11A + 0.661A (-456W @ 3p)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

andig, 2025-04-23:
> I have verified that the issue is reproducible with the latest nightly build

Hast Du? In jedem Fall bitte .2 testen.

closed 2025-04-23 by andig

eikarramba, 2025-04-24:
> > I have verified that the issue is reproducible with the latest nightly build
> 
> Hast Du? In jedem Fall bitte .2 testen.

Mit .2 genau das gleiche Problem.

andig, 2025-04-27:
Im aktuellen Nightly sind jetzt wieder 10s timeout eingestellt.