evcc-io/evcc #20796: PV production calculation wrong since v0.203.2 (sungrow & byd)

MrMontesa, 2025-04-23:
### Describe the bug

Seit dem Update auf die v0.203.2 ist die Berechnung der PV Produktion inkorrekt.

[image]

Während das System eigentlich das hier liefert:

[image]

Bei mir läuft eine Sungrow SH10RT via modbus und ne BYD Batterie. Die Homeassisant werte kommen aus dem gleichen modbus setup via modbus_proxy der als HA addon läuft.

Sagt bescheid, wenn ich andere Logs oder konfigs ziehen soll um das Problem besser zu erkennen.
Vielen Dank & Gruß

### Steps to reproduce

1. update to v0.203.2
2.
3.
...

### Configuration details

```yaml
Ich hab vor ein paar Versionen auf db umgestellt.
/app # evcc dump --cfg
[main  ] INFO 2025/04/23 11:15:54 evcc 0.203.2

Fehlermeldung:

    Config File "evcc" Not Found in "[/root /app /etc]"

Version: `0.203.2`
```

### Log details

```text
log:

[site ] DEBUG 2025/04/23 10:58:50 ----
[lp-1 ] DEBUG 2025/04/23 10:58:50 charge power: 0W
[lp-1 ] DEBUG 2025/04/23 10:58:50 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/23 10:58:50 grid power: -1W
[site ] DEBUG 2025/04/23 10:58:50 pv 1 power: 368W
[site ] DEBUG 2025/04/23 10:58:50 battery 1 power: -3287W
[site ] DEBUG 2025/04/23 10:58:50 battery 1 soc: 64%
[site ] DEBUG 2025/04/23 10:58:50 battery has priority at soc 64% (< 90%)
[site ] DEBUG 2025/04/23 10:58:50 site power: 99W
[lp-1 ] DEBUG 2025/04/23 10:58:50 charge total import: 2368.397kWh
[lp-1 ] DEBUG 2025/04/23 10:58:50 charger status: A
[site ] DEBUG 2025/04/23 10:59:20 ----
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.2

closed 2025-04-23 by andig

MrMontesa, 2025-04-23:
Hey @andig ,
danke für die schnelle Antwort. Wie zieht man denn die config, wenn das in der DB ist? Ein einfaches `evcc config` innerhalb des HA docker containers bringt nur:
```
/app # evcc config
[main  ] INFO 2025/04/23 12:50:56 evcc 0.203.2
[main  ] FATAL 2025/04/23 12:50:56 Config File "evcc" Not Found in "[/root /app /etc]"
/app # 
```
Heute morgen stimmte der "Production" Wert nicht. Also auch "Consumption". Produktion war 368W in EVCC, während es 3,47kW sein sollte. Und Consumption war 0 wobei es 405W gewesen sein sollte.

Jetzt ist die Batterie im Keller voll und die "Produktion" stimmt wieder. Allerdings ist jetzt die "Grid export" bei 0, wobei ich aktuell mit 18kW ins Netz lade.
Hier noch mal zwei Screenshots:

[image]

[image]

andig, 2025-04-23:
Es reicht wenn du deine config postest- ohne macht wieder öffnen keinen Sinn.

MrMontesa, 2025-04-23:
Wie gesagt hab ich auf config.db umgestellt. evcc dump cfg gibt folgendes aus
```
/config # evcc dump --cfg
[main  ] INFO 2025/04/23 14:43:12 evcc 0.203.2
[main  ] INFO 2025/04/23 14:43:12 using config file: /config/evcc.yaml
[mqtt  ] ERROR 2025/04/23 14:43:12 tcp://localhost:1883 connection lost: EOF

Konfiguration (/config/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

#log: debug
log: error
levels:
  cache: error
  #easee: trace

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: evcc
  user: *****
  password: *****

Version: `0.203.2`
```

Ansonsten hier noch ein paar Screenshots der config:

[image]
[image]

Gibts ne Möglichkeit die config aus der DB zu parsen, so dass sie evcc.yaml content enhält?

MrMontesa, 2025-04-23:
> Wir brauchen bitte die Config:
> 
> ```
> evcc config
> ```
> 
> > Seit dem Update auf die v0.203.2 ist die Berechnung der PV Produktion inkorrekt.
> 
> Was war die letzte funktionierende Version? Welcher Wert konkret ist falsch?

Bei der Config hab ich jetzt mein bestes gegeben. 
https://github.com/evcc-io/evcc/issues/20796#issuecomment-2824191164
Wenn du da mehr Details brauchst, brauch ich ne Anleitung wie man an die Config dran kommt.

Bei der Version bin ich mir nicht 100% sicher, aber ich hatte vorher die 0.203.1 drauf. Da war mir das bisher nichts aufgefallen. Hab heute morgen das update auf die .2 gemacht und seit dem auch noch nicht das Auto geladen.

Welche Werte nicht stimmen?
https://github.com/evcc-io/evcc/issues/20796#issuecomment-2823904018

andig, 2025-04-24:
> /app # evcc config

Geht nciht

> /config # evcc dump --cfg

Geht. Vllt. musst du mal den richtigen Pfad probieren? Ohne Config wird hier nix weiter gehen.

andig, 2025-04-25:
Mach doch bitte mal das Angefragte:

    evcc config

...und zwar in dem Ordner wo es funktionert!

MrMontesa, 2025-04-25:
```
/app # cd /config/
/config # evcc config
[main  ] INFO 2025/04/25 08:59:04 evcc 0.203.2
[main  ] INFO 2025/04/25 08:59:04 using config file: /config/evcc.yaml
[mqtt  ] ERROR 2025/04/25 08:59:04 tcp://localhost:1883 connection lost: EOF
```
Thats it.

andig, 2025-04-25:
Das ist ja schonmal anders als     

> [main  ] FATAL 2025/04/23 12:50:56 Config File "evcc" Not Found in "[/root /app /etc]"

Ich hab keine Ahnung, warum er sich jetzt ausgerechnet mit MQTT nicht verbinden sollte. Für Dump der DB ist das auch nicht notwendig, aber anscheinend mit enthalten. Da müssten wir nochmal schauen ob sich das ausbauen ließe.

> clientid: evcc

Ach der Grund dürfte klar sein- es kann nur eine Verbindung geben. Nimm einfach die clientid raus, dann ist alles gut...

MrMontesa, 2025-04-25:
Wenn ich die clientid raus nehme, ist in der Tat die MQTT Fehlermeldung weg.
Kommt leider trotzdem nicht mehr

```
/config # evcc config
[main  ] INFO 2025/04/25 09:46:00 evcc 0.203.2
[main  ] INFO 2025/04/25 09:46:00 using config file: /config/evcc.yaml
```

MrMontesa, 2025-04-30:
Hey,
ich stehe etwas auf dem Schlauch wie ich dir bei der config weiter helfen kann. Aktuell verhält sich das System komisch.
Obwohl ich aktuell 4WK Strom ins Netz exportiere, lädt EVCC das Auto nur mit minimal surplus

[image]

[image]

```
[lp-1 ] ERROR 2025/04/30 12:33:09 vehicle soc: unexpected status: 502 (Bad Gateway)
[lp-1 ] DEBUG 2025/04/30 12:33:09 pv charge current: 7A = 7A + 0.00435A (-1W @ 1p)
[site ] DEBUG 2025/04/30 12:33:39 ----
[lp-1 ] DEBUG 2025/04/30 12:33:39 charge power: 1489W
[lp-1 ] DEBUG 2025/04/30 12:33:39 charge currents: [6.8 0.009 0.01]A
[site ] DEBUG 2025/04/30 12:33:39 grid power: -1W
[site ] DEBUG 2025/04/30 12:33:39 pv 1 power: 5871W
[site ] DEBUG 2025/04/30 12:33:39 battery 1 power: 0W
[site ] DEBUG 2025/04/30 12:33:39 battery 1 soc: 100%
[site ] DEBUG 2025/04/30 12:33:39 site power: -1W
[lp-1 ] DEBUG 2025/04/30 12:33:39 detected active phases: 1p
[lp-1 ] DEBUG 2025/04/30 12:33:39 charge total import: 2384.459kWh
[lp-1 ] DEBUG 2025/04/30 12:33:39 charger status: C
[lp-1 ] ERROR 2025/04/30 12:33:39 vehicle soc: unexpected status: 502 (Bad Gateway)
[lp-1 ] DEBUG 2025/04/30 12:33:39 pv charge current: 7A = 7A + 0.00435A (-1W @ 1p)
[site ] DEBUG 2025/04/30 12:34:09 ----
[lp-1 ] DEBUG 2025/04/30 12:34:09 charge power: 1491W
[lp-1 ] DEBUG 2025/04/30 12:34:09 charge currents: [6.8 0.009 0.01]A
[site ] DEBUG 2025/04/30 12:34:09 grid power: -1W
[site ] DEBUG 2025/04/30 12:34:09 pv 1 power: 5904W
[site ] DEBUG 2025/04/30 12:34:09 battery 1 power: 0W
[site ] DEBUG 2025/04/30 12:34:09 battery 1 soc: 100%
[site ] DEBUG 2025/04/30 12:34:09 site power: -1W
[lp-1 ] DEBUG 2025/04/30 12:34:09 detected active phases: 1p
[lp-1 ] DEBUG 2025/04/30 12:34:09 charge total import: 2384.459kWh
[lp-1 ] DEBUG 2025/04/30 12:34:09 charger status: C
[lp-1 ] ERROR 2025/04/30 12:34:09 vehicle soc: unexpected status: 502 (Bad Gateway)
[lp-1 ] DEBUG 2025/04/30 12:34:09 pv charge current: 7A = 7A + 0.00435A (-1W @ 1p)
```
Die 502 Fehlermeldung ist erwartet, da der Fiat connect Vertrag ausgelaufen ist, und ich mich nicht mehr einloggen kann.

Ich werd mal versuchen aussenhalb von HA nen container als test system aufzubauen. Durch mod_proxy müsste das gehen. Dann kann ich auch noch mal unterschiedliche Versionen testen.
Wenn du sonst noch ne Idee hast, lass es mich wissen.

MrMontesa, 2025-04-30:
Just did a container test setup to compare. Started with version v201.1.
Screenshot on the left shows old version from container, while right is most recent version in HA addon container.

[image]

andig, 2025-04-30:
Die Screenshots sind nutzlos. Beide zeigen wunderbar ladendes Auto. Ansonsten- da die Rechnung laut Log- stimmt:

> Welcher Wert konkret ist falsch?

MrMontesa, 2025-04-30:
Wie im Screenshot von heute morgen zu sehen https://github.com/evcc-io/evcc/issues/20796#issuecomment-2841561647
Ist der Strom Export nicht wie in EVCC 0, sondern gut 4kw gewesen.

Warum exportiere ich 4kw, wenn der Charger auf automatic steht und von 1,49kw Ladestrom hoch schalten könnte?

Im Vergleichs screenshot mit beiden Version ist es auch "Grid export". Alte Version ist export 0,9kw (gelber Anteil im Bar chart). In der neuen Version ist der Wert 0. Backend ist der gleiche modbus_proxy. Woher die unterschiedlichen werte?

andig, 2025-04-30:
Dann bitte

    evcc meter —log trace

und den falschen Wert bestätigen. Und dann bleibt die Frage, was dein grid meter ist.

andig, 2025-05-02:
> Im Produktiv System kommt bei evcc meter --log trace nur das hier:

Ich sehe keine Messwerte

MrMontesa, 2025-05-02:
Ich auch nicht!
Ich hab die DB config mal gelöscht und dann `experimentell = false` gesetzt. Nach dem import meiner alten evcc.yaml sehen die Werte jetzt wieder normal aus.
```
/config # evcc meter --log trace
[main  ] INFO 2025/05/02 10:21:40 evcc 0.203.4
[main  ] INFO 2025/05/02 10:21:40 using config file: /config/evcc.yaml
[db    ] INFO 2025/05/02 10:21:40 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/05/02 10:21:40 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/05/02 10:21:40 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2025/05/02 10:21:40 SELECT * FROM `settings` 0 <nil>
[mqtt  ] INFO 2025/05/02 10:21:40 connecting evcc at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/02 10:21:40 tcp://localhost:1883 connected
[db    ] TRACE 2025/05/02 10:21:40 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[mqtt  ] ERROR 2025/05/02 10:21:40 tcp://localhost:1883 connection lost: EOF
[mqtt  ] DEBUG 2025/05/02 10:21:40 tcp://localhost:1883 connected
[db    ] TRACE 2025/05/02 10:21:40 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text,`title` text,`icon` text,`product` text,`value` text) 0 <nil>
[db    ] TRACE 2025/05/02 10:21:40 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
pv
--
[pv    ] TRACE 2025/05/02 10:21:40 modbus: send 00 01 00 00 00 06 01 04 13 98 00 02
[pv    ] TRACE 2025/05/02 10:21:40 modbus: recv 00 01 00 00 00 07 01 04 04 12 ff 00 00
[pv    ] TRACE 2025/05/02 10:21:40 modbus: send 00 02 00 00 00 06 01 04 32 ca 00 02
[pv    ] TRACE 2025/05/02 10:21:40 modbus: recv 00 02 00 00 00 07 01 04 04 c5 47 00 02
Power:  4863W
Energy: 18157.5kWh

grid
----
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 03 00 00 00 06 01 04 32 d1 00 02
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 03 00 00 00 07 01 04 04 00 04 00 00
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 04 00 00 00 06 01 04 32 ec 00 02
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 04 00 00 00 07 01 04 04 d0 86 00 00
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 05 00 00 00 06 01 04 15 e2 00 02
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 05 00 00 00 07 01 04 04 00 25 00 00
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 06 00 00 00 06 01 04 13 9a 00 01
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 06 00 00 00 05 01 04 02 08 c3
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 07 00 00 00 06 01 04 15 e4 00 02
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 07 00 00 00 07 01 04 04 00 70 00 00
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 08 00 00 00 06 01 04 13 9b 00 01
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 08 00 00 00 05 01 04 02 08 d9
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 09 00 00 00 06 01 04 15 e6 00 02
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 09 00 00 00 07 01 04 04 ff 7d ff ff
[grid  ] TRACE 2025/05/02 10:21:40 modbus: send 00 0a 00 00 00 06 01 04 13 9c 00 01
[grid  ] TRACE 2025/05/02 10:21:40 modbus: recv 00 0a 00 00 00 05 01 04 02 08 e4
Power:          -4W
Energy:         5338.2kWh
Current L1..L3: 0.165A 0.494A -0.576A

battery
-------
[battery] TRACE 2025/05/02 10:21:40 modbus: send 00 0b 00 00 00 06 01 04 32 c8 00 01
[battery] TRACE 2025/05/02 10:21:40 modbus: recv 00 0b 00 00 00 05 01 04 02 00 0b
[battery] TRACE 2025/05/02 10:21:40 modbus: send 00 0c 00 00 00 06 01 04 32 dc 00 01
[battery] TRACE 2025/05/02 10:21:40 modbus: recv 00 0c 00 00 00 05 01 04 02 00 6b
[battery] TRACE 2025/05/02 10:21:40 modbus: send 00 0d 00 00 00 06 01 04 32 dd 00 01
[battery] TRACE 2025/05/02 10:21:40 modbus: recv 00 0d 00 00 00 05 01 04 02 12 31
[battery] TRACE 2025/05/02 10:21:40 modbus: send 00 0e 00 00 00 06 01 04 32 e2 00 02
[battery] TRACE 2025/05/02 10:21:40 modbus: recv 00 0e 00 00 00 07 01 04 04 a1 6c 00 00
[battery] TRACE 2025/05/02 10:21:40 modbus: send 00 0f 00 00 00 06 01 04 32 de 00 01
[battery] TRACE 2025/05/02 10:21:40 modbus: recv 00 0f 00 00 00 05 01 04 02 02 89
Power:        -4657W
Energy:       4132.4kWh
Soc:          65%
Controllable: true

/config # 
```
Vielleicht hatte es die DB zerhauen...

andig, 2025-05-02:
Das wäre sehr mekrwürdig. Kannst Du nochmal DB probieren und schauen ob `evcc meter` was sagt? Falls ja: was ist anders als bei der alten Version?

MrMontesa, 2025-05-02:
ok!
```
/config # evcc migrate --log trace
[main  ] INFO 2025/05/02 11:46:44 evcc 0.203.4
[main  ] INFO 2025/05/02 11:46:44 using config file: /config/evcc.yaml
[db    ] INFO 2025/05/02 11:46:44 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/05/02 11:46:44 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/05/02 11:46:44 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/05/02 11:46:44 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/05/02 11:46:44 SELECT * FROM `settings` 0 <nil>
[main  ] INFO 2025/05/02 11:46:44 migrating:
[main  ] INFO 2025/05/02 11:46:44 - global settings
[main  ] INFO 2025/05/02 11:46:44 - network
[main  ] INFO 2025/05/02 11:46:44 - mqtt
[main  ] INFO 2025/05/02 11:46:44 - influx
[main  ] INFO 2025/05/02 11:46:44 - hems
[main  ] INFO 2025/05/02 11:46:44 - eebus
[main  ] INFO 2025/05/02 11:46:44 - modbusproxy
[main  ] INFO 2025/05/02 11:46:44 - messaging
[main  ] INFO 2025/05/02 11:46:44 - tariffs
[main  ] INFO 2025/05/02 11:46:44 - circuits
[main  ] INFO 2025/05/02 11:46:44 - device configs
[db    ] TRACE 2025/05/02 11:46:44 INSERT INTO `settings` (`key`,`value`) VALUES ("interval","30000000000"),("sponsorToken","eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3OTAzMzc2MDAsImlhdCI6MTY5NTcyOTYwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6Ik1yTW9udGVzYSIsInNyYyI6ImdoIn0.Ym7qm8557fROgFgZlI35eqP5aeuvHtyENbqSNqPztCM"),("network","{""schema"":""http"",""host"":""evcc.local"",""port"":7070}"),("mqtt","{""broker"":""localhost:1883"",""user"":""mqtt-user"",""password"":""section"",""clientID"":""evcc"",""insecure"":false,""caCert"":"""",""clientCert"":"""",""clientKey"":"""",""topic"":""evcc""}"),("influx","{""url"":"""",""database"":"""",""token"":"""",""org"":"""",""user"":"""",""password"":"""",""insecure"":false}") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 5 <nil>
```
server restart und `experimental=on`
```
/config # evcc meter
[main  ] INFO 2025/05/02 11:49:24 evcc 0.203.4
[main  ] INFO 2025/05/02 11:49:24 using config file: /config/evcc.yaml
[mqtt  ] ERROR 2025/05/02 11:49:24 tcp://localhost:1883 connection lost: EOF
pv
--
Power:  6238W
Energy: 18165.6kWh

grid
----
Power:          -6114W
Energy:         5338.2kWh
Current L1..L3: -8.75A -8.5A -9.7A

battery
-------
Power:        0W
Energy:       4132.4kWh
Soc:          100%
Controllable: true
```
Aktuell würde ich sagen: Sieht normal aus. Der Test-container mit der alten Version zeigt das gleiche an und gibt im `evcc meter` auch das gleiche aus.