evcc-io/evcc #20811: Phoenix Chargecontroller  Modbus Voltage wrong

jollyjinx, 2025-04-23:
### Describe the bug

Beim Phoenix Charge Controller (em-cp-pp-eth) wird scheinbar der Faktor der für die Umrechnung der Voltage Werte benutzt wird nicht ? oder korrekt ausgelesen.
Im log sehe ich:

[em-eth] TRACE 2025/04/23 17:26:13 modbus: recv 00 52 00 00 00 0f b4 04 0c 59 1c 00 00 00 00 00 00 00 00 00 00
[lp-1 ] DEBUG 2025/04/23 17:26:13 charge voltages: [22.8 0 0]V

also in klartext:

00 07 | Transaction ID
-- | --
00 00 | Protocol ID (0 = Modbus)
00 0f | Length (15 bytes to follow)
b4 | Unit ID (slave address = 0xB4)
04 | Function Code = 0x04 (Read Input Registers)
0c | Byte count = 12 bytes (6 registers = 3 * 2 bytes)
58 f4 | Register 0: 0x58f4 = 22772 (big-endian)
00 00 (x5) | Registers 1-5: All 0x0000

da wird also der wert 22772 mit 0.001 multipliziert statt mit 0.01
22812 / 100.0 = 228.12V  ✅
22812 / 1000.0 = 22.812V  ❌

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Benutzte Hardware:
- evcc docker container auf rpi4 
- sma Homemanager
- sma Solarwechselrichter (allerdings via mqtt in evcc eingetragen)
- Phoenix Chargecontroller em-cp-pp-eth mit Firmware Version 2.1.3 der via Modbus TCP angeschlossen ist,  Aktuell nur Einphasig angeschlossen damit ich feingranularer die PV ausnütze , daher sind die anderen Phasen 0V.
- Eastron SDM720 Zähler der an dem Phoenix via Mobus RS585 angeschlossen

In der evcc Config habe ich den phoenix-em-eth ausgewählt
```

### Log details

```text
[site ] DEBUG 2025/04/23 17:26:13 ----
[em-eth] TRACE 2025/04/23 17:26:13 modbus: send 00 50 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/04/23 17:26:13 modbus: recv 00 50 00 00 00 07 b4 04 04 00 83 00 00
[lp-1 ] DEBUG 2025/04/23 17:26:13 charge power: 1310W
[em-eth] TRACE 2025/04/23 17:26:13 modbus: send 00 51 00 00 00 06 b4 04 00 72 00 06
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/loadpoints/1/chargePower: '1310'
[em-eth] TRACE 2025/04/23 17:26:13 modbus: recv 00 51 00 00 00 0f b4 04 0c 16 b2 00 00 00 00 00 00 00 00 00 00
[lp-1 ] DEBUG 2025/04/23 17:26:13 charge currents: [5.81 0 0]A
[site ] DEBUG 2025/04/23 17:26:13 grid power: -310W
[site ] DEBUG 2025/04/23 17:26:13 grid powers: [975 -668 -617]W
[site ] DEBUG 2025/04/23 17:26:13 grid currents: [4.4 -2.93 -2.75]A
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/controllable: ''
[influx] TRACE 2025/04/23 17:26:13 write chargePower=map[value:1310] (map[loadpoint:Ladestation vehicle:Admiral])
[influx] TRACE 2025/04/23 17:26:13 write gridPower=map[value:-310.5] (map[])
[site ] DEBUG 2025/04/23 17:26:13 pv 1 power: 1991W
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/powers: '3'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/powers/1: '974.7'
[site ] DEBUG 2025/04/23 17:26:13 site power: -310W
[em-eth] TRACE 2025/04/23 17:26:13 modbus: send 00 52 00 00 00 06 b4 04 00 6c 00 06
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/batteryGridChargeActive: 'false'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pvPower: '1991'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/title: ''
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/powers/2: '-668'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pv/1/controllable: ''
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pvEnergy: '0'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/loadpoints/1/smartCostActive: 'false'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pv: '1'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pv/1/title: ''
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pv/1/icon: ''
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/currents: '3'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/pv/1/currents: ''
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/currents/1: '4.4'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/powers/3: '-617.3'
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/power: '-310.5'
[em-eth] TRACE 2025/04/23 17:26:13 modbus: recv 00 52 00 00 00 0f b4 04 0c 59 1c 00 00 00 00 00 00 00 00 00 00
[lp-1 ] DEBUG 2025/04/23 17:26:13 charge voltages: [22.8 0 0]V
[influx] TRACE 2025/04/23 17:26:13 write gridEnergy=map[value:12562.0978] (map[])
[mqtt ] TRACE 2025/04/23 17:26:13 send evcc/site/grid/energy: '12562'
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.2

premultiply, 2025-04-23:
Wie ist denn der Controller selbst bzgl. angeschlossenem Zähler konfiguriert?

jollyjinx, 2025-04-23:
Die angezeigten Werte in der UI sind korrekt:

[image]
</br>

konfiguration der modbus werte:

[image]
</br>

evtl. muss man den faktor noch auslesen den der Phoenix verwendet , laut Dokumentation:

https://asset.conrad.com/media10/add/160267/c1/-/en/000554895ML02/gebruiksaanwijzing-554895-em-cp-pp-eth-oplader.pdf

kann man den conversion factor bei Addresse 352 32bittig finden (Seite 103) 
bei mir sehe ich da,  wie in der Oberfläche 0.1

premultiply, 2025-04-23:
Dann trag doch in der Konfiguration den richtigen Faktor 1.0 statt 0.1 ein.

jollyjinx, 2025-04-23:
Der Faktor von 0,1 ist ja für den Phoenix Charger korrekt, damit es in der Phoenix Oberfläche richtig dargestellt wird. Einzig evcc stellt den Wert falsch im log dar und trägt den dann auch falsch in das mqtt ein.

premultiply, 2025-04-23:
Ist das wirklich so? Erscheint mir irgendwie Merkwürdig.
Soweit ich weiss liefert der Eastron-Zähler seine Daten mit Faktor 1 als Float in der SI-Einheit.
Ich würde nicht verstehen wo dabei ein Faktor 0.1 herkommen sollte.

jollyjinx, 2025-04-24:
Ja das ist so, gerade nochmal ausprobiert:

[image]

[image]

andig, 2025-04-25:
> evtl. muss man den faktor noch auslesen den der Phoenix verwendet , laut Dokumentation:

Das passt ja vorne und hinten nicht. Wir dividieren durch 1000. Da einen Faktor von 0.1 stattdessen zu nutzen wäre ja auch falsch- es sei denn es gäbe eine div 100 * "konfigurierter Faktor" Logik. 

Falls das niemand von Phoenix erklären könnte würde ich hier als "works as implemented" schließen.

jollyjinx, 2025-04-25:
Ich denke der factor 1000 ist default wenn factor für das Messgerät 1.0 ist. Man muss also nur den (originalwert / ( 1000.0 * factor für messgerät ) rechnen und  schon geht das für alles messgeräte korrekt:

realvoltage  =  MODBUS_VOLTAGE  / ( 1000.0 * MODBUS_FACTOR )

andig, 2025-04-27:
Klingt vernünftig. @premultiply ?

jollyjinx, 2025-05-20:
ich habe schon in dem andern thread geschrieben - wie kann ich das denn testen ? ich habe hier einen raspberry pi4 mit ubuntu und benutze den docker container evcc/evcc ?
ich habe keine entwicklungsumgebung hier für evcc sonst hätte ich das selbst kurz gefixt.

maikheinrich, 2025-06-07:
Hallo zusammen,

ich habe mit meinem Phoenix EM-ETH und einem Phoenix EEM-350 das gleiche Problem. Im Webinterface des EM-ETH ist die angezeigt Spannung korrekt, aber EVCC zeigt diese falsch an.

Aufgefallen ist es mir allerdings erst in der Version 0.204.1 da dort in der WebUI unter dem Ladepunkt 45 W Ladeleistung anstelle von 4,5kW angezeigt wurden. 

Hier einmal der Output von evcc charger Version 0.204.1

[main  ] INFO 2025/06/03 18:26:16 evcc 0.204.1
[main  ] INFO 2025/06/03 18:26:16 using config file: /etc/evcc.yaml
evcc_3
------
Charge status: A
Enabled:       true

evcc_1
------
Power:          0W
Energy:         2561.8kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 2.36V 2.37V 2.36V
Charge status:  A
Enabled:        false
Duration:       2h23m16s
Max Current:    6.0A

evcc_2
------
Power:          29W
Energy:         14396.5kWh
Current L1..L3: 6.12A 0.496A 6.1A
Voltage L1..L3: 2.37V 2.37V 2.36V
Charge status:  C
Enabled:        true
Duration:       25m59s
Max Current:    6.0A

und hier die ältere Version 0.203.6. Auch hier ist es falsch. Ist mir allerdings nicht aufgefallen, da in der WebGUI von EVCC die Ladeleistung passt.

[main  ] INFO 2025/06/03 18:23:29 evcc 0.203.6
[main  ] INFO 2025/06/03 18:23:29 using config file: /etc/evcc.yaml
evcc_3
------
Charge status: A
Enabled:       true

evcc_2
------
Power:          4310W
Energy:         14396.4kWh
Current L1..L3: 6.11A 6.15A 6.1A
Voltage L1..L3: 23.8V 23.6V 23.7V
Charge status:  C
Enabled:        true
Duration:       24m10s
Max Current:    6.0A

evcc_1
------
Power:          0W
Energy:         2561.8kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 23.8V 23.6V 23.7V
Charge status:  A
Enabled:        false
Duration:       2h23m16s
Max Current:    6.0A

Falls benötigt kann ich gerne weitere Test durchführen.

Liebe Grüße Maik

andig, 2025-06-07:
Bitte

    evcc meter --log trace

maikheinrich, 2025-06-07:
Hi andig,

hier der output. Leider sehe ich hier keinen Ladepunkt, sondern nur Netz, Batterie und Wewchselrichter.

root@evcc:~/projects# evcc meter --log trace --database /var/lib/evcc/evcc.db
[main  ] INFO 2025/06/07 14:00:52 evcc 0.203.6
[main  ] INFO 2025/06/07 14:00:52 using config file: /etc/evcc.yaml
[db    ] INFO 2025/06/07 14:00:52 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/06/07 14:00:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT * FROM `settings` 36 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/07 14:00:52 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
e3dc_can_4
----------
[e3dc_can_4] TRACE 2025/06/07 14:00:52 modbus: send 00 01 00 00 00 06 01 03 9c e1 00 02
[e3dc_can_4] TRACE 2025/06/07 14:00:52 modbus: recv 00 01 00 00 00 07 01 03 04 00 04 00 00
[e3dc_can_4] TRACE 2025/06/07 14:00:52 modbus: send 00 02 00 00 00 06 01 03 9c e3 00 02
[e3dc_can_4] TRACE 2025/06/07 14:00:52 modbus: recv 00 02 00 00 00 07 01 03 04 00 0a 00 00
[e3dc_can_4] TRACE 2025/06/07 14:00:52 modbus: send 00 03 00 00 00 06 01 03 9c e5 00 02
[e3dc_can_4] TRACE 2025/06/07 14:00:52 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 00 00
Power: 14W

e3dc_netz
---------
[e3dc_netz] TRACE 2025/06/07 14:00:52 modbus: send 00 04 00 00 00 06 00 03 9c 89 00 02
[e3dc_netz] TRACE 2025/06/07 14:00:52 modbus: recv 00 04 00 00 00 07 00 03 04 00 41 00 00
Power: 65W

e3dc_pv_int
-----------
[e3dc_pv_int] TRACE 2025/06/07 14:00:52 modbus: send 00 05 00 00 00 06 01 03 9c 83 00 02
[e3dc_pv_int] TRACE 2025/06/07 14:00:52 modbus: recv 00 05 00 00 00 07 01 03 04 1c 15 00 00
Power: 7189W

e3dc_pv_ext
-----------
[e3dc_pv_ext] TRACE 2025/06/07 14:00:52 modbus: send 00 06 00 00 00 06 01 03 9c 8b 00 02
[e3dc_pv_ext] TRACE 2025/06/07 14:00:52 modbus: recv 00 06 00 00 00 07 01 03 04 ea 98 ff ff
Power: 5480W

e3dc_battery
------------
[e3dc_battery] TRACE 2025/06/07 14:00:52 modbus: send 00 07 00 00 00 06 00 03 9c 85 00 02
[e3dc_battery] TRACE 2025/06/07 14:00:52 modbus: recv 00 07 00 00 00 07 00 03 04 07 6b 00 00
[e3dc_battery] TRACE 2025/06/07 14:00:52 modbus: send 00 08 00 00 00 06 00 03 9c 92 00 01
[e3dc_battery] TRACE 2025/06/07 14:00:52 modbus: recv 00 08 00 00 00 05 00 03 02 00 5e
Power:    -1899W
Soc:      94%
Capacity: 38.0kWh

aktuelle Version teste ich gleich

LG Maik

maikheinrich, 2025-06-07:
Hier die Version 0.204.2

root@evcc:~/projects# evcc meter --log trace --database /var/lib/evcc/evcc.db
[main  ] INFO 2025/06/07 14:11:40 evcc 0.204.2
[main  ] INFO 2025/06/07 14:11:40 using config file: /etc/evcc.yaml
[db    ] INFO 2025/06/07 14:11:40 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/06/07 14:11:40 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT * FROM `settings` 36 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/07 14:11:40 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
e3dc_can_4
----------
[e3dc_can_4] TRACE 2025/06/07 14:11:40 modbus: send 00 01 00 00 00 06 01 03 9c e1 00 02
[e3dc_can_4] TRACE 2025/06/07 14:11:40 modbus: recv 00 01 00 00 00 07 01 03 04 00 04 00 00
[e3dc_can_4] TRACE 2025/06/07 14:11:40 modbus: send 00 02 00 00 00 06 01 03 9c e3 00 02
[e3dc_can_4] TRACE 2025/06/07 14:11:40 modbus: recv 00 02 00 00 00 07 01 03 04 00 0b 00 00
[e3dc_can_4] TRACE 2025/06/07 14:11:40 modbus: send 00 03 00 00 00 06 01 03 9c e5 00 02
[e3dc_can_4] TRACE 2025/06/07 14:11:40 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 00 00
Power: 15W

e3dc_netz
---------
[e3dc_netz] TRACE 2025/06/07 14:11:40 modbus: send 00 04 00 00 00 06 00 03 9c 89 00 02
[e3dc_netz] TRACE 2025/06/07 14:11:40 modbus: recv 00 04 00 00 00 07 00 03 04 ff f4 ff ff
Power: -12W

e3dc_pv_int
-----------
[e3dc_pv_int] TRACE 2025/06/07 14:11:40 modbus: send 00 05 00 00 00 06 01 03 9c 83 00 02
[e3dc_pv_int] TRACE 2025/06/07 14:11:40 modbus: recv 00 05 00 00 00 07 01 03 04 09 f0 00 00
Power: 2544W

e3dc_pv_ext
-----------
[e3dc_pv_ext] TRACE 2025/06/07 14:11:40 modbus: send 00 06 00 00 00 06 01 03 9c 8b 00 02
[e3dc_pv_ext] TRACE 2025/06/07 14:11:40 modbus: recv 00 06 00 00 00 07 01 03 04 f9 0f ff ff
Power: 1777W

e3dc_battery
------------
[e3dc_battery] TRACE 2025/06/07 14:11:40 modbus: send 00 07 00 00 00 06 00 03 9c 85 00 02
[e3dc_battery] TRACE 2025/06/07 14:11:40 modbus: recv 00 07 00 00 00 07 00 03 04 f9 2b ff ff
[e3dc_battery] TRACE 2025/06/07 14:11:40 modbus: send 00 08 00 00 00 06 00 03 9c 92 00 01
[e3dc_battery] TRACE 2025/06/07 14:11:40 modbus: recv 00 08 00 00 00 05 00 03 02 00 5f
Power:    1749W
Soc:      95%
Capacity: 38.0kWh

root@evcc:~/projects#

maikheinrich, 2025-06-07:
oot@evcc:~/projects# evcc charger --database /var/lib/evcc/evcc.db
[main  ] INFO 2025/06/07 14:12:57 evcc 0.204.2
[main  ] INFO 2025/06/07 14:12:57 using config file: /etc/evcc.yaml
evcc_3
------
Charge status: A
Enabled:       true

evcc_1
------
Power:          0W
Energy:         2561.8kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 2.4V 2.39V 2.4V
Charge status:  A
Enabled:        false
Duration:       2h23m16s
Max Current:    6.0A

evcc_2
------
Power:          44W
Energy:         14450.8kWh
Current L1..L3: 6.1A 6.18A 6.1A
Voltage L1..L3: 2.4V 2.39V 2.4V
Charge status:  C
Enabled:        true
Duration:       56m5s
Max Current:    6.0A

root@evcc:~/projects# 

[image]

maikheinrich, 2025-06-07:
Hier nochmal ein Foto von der 0.203.6. Hier ist zumindest in der WebUI alles ok...

[image]

andig, 2025-06-08:
Entschuldige, wir brauchen bitte charger, nicht meter. Mein Fehler 🤦

maikheinrich, 2025-06-08:
Hi andig,

hier der output von 0.203.6

#root@evcc:~#evcc charger --log trace --database /var/lib/evcc/evcc.db
[main  ] INFO 2025/06/08 12:11:43 evcc 0.203.6
[main  ] INFO 2025/06/08 12:11:43 using config file: /etc/evcc.yaml
[db    ] INFO 2025/06/08 12:11:43 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/06/08 12:11:43 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/08 12:11:43 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/08 12:11:43 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/08 12:11:43 SELECT * FROM `settings` 36 <nil>
[db    ] TRACE 2025/06/08 12:11:44 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/08 12:11:44 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/08 12:11:44 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/08 12:11:44 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 01 00 00 00 07 b4 04 04 5c f8 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 01 00 00 00 07 b4 04 04 5c da 00 00
evcc_3
------
[script] DEBUG 2025/06/08 12:11:44 /bin/bash -c echo A: A
[script] DEBUG 2025/06/08 12:11:44 /bin/bash -c echo true: true
Charge status: A
Enabled:       true

evcc_2
------
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 02 00 00 00 07 b4 04 04 00 00 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 03 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 03 00 00 00 07 b4 04 04 11 e2 00 16
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 04 00 00 00 06 b4 04 00 72 00 06
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 04 00 00 00 0f b4 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 05 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 05 00 00 00 0f b4 04 0c 5c f8 00 00 5c 9e 00 00 5d 02 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 06 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 06 00 00 00 05 b4 04 02 00 42
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 07 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 07 00 00 00 04 b4 01 01 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 08 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 08 00 00 00 07 b4 04 04 21 8a 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 09 00 00 00 06 b4 03 01 2c 00 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 09 00 00 00 05 b4 03 02 00 06
Power:          0W
Energy:         14463.7kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 23.8V 23.7V 23.8V
Charge status:  B
Enabled:        true
Duration:       2h23m6s
Max Current:    6.0A

evcc_1
------
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 02 00 00 00 07 b4 04 04 00 00 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 03 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 03 00 00 00 07 b4 04 04 e8 b4 00 03
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 04 00 00 00 06 b4 04 00 72 00 06
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 04 00 00 00 0f b4 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 05 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 05 00 00 00 0f b4 04 0c 5c da 00 00 5c 9e 00 00 5d 02 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 06 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 06 00 00 00 05 b4 04 02 00 41
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 07 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 07 00 00 00 04 b4 01 01 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 08 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 08 00 00 00 07 b4 04 04 21 94 00 00
[em-eth] TRACE 2025/06/08 12:11:44 modbus: send 00 09 00 00 00 06 b4 03 01 2c 00 01
[em-eth] TRACE 2025/06/08 12:11:44 modbus: recv 00 09 00 00 00 05 b4 03 02 00 06
Power:          0W
Energy:         2561.8kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 23.8V 23.7V 23.8V
Charge status:  A
Enabled:        false
Duration:       2h23m16s
Max Current:    6.0A

maikheinrich, 2025-06-08:
und hier die aktuelle Version

root@evcc:~# evcc charger --log trace --database /var/lib/evcc/evcc.db
[main  ] INFO 2025/06/08 12:14:22 evcc 0.204.2
[main  ] INFO 2025/06/08 12:14:22 using config file: /etc/evcc.yaml
[db    ] INFO 2025/06/08 12:14:22 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/06/08 12:14:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT * FROM `settings` 36 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/08 12:14:22 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 01 00 00 00 07 b4 04 04 5c d0 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 01 00 00 00 07 b4 04 04 5c bc 00 00
evcc_3
------
[script] DEBUG 2025/06/08 12:14:22 /bin/bash -c echo A: A
[script] DEBUG 2025/06/08 12:14:22 /bin/bash -c echo true: true
Charge status: A
Enabled:       true

evcc_2
------
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 02 00 00 00 07 b4 04 04 00 00 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 03 00 00 00 06 b4 03 01 6c 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 03 00 00 00 07 b4 03 04 d7 0a 3c 23
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 04 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 04 00 00 00 07 b4 04 04 11 e2 00 16
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 05 00 00 00 06 b4 03 01 74 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 05 00 00 00 07 b4 03 04 cc cd 3d cc
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 06 00 00 00 06 b4 04 00 72 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 06 00 00 00 0f b4 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 07 00 00 00 06 b4 03 01 66 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 07 00 00 00 0f b4 03 0c 00 00 3f 80 00 00 3f 80 00 00 3f 80
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 08 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 08 00 00 00 0f b4 04 0c 5c d0 00 00 5c 76 00 00 5c da 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 09 00 00 00 06 b4 03 01 60 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 09 00 00 00 0f b4 03 0c 00 00 41 20 00 00 41 20 00 00 41 20
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 0a 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 0a 00 00 00 05 b4 04 02 00 42
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 0b 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 0b 00 00 00 04 b4 01 01 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 0c 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 0c 00 00 00 07 b4 04 04 21 8a 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 0d 00 00 00 06 b4 03 01 2c 00 01
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 0d 00 00 00 05 b4 03 02 00 06
Power:          0W
Energy:         14463.7kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 2.38V 2.37V 2.38V
Charge status:  B
Enabled:        false
Duration:       2h23m6s
Max Current:    6.0A

evcc_1
------
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 02 00 00 00 07 b4 04 04 00 00 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 03 00 00 00 06 b4 03 01 6c 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 03 00 00 00 07 b4 03 04 d7 0a 3c 23
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 04 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 04 00 00 00 07 b4 04 04 e8 b4 00 03
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 05 00 00 00 06 b4 03 01 74 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 05 00 00 00 07 b4 03 04 cc cd 3d cc
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 06 00 00 00 06 b4 04 00 72 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 06 00 00 00 0f b4 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 07 00 00 00 06 b4 03 01 66 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 07 00 00 00 0f b4 03 0c 00 00 3f 80 00 00 3f 80 00 00 3f 80
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 08 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 08 00 00 00 0f b4 04 0c 5c bc 00 00 5c 62 00 00 5c d0 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 09 00 00 00 06 b4 03 01 60 00 06
[... cut ...]
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 0c 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 0c 00 00 00 07 b4 04 04 21 94 00 00
[em-eth] TRACE 2025/06/08 12:14:22 modbus: send 00 0d 00 00 00 06 b4 03 01 2c 00 01
[em-eth] TRACE 2025/06/08 12:14:22 modbus: recv 00 0d 00 00 00 05 b4 03 02 00 06
Power:          0W
Energy:         2561.8kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 2.37V 2.37V 2.38V
Charge status:  A
Enabled:        false
Duration:       2h23m16s
Max Current:    6.0A

andig, 2025-06-08:
Bitte als Code formatieren: `<>` Button benutzen

maikheinrich, 2025-06-08:
> Bitte als Code formatieren: `<>` Button benutzen

erledigt...

maikheinrich, 2025-06-09:
Hi @premultiply,

ich habe eben den nightly build installiert. Am Phoenix Controller werden jetzt Spannung und Strom korrekt gelesen. Leider ist jetzt Energy falsch/zu gering. Energy x100 und dann passt es. Weiterhin wird jetzt in der WebUi 0 W am Ladepunkt angezeigt.
Siehe Screenshot.

Liebe Grüße

Maik
`[main  ] INFO 2025/06/09 12:44:00 evcc 0.204.2 (08a5e960)
[main  ] INFO 2025/06/09 12:44:00 using config file: /etc/evcc.yaml
[db    ] INFO 2025/06/09 12:44:00 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/06/09 12:44:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT * FROM `settings` 36 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/09 12:44:00 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 01 00 00 00 07 b4 04 04 5c f8 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 01 00 00 00 07 b4 04 04 5c e4 00 00
evcc_3
------
[script] DEBUG 2025/06/09 12:44:00 /bin/bash -c echo A: A
[script] DEBUG 2025/06/09 12:44:00 /bin/bash -c echo true: true
Charge status: A
Enabled:       true

evcc_2
------
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 02 00 00 00 07 b4 04 04 01 5e 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 03 00 00 00 06 b4 03 01 6c 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 03 00 00 00 07 b4 03 04 d7 0a 3c 23
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 04 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 04 00 00 00 07 b4 04 04 11 ec 00 16
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 05 00 00 00 06 b4 03 01 74 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 05 00 00 00 07 b4 03 04 cc cd 3d cc
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 06 00 00 00 06 b4 04 00 72 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 06 00 00 00 0f b4 04 0c 13 b1 00 00 13 9a 00 00 13 89 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 07 00 00 00 06 b4 03 01 66 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 07 00 00 00 0f b4 03 0c 00 00 3f 80 00 00 3f 80 00 00 3f 80
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 08 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 08 00 00 00 0f b4 04 0c 5c f8 00 00 5c 94 00 00 5d 02 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 09 00 00 00 06 b4 03 01 60 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 09 00 00 00 0f b4 03 0c 00 00 41 20 00 00 41 20 00 00 41 20
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0a 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0a 00 00 00 05 b4 04 02 00 43
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0b 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0b 00 00 00 04 b4 01 01 01
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0c 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0c 00 00 00 07 b4 04 04 00 54 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0d 00 00 00 06 b4 03 01 2c 00 01
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0d 00 00 00 05 b4 03 02 00 06
Power:          0W
Energy:         144.6kWh
Current L1..L3: 5.04A 5.02A 5A
Voltage L1..L3: 238V 237V 238V
Charge status:  C
Enabled:        true
Duration:       1m24s
Max Current:    6.0A

evcc_1
------
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 02 00 00 00 07 b4 04 04 00 00 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 03 00 00 00 06 b4 03 01 6c 00 02
[... 6 lines cut ...]
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 06 00 00 00 0f b4 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 07 00 00 00 06 b4 03 01 66 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 07 00 00 00 0f b4 03 0c 00 00 3f 80 00 00 3f 80 00 00 3f 80
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 08 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 08 00 00 00 0f b4 04 0c 5c e4 00 00 5c a8 00 00 5d 02 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 09 00 00 00 06 b4 03 01 60 00 06
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 09 00 00 00 0f b4 03 0c 00 00 41 20 00 00 41 20 00 00 41 20
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0a 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0a 00 00 00 05 b4 04 02 00 41
[... cut ...]
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0c 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0c 00 00 00 07 b4 04 04 21 94 00 00
[em-eth] TRACE 2025/06/09 12:44:00 modbus: send 00 0d 00 00 00 06 b4 03 01 2c 00 01
[em-eth] TRACE 2025/06/09 12:44:00 modbus: recv 00 0d 00 00 00 05 b4 03 02 00 06
Power:          0W
Energy:         25.6kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 238V 237V 238V
Charge status:  A
Enabled:        false
Duration:       2h23m16s
Max Current:    6.0A`

[image]

maikheinrich, 2025-06-09:
Hi @premultiply 

anbei die Screenshots von einen Ladepunkt. Der 2. ist identisch konfiguriert (hat halt nur andere Zählstände)...

Vielen Dank

Maik

[image]
[image]

andig, 2025-06-11:
@premultiply willst du erstmal einen revert PR machen damit master wieder fehlerfrei ist? Es sieht aus, als ob auch hier kein Test passieren würde- den zweiten Teil sollten wir uns dann sparen :(

andig, 2025-06-14:
@premultiply PR hat den Scaler entfernt...

closed 2025-06-18 by andig

andig, 2025-06-24:
Hallelujah :)

PR #21739 (2025-06-12): Phoenix: fix EM ETH measurand scaling
Fixes #20811 (second attempt)

...and brings other minor code fixes & improvements to EM ETH implementation.

/cc @jollyjinx @maikheinrich