evcc-io/evcc #20812: Improve heating visualization and energy computation

chhtz, 2025-04-23:
**Is your feature request related to a problem? Please describe.**

Heatpump visualization does not represent what heatpump actually does, but what evcc suggests it to do. E.g. "Heizen" or "Standby" even though the HP will decide what it actually does.

**Describe the solution you'd like**

If the heatpump allows reading its current mode (heating buffer or heating boiler or standby (advanced: defrost, etc)) take that for visualization and to log energy consumption (if possible separately for water and buffer). Alternatively, take a threshold below which HP is assumed to be in Standby mode.

It would also be great if two buffer temperatures can be visualized below each other.

**Describe alternatives you've considered**
Regarding separate temperatures, and alternative might be to create separate HPs, but that sounds complicated, especially it won't solve the problem of detecting when the HP starts heating on its own (which it of course still should be able to do) 

**Additional context**

Issue #19753 is strongly related, though it seems to be more focussed on actual behavior. 

(Ich spreche auch deutsch, falls das die Diskussion vereinfacht, das Template war aber englisch)

chhtz, 2025-04-23:
Another thing that could be visualized is the heat output of the HP (from which an average COP could be computed)

Tombra1889, 2025-04-23:
COP hatte ich schonmal vorgeschlagen, wäre dann aber wohl verfälscht durch den aktuellen Energieabfluss aus dem Heizsystem: https://github.com/evcc-io/evcc/discussions/1875#discussioncomment-10714050

Eine Anlieferung im Template via mqtt wäre aber ggf. etwas, dann kann man selbst entscheiden, ob man einen Wert anzeigt, der ggf. vom Hersteller kommt oder man mit der eigenen Berechnung leben kann.

chhtz, 2025-04-23:
Ja, selbst berechnen fände ich auch schwierig/ungenau. Man kann halt die Hersteller-Werte anzeigen, wenn sie sowieso zur Verfügung stehen.

closed 2025-04-24 by andig