evcc-io/evcc #20813: Tasmota-SML zeigt falschen Energiewert

VolkerK62, 2025-04-23:
### Describe the bug

mit Version 0.203.0 wird richtigerweise der Bezugswert ausgegeben
mit Version 0.203.2 wird der Einspeisewert ausgegeben

Vermutlich durch #19072 

### Steps to reproduce

.

### Configuration details

```yaml
- name: grid_tasmota
  type: template
  template: tasmota-sml
  usage: grid
  host: 192.168.178.27
```

### Log details

```text
Output mit 0.203.0
wsl@DESKTOP-7RU7VSU:~$ evcc meter grid_tasmota
[main  ] INFO 2025/04/23 18:16:13 evcc 0.203.0
Power:  45W
Energy: 12433.0kWh

Output mit 0.203.2
wsl@DESKTOP-7RU7VSU:~$ evcc meter grid_tasmota --log trace
[main  ] INFO 2025/04/23 18:31:37 evcc 0.203.2 (36330ac2)
[tasmota] TRACE 2025/04/23 18:31:37 GET http://192.168.178.27/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2025/04/23 18:31:37 {"StatusSNS":{"Time":"2025-04-23T17:31:37","SML":{"Power_curr":44,"Total_in":12433,"Total_out":21792}}}
Power:  44W
Energy: 21792.0kWh
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.2 (36330ac2)

VolkerK62, 2025-04-23:
liegt hier evtl der Fehler?
https://github.com/evcc-io/evcc/blob/bacf49fd12473fe56bbf86c70cfbb3e9eb978f5c/meter/tasmota/connection.go#L217

thierolm, 2025-04-24:
Aaaaah, richtig, da hab ich vor lauter Shelly Änderungen ...Out genommen, was aus Sicht des SML Zählers ja anders rum ist ...

PR #20826 (2025-04-24): Shelly: use TotalIn for sml totalenergy
Fixes #20813
closed 2025-04-24 by andig