evcc-io/evcc #20823: Hyundai Bluelink auto start Laden

DavidBachDerEchte, 2025-04-24:
**Is your feature request related to a problem? Please describe.**
Ich habe einen Kona und eine ABB Wallbox. Wenn EVCC das Auto laden will, kommt oft "Warten auf Fahrzeug". Dann muss ich erst in die Hyundai Bluelink App hinein gehen und das Auto via die App Starten zum Laden.

**Describe the solution you'd like**
Es wäre super, wenn das EVCC direkt regeln kann, dann muss ich nicht immer auf die EVCC App schauen und zu hoffen, dass das Auto lädt.

DavidBachDerEchte, 2025-04-24:
Bzw. Wenn es schon sowas gibt, wie kann ich das dann in meine Config aufnehmen, damit es klappt? (könnt ihr mir eine Anleitung geben?)

VolkerK62, 2025-04-24:
Hmm ... hier ist ein ähnliches Problem https://github.com/evcc-io/evcc/discussions/20520#discussioncomment-12923529

Kia (aber ohne Bluelink) und ABB Wallbox. Braucht auch einen Anschubser (Kabel raus/rein), damit das Laden funktioniert.

leo-olecrt, 2025-04-24:
Pretty good problem, thank you