evcc-io/evcc #20828: upgrading 0.203.1->0.203.2 fails "easee home" charger with timeout; server won't start

t-rood, 2025-04-24:
### Describe the bug

I'm using the docker version. Running state **0.203.1** with two configured **easee home** chargers. Upgrading to **0.203.2** fails with "timeout" when creating the charger objects. Everything is configured in evcc.yaml. Nightly also fails in the same way.

log excerpt:

`[main ] FATAL 2025/04/24 09:12:53 charger [my_charger2] cannot create charger 'my_charger2': cannot create charger type 'template': cannot create charger type 'easee': i/o timeout
`

The same fails when trying to add the charger through UI. Rolling back to 0.203.1 restores functionality.

Can somebody look into what happened there? Must be related to API/template internals. Easee app, web and API seem fine. Easee integrations with tibber still work fine.

Thank you!

### Steps to reproduce

1. use config as above, using valid credentials and charger serials
2. compare 0.203.1 and 0.203.2
3. fails on startup
4. even cannot be configured through the UI, same timeout

### Configuration details

```yaml
chargers:
  - name: my_charger1
    type: template
    template: easee
    user: ****
    password: ****
    charger: EH****
    timeout: 20s
  - name: my_charger2
    type: template
    template: easee
    user: ****
    password: ****
    charger: EH****
    timeout: 20s
```

### Log details

```text
[main ] FATAL 2025/04/24 09:12:53 charger [my_charger2] cannot create charger 'my_charger2': cannot create charger type 'template': cannot create charger type 'easee': i/o timeout
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.2

StefanSchoof, 2025-04-24:
Fixed in master, https://github.com/evcc-io/evcc/issues/20791

t-rood, 2025-04-24:
oops. I missed that. That you.

closed 2025-04-24 by t-rood