evcc-io/evcc #20829: adding solar tariffs in evcc.yaml cause server failure on start // 0.203.1

t-rood, 2025-04-24:
### Describe the bug

when I add my two solar strings into tariffs, the server won't launch any more and even doesn't accept my admin password. Removing the configured tarif items restores functionality.

### Steps to reproduce

1. add the shown config sections under "tariffs:" as the last item
2. try to launch the server
3. failed

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: template
    template: tibber
    token: ****
  feedin:
    type: fixed
    price: 0.082 # EUR/kWh
  co2:
    type: template
    template: green-grid-compass
    apiKey: ****
    zone: DE_LU
  solar:
    - type: template
      template: forecast-solar
      lat: 51.2213
      lon: 6.9073
      dec: 35 # 0 = horizontal, 90 = vertikal
      kwp: 5.23
      az: 57 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
      interval: 1h
    - type: template
      template: forecast-solar
      lat: 51.2213
      lon: 6.9073
      dec: 35 # 0 = horizontal, 90 = vertikal
      kwp: 4.79
      az: -123 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
      interval: 1h
```

### Log details

```text
cannot access log/UI, admin login is blocked, then
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

andig, 2025-04-24:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-04-24 by andig

t-rood, 2025-04-24:
sorry, was caught by UFT-8 encoding in configuration file. my bad.