evcc-io/evcc #20834: easee /// kein automatische Umschalten 3- auf 1-phasig (203.2)

detheese, 2025-04-24:
### Describe the bug

Moin,

Seit dem Update habe ich folgendes Problem: zwei Easee Wallboxen mit Prio 1 und 2. Bei entsprechendem Überschuss startet die Easee nur mit 3 Phasen — stoppt dann und geht nicht runter auf passende 3 KW mit 1 Phase. Es passiert nichts.
Stellte ich manuell um auf 1-phasig läuft es wie gewohnt

### Steps to reproduce

1. automatische Phasenumstaltung am Loadpunkt
2. neustart evcc
3. laden beginnt mit 11 kW 
4. Ladestopp und Ende Gelände (bei Überschuss um 3-4 kW) 
5.  Batterieeinstellung stimmt

### Configuration details

```yaml
## minimal configuration example

sponsortoken: xxxxxdNU

plant: 1xxxxxxxb36

interval: 50s

site:
  title: Überschuss-Laden # display name for UI
  
  residualPower: 100
  meters:
    grid: my_grid
    pv:
      - my_pv
      - my_pv2
    battery:
      - my_battery
    

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Garage # display name for UI
    charger: my_charger2 # charger
    vehicle: cupra # default vehicle
    priority: 2
    mode: pv
    

  - title: Hof # display name for UI
    charger: my_charger  # charger
    vehicle: tesla # default vehicle
    priority: 1
    mode: pv
    

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  # replace with your real grid meter
  - name: my_grid
    type: template
    template: huawei-sun2000
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 16
    host: 192.168.100.199 # Hostname
[... 783 lines cut ...]
[site  ] DEBUG 2025/04/24 15:16:57 pv 1 power: 1962W
[site  ] DEBUG 2025/04/24 15:17:07 pv power: 3409W
[site  ] DEBUG 2025/04/24 15:17:09 grid currents: [-4.07 -3.3 7.4]A
[site  ] DEBUG 2025/04/24 15:17:10 battery 1 soc: 63%
[site  ] DEBUG 2025/04/24 15:17:11 site power: -2731W (including 2579W prioritized power)
[lp-1  ] DEBUG 2025/04/24 15:17:11 charge total import: 1521.333kWh
[lp-1  ] DEBUG 2025/04/24 15:17:11 charger status: A
[site  ] DEBUG 2025/04/24 15:17:11 solar forecast: accumulated 0.712kWh, produced 0.910kWh, scale 1.278
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

203.2

andig, 2025-04-24:
@detheese könntest Du mal bitte in 30min das Nightly prüfen?

GrimmiMeloni, 2025-04-24:
Das geteilte Log zeigt das beschriebene Problem nicht. Dort wird die ganze Zeit mit einer Phase geladen.

detheese, 2025-04-25:
@GrimmiMeloni ... Dann warten wir mal dass die Sonne hochkommt. Ich denke gegen halb elf haben wir Daten.

andig, 2025-04-25:
> Könnte der Effekt von Gestern an vielen Neustart (Änderungen in der YAML) liegen? Vielleicht hat die Tesla-API da nicht mehr mitgespielt und dieses komische Verhalten erzeugt?

Nein...

closed 2025-04-25 by andig

GrimmiMeloni, 2025-04-25:
> [@GrimmiMeloni](https://github.com/GrimmiMeloni) Was sind die besten Easee-Intervale? Meine Huawei ist mit 45 Sekunden recht träge eingestellt?

Das (Regel)Interval in evcc ist global.
Am Charger kann ein Timeout definiert werden. Da gibt es keinen "besten" Wert. Default ist 20s. Das war einige Zeit ganz OK. Aktuell bin ich mit 25s unterwegs, weil um den Jahreswechsel Easee vermehrt Latenzprobleme hatte.