evcc-io/evcc #20846: Support for Zappi Charger (from myenergy)

alamagar, 2025-04-25:
Hello,
Is the Zappi charger from myenergy supported?

[https://www.myenergi.com/product/zappi-ev-charger/?attribute_pa_tethered=type-2-tethered&attribute_pa_colour=white](url)

I am receiving a proposal to install this charger and I would like to be sure it is supported by EVCC

Many thanks in advance.

andig, 2025-04-25:
See https://github.com/search?q=repo%3Aevcc-io%2Fevcc+zappi

mfuchs1984, 2025-04-25:
It seems it's now possible to use it through coop when activating a so called commercial mode, see https://github.com/evcc-io/evcc/discussions/20441#discussioncomment-12756305

andig, 2025-05-16:
Given https://github.com/evcc-io/evcc/issues/20846#issuecomment-2829528924 it seems there's nothing to do here.

closed 2025-05-16 by andig

mfuchs1984, 2025-05-17:
See https://github.com/evcc-io/evcc/discussions/21270 for a tutorial.