evcc-io/evcc #20848: Get RFID Tag from E.ON Drive vBox pro aka Vestel EVC04

DonRoberto1985, 2025-04-25:
Hi Guys,

I would like to split the bill of our two eletric cars, because one is company owned. At this time I need to use the EVCC App to set the correct car, because EVCC doesn't recognise the RFID Tags.

The Tags are working, I see them in the HMI protocols.

EVCC is connected using ModBus and the RFID cards are working as expected. Plug&Charge is disabled.

BTW: EVCC is also working as expected - except the RFID thing ;)

THX & BR
Robert

mfuchs1984, 2025-04-25:
Should be technically possible since the RFID tag is available through Modbus registers 1516-1530

Session RFID Tag [1516,1530] 15 R String RFID tag of active charge session

premultiply, 2025-04-25:
And you don't need a RFID card if both of your cars are configured correctly and have an online connection.
In this case evcc will figure out automatically which of the cars is connected in most cases.

DonRoberto1985, 2025-04-25:
> And you don't need a RFID card if both of your cars are configured correctly and have an online connection. In this case evcc will figure out automatically which of the cars is connected in most cases.

My company car is a Volvo. When I read  the manual for Volvo, it's more a security issue than a feature. IF Volvo gives you the API access...

Also if a thrid car would like to charge it's good to use the third RFID card, just for documentation :)

andig, 2025-04-25:
...do we know the Modbus registers for RFID?

premultiply, 2025-04-25:
https://github.com/evcc-io/evcc/issues/20848#issuecomment-2829584424 🤷🏻‍♂️

mfuchs1984, 2025-04-25:
Modbus spec is available in https://github.com/evcc-io/evcc/issues/18891

closed 2025-04-25 by andig

DonRoberto1985, 2025-04-25:
dumb question due to fixing this...should I install a nightly or a stable update?

mfuchs1984, 2025-04-29:
Is the wallbox software up to date? Can you run evcc charger with trace logging?

DonRoberto1985, 2025-04-29:
> Is the wallbox software up to date? Can you run evcc charger with trace logging?

E.ON said yes, up2date...

Here comes the Log :)

[evcc-20250429-070438-trace.log](https://github.com/user-attachments/files/19954450/evcc-20250429-070438-trace.log)

mfuchs1984, 2025-04-29:
Which firmware version? Is there a firmware date?

mfuchs1984, 2025-04-29:
Was more thinking about a log from
`evcc charger --log trace`

DonRoberto1985, 2025-04-29:
> Which firmware version? Is there a firmware date?

HMI Softwarestand : | v3.118.0

Leistungsplatine Softwarestand : | 1.0.130.0

DonRoberto1985, 2025-04-29:
> Was more thinking about a log from `evcc charger --log trace`

Oh sorry...

[main  ] INFO 2025/04/29 07:24:22 evcc 0.203.4
[main  ] INFO 2025/04/29 07:24:22 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/29 07:24:22 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/04/29 07:24:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/04/29 07:24:22 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/04/29 07:24:22 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/04/29 07:24:22 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2025/04/29 07:24:23 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/04/29 07:24:23 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/04/29 07:24:23 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/04/29 07:24:23 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 01 00 00 00 06 ff 04 01 94 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: close connection: read tcp 192.168.178.107:51626->192.168.178.110:502: read: connection reset by peer
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 02 00 00 00 06 ff 03 07 d2 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 02 00 00 00 05 ff 03 02 00 14
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 03 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 03 00 00 00 07 ff 04 04 00 00 29 04
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 04 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 04 00 00 00 07 ff 04 04 00 00 05 bc
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 05 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 05 00 00 00 05 ff 04 02 3a 66
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 06 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 06 00 00 00 05 ff 04 02 3b d8
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 07 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 07 00 00 00 05 ff 04 02 3b c4
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 08 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 09 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 09 00 00 00 05 ff 04 02 00 e6
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 0a 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 e6
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 0b 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 03
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 0c 00 00 00 06 ff 04 03 e9 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 0c 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 0d 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 0d 00 00 00 05 ff 04 02 00 03
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 0e 00 00 00 06 ff 04 03 e9 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 0e 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 0f 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 0f 00 00 00 07 ff 04 04 00 00 26 02
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 10 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 10 00 00 00 07 ff 04 04 00 00 0e 47
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 11 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 11 00 00 00 05 ff 03 02 00 10
[vestel] TRACE 2025/04/29 07:24:23 modbus: send 00 12 00 00 00 06 ff 04 05 ec 00 0f
[vestel] TRACE 2025/04/29 07:24:23 modbus: recv 00 12 00 00 00 03 ff 84 02
Power:          10500W
Energy:         146.8kWh
Current L1..L3: 14.9A 15.3A 15.3A
Voltage L1..L3: 231V 230V 230V
Charge status:  C
Enabled:        true
Charged:        9.7kWh
Duration:       1h0m55s
Max Current:    16.0A
Identifier:     modbus: exception '2' (illegal data address), function '4'

andig, 2025-04-29:
@DonRoberto1985 could you just try to read the registers with any modbus tool? If we can‘t figure out how e‘ll just need to revert this change.

mfuchs1984, 2025-04-29:
I think the firmware 3.118.0 is just too old. When looking at the vestel phase switching thread, these guys have a more recent version 3.187.0. Phase switching is also not working with the outdated 3.118.0 on the EON boxes.

https://github.com/evcc-io/evcc/discussions/11701#discussioncomment-12525761

DonRoberto1985, 2025-04-29:
> [@DonRoberto1985](https://github.com/DonRoberto1985) could you just try to read the registers with any modbus tool? If we can‘t figure out how e‘ll just need to revert this change. Or provide repote access to the modbus port…

[image]

&

[image]

In the Inputregisters I can see some stuff but not for 1516-1530...

DonRoberto1985, 2025-04-29:
> I think the firmware 3.118.0 is just too old. When looking at the vestel phase switching thread, these guys have a more recent version 3.187.0. Phase switching is also not working with the outdated 3.118.0 on the EON boxes.
> 
> [#11701 (comment)](https://github.com/evcc-io/evcc/discussions/11701#discussioncomment-12525761)

THX for this hint...I'm in the updateprocess right now :)

andig, 2025-04-29:
I‘ll wait for retry then

andig, 2025-04-30:
We need a comment in the template that mentions required firmware. PR welcome.

DonRoberto1985, 2025-05-02:
The identifier is now visible correct via evcc charger. Do you have any idea why evcc doesn't recognise my car and shows Guestcar instead?

DonRoberto1985, 2025-05-02:
> Great, first step completed. Can you upload a log file covering the car connection to the charger?

Of cource. From the WebIF? Trace, Debug or something other? :)

mfuchs1984, 2025-05-02:
Just a debug log covering the moment when you connect the car. Web interface is fine.

DonRoberto1985, 2025-05-02:
> Just a debug log covering the moment when you connect the car. Web interface is fine.

[site  ] DEBUG 2025/05/02 14:50:54 ----
[lp-1  ] DEBUG 2025/05/02 14:50:54 charge power: 0W
[lp-1  ] DEBUG 2025/05/02 14:50:56 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/02 14:50:56 grid power: 160W
[site  ] DEBUG 2025/05/02 14:50:57 pv 1 power: 283W
[site  ] DEBUG 2025/05/02 14:50:57 site power: 160W
[lp-1  ] DEBUG 2025/05/02 14:50:57 charge total import: 190.800kWh
[lp-1  ] DEBUG 2025/05/02 14:50:57 charger status: B
[lp-1  ] INFO 2025/05/02 14:50:57 car connected
[lp-1  ] DEBUG 2025/05/02 14:50:57 vehicle api refresh
[lp-1  ] DEBUG 2025/05/02 14:50:57 charge total import: 190.800kWh
[lp-1  ] DEBUG 2025/05/02 14:50:57 set charge current limit: 16A
[lp-1  ] DEBUG 2025/05/02 14:50:57 charger enable
[lp-1  ] DEBUG 2025/05/02 14:50:57 wake-up timer: start
[site  ] DEBUG 2025/05/02 14:50:57 ----

mfuchs1984, 2025-05-02:
This is how it looks like here (with a different charger)

```
[lp-1 ] INFO 2025/05/02 16:56:39 car connected
[lp-1 ] DEBUG 2025/05/02 16:56:39 vehicle api refresh
[lp-1 ] DEBUG 2025/05/02 16:56:39 charge total import: 7352.202kWh
[lp-1 ] DEBUG 2025/05/02 16:56:39 charger vehicle id: 0C86C7017447
[lp-1 ] INFO 2025/05/02 16:56:39 vehicle updated: unknown -> BMW i4
[lp-1 ] DEBUG 2025/05/02 16:56:39 set charge mode: pv
```

mfuchs1984, 2025-05-02:
The tag ist still in your yaml, format including identation is correct? Do you edit the correct yaml (check the log when starting evcc)? Do you present the tag before or after connecting the car?

DonRoberto1985, 2025-05-02:
> The tag ist still in your yaml, format including identation is correct? Do you edit the correct yaml (check the log when starting evcc)? Do you present the tag before or after connecting the car?

Yes I'm using the correct yaml. Everytime I do a typo I can see that I'm using the correct one ;)

YAML:

[image]

I've also thought about the order card->car or car->card, but this changed nothing.

DonRoberto1985, 2025-05-06:
Update from my side:

- in the experimental view and also in the log I'm able to see the correct RFID tag, but EVCC doesn't switch from guest to custom
- I've also put a test car through the experimental view but this doesn't work too
- I tried quotation marks (single and double)

[image]

Could it have to do with the generic(offline) car I've created?

mfuchs1984, 2025-05-06:
Maybe, can you try without the offline car?

mfuchs1984, 2025-05-06:
@andig while getting the RFID tag works well now with recent firmware, people with older firmware run into problems now, an it seems not to be obvious how to upgrade for normal users, see https://github.com/evcc-io/evcc/discussions/21048. 

Wouldn't it make sense to test the RFID registers in `NewVestel` and only decorate `Identify` when they are readable? 
In addition the documentation can be enhanced to inform users that a certain firmware version is needed for RFID, maybe with some instructions (@DonRoberto1985 the documentation update needs your help then).

andig, 2025-05-06:
Yes, absoltely. Wanna do a PR?

mfuchs1984, 2025-05-06:
Sure, should be possible on Thursday.

DonRoberto1985, 2025-05-07:
> Maybe, can you try without the offline car?

I tried without luck...as simple solution I put my Volvo into the loadpoint. The other car get much less loads. These I need so switch in the UI afterwards.

I've also tried to implement my Volvo native but this is still crap as hell...

mfuchs1984, 2025-06-07:
> > Maybe, can you try without the offline car?
> 
> I tried without luck...as simple solution I put my Volvo into the loadpoint. The other car get much less loads. These I need so switch in the UI afterwards.
> 
> I've also tried to implement my Volvo native but this is still crap as hell...

@DonRoberto1985 you can try using RFID with the new release 0.204.2 again, should hopefully work now, feedback appreciated.