evcc-io/evcc #20861: Toyota integration not working any more

Albert2222222, 2025-04-25:
### Describe the bug

Toyota changed its API a few days ago, so no more SoC etc. is displayed in evcc.

The Toyota Home Assistant Integration encoutered the same issue recently, here is how they solved it:
https://github.com/pytoyoda/ha_toyota/issues/77

This should help evcc as well to solve the issue.

### Steps to reproduce

Add any Toyot vehicle in evcc

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: toyota
    title: bZ4X
    capacity: 64 # Akkukapazität in kWh (optional)
    user: xxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: yyyyy # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    icon: car # Icon in der Benutzeroberfläche (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    mode: # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
    minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)
```

### Log details

```text
[main ] ERROR 2025/04/25 15:19:00 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'toyota': cannot get vehicles: unexpected status: 500 (Internal Server Error)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.3

andig, 2025-04-27:
/cc @thomasbecker

thomasbecker, 2025-04-28:
Gucke ich mir an. Kann noch nicht versprechen, wann.

Albert2222222, 2025-05-05:
Any news here? My bz4x still not working :(

The library the above integration used is this one:
https://github.com/pytoyoda/pytoyoda

A diff from v3.01 to 3.04 of this lib should display the changes needed, I guess just the client version needs to be increased

andig, 2025-05-06:
@Albert2222222 aounds good. Do you want to open a PR?

closed 2025-05-07 by andig