evcc-io/evcc #20865: Since 2 updates evcc does not start my Passat GTE anymore

albertusQuelle, 2025-04-25:
### Describe the bug

After ca 5 attempts it stops trying t oload. 
My evcc is connected. Car as well. 

### Steps to reproduce

1. Select either PV, minPV or Fast
2. Car connected
3. Car is "Ladebereit and connected" but does not start
...

It occurs since 0.203.2

### Configuration details

```yaml
/app # evcc dump --cfg
[main  ] INFO 2025/04/25 21:37:55 evcc 0.203.3

Fehlermeldung:

    Config File "evcc" Not Found in "[/root /app /etc]"

Version: `0.203.3`

But i can find it in Homeassistant

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: fadsfads

interval: 30s # control cycle interval

sponsortoken: fdasfas

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

#modbusproxy:
#  - port: 5021
 #   device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ043VSX-if00-port0
 #   baudrate: 9600
  #  comset: "8N1"

meters:
- type: template
  template: deye-hybrid-3p
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid  
  modbus: rs485serial  
  name: grid1
- type: template
  template: deye-hybrid-3p 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
[... 564 lines cut ...]
[lp-1  ] DEBUG 2025/04/25 21:45:36 charger status: B
[ev-eth] TRACE 2025/04/25 21:45:36 modbus: send 00 19 00 00 00 06 ff 01 01 90 00 01
[ev-eth] TRACE 2025/04/25 21:45:36 modbus: recv 00 19 00 00 00 04 ff 01 01 01
[lp-1  ] DEBUG 2025/04/25 21:45:36 plan: charge 1h10m2s between 2025-04-25 21:49:58 +0200 CEST until 2025-04-25 23:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1  ] TRACE 2025/04/25 21:45:36   slot from: 2025-04-25 21:49:58 +0200 CEST to 2025-04-25 23:00:00 +0200 CEST cost 0.000
[lp-1  ] DEBUG 2025/04/25 21:45:36 wake-up vehicle, attempts left: 4
[id    ] TRACE 2025/04/25 21:45:36 POST https://emea.bff.cariad.digital/vehicle/v1/vehicles/***/charging/start
[id    ] TRACE 2025/04/25 21:45:37 {"data":{"requestID":"8aee8e7e-4417-4849-9d13-2d69665ad90f"}}
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.3

TobiasHuber1980, 2025-04-26:
1. Ich hoffe, das sind nicht deine echten Zugangsdaten! (VW/Tibber)
2. Das Verhalten kommt evtl. vom Planer
```
[lp-1  ] DEBUG 2025/04/25 21:42:35 plan: charge 1h10m2s between 2025-04-25 21:49:58 +0200 CEST until 2025-04-25 23:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1  ] TRACE 2025/04/25 21:42:35   slot from: 2025-04-25 21:49:58 +0200 CEST to 2025-04-25 23:00:00 +0200 CEST cost 0.000
```

andig, 2025-04-26:
Ich kann in dem Log nichts erkennen- könntest Du die ganzen TRACE Zeilen bis auf VW bitte mal raus nehmen?

albertusQuelle, 2025-04-27:
> 1. Ich hoffe, das sind nicht deine echten Zugangsdaten! (VW/Tibber)
> 2. Das Verhalten kommt evtl. vom Planer
>
> ```
> [lp-1  ] DEBUG 2025/04/25 21:42:35 plan: charge 1h10m2s between 2025-04-25 21:49:58 [...]

- Fake Date 😏
- planner ist nun deaktiviert. Trotzdem geht es nicht 

> Ich kann in dem Log nichts erkennen- könntest Du die ganzen TRACE Zeilen bis auf VW bitte mal raus nehmen?
> 

Das verstehe ich nicht 

> > It occurs since 0.203.2
> 
> Hast Du ein Vergleichslog des gleichen Sachverhalts auf funktionierender Version?

Leider nein. An meiner zweiten Wallbox die nicht an evcc hängt geht es aber

andig, 2025-04-27:
> Das verstehe ich nicht

Ich kann in dem Wust nichts erkennen. Trace ausser fürs Auto brauchts hier auch nicht.

albertusQuelle, 2025-04-27:
Wie wäre denn die Query um das zu extrahieren? 
Lg

TobiasHuber1980, 2025-04-27:
ohne configfile startet evcc im demo modus. daher auch die "falschen autos"

albertusQuelle, 2025-04-27:
Und warum startet evcc dann normal ? Ich habe am Configfile nie was verändert und auch auf der UI schaut erstmal alles korrekt aus?

[image]
[image]

andig, 2025-04-27:
Es fehlt weiter das angefragt debug log. Wir können wieder auf machen wenns das gibt.

closed 2025-04-27 by andig

albertusQuelle, 2025-04-27:
Dürfte ich dich bitten mir die Frage zu beantworten 
https://github.com/evcc-io/evcc/issues/20865#issuecomment-2833288418
War das dann oben der erweiterte trace log? 
Etwas Details wären halt schon hilfreich statt gleich zu closen.

andig, 2025-04-27:
```
[lp-1  ] DEBUG 2025/04/25 21:43:43 set charge mode: now
[site  ] DEBUG 2025/04/25 21:43:43 ----
[lp-1  ] DEBUG 2025/04/25 21:43:44 set charge current limit: 16A
[lp-1  ] DEBUG 2025/04/25 21:44:06 charger status: B
```

Dein Auto lädt einfach nicht- warum musst Du VW fragen.

Ob das Aufwecken funktioniert kann man nicht sagen da Du den relevanten Teil weg gelassen hast:

```
[lp-1  ] DEBUG 2025/04/25 21:45:36 wake-up vehicle, attempts left: 4
[id    ] TRACE 2025/04/25 21:45:36 POST https://emea.bff.cariad.digital/vehicle/v1/vehicles/***/charging/start
[id    ] TRACE 2025/04/25 21:45:37 {"data":{"requestID":"8aee8e7e-4417-4849-9d13-2d69665ad90f"}}
```

Leider erfordert die Suche in unvollständigen oder ungeeigneten Logs sehr viel Zeit. Die können wir leider nicht immer zur Verfügung stellen.

albertusQuelle, 2025-04-27:
Habe nun evcc komplett deinstalliert. 
Mit meiner yaml wieder neu installiert. 
Aktuellste version. 

leider das gleiche. 
Habe im Log level as auto auf Trace gestellt. 

Interessant: an der ungesteuerten Wallbox läd das Auto wie gewohnt. 

Hilft dieser Log nun?

`[lp-1  ] DEBUG 2025/04/27 19:45:08 wake-up vehicle, attempts left: 4
[id    ] TRACE 2025/04/27 19:45:08 POST https://emea.bff.cariad.digital/vehicle/v1/vehicles/***/charging/start
[id    ] TRACE 2025/04/27 19:45:08 {"data":{"requestID":"cf04f544-6c24-432b-b58f-db8866b813b4"}}
[db    ] TRACE 2025/04/27 19:45:33 INSERT INTO `settings` (`key`,`value`) VALUES ("lp1.title","Car Port"),("solarAccYield","{""pv2"":{""accumulated"":0.2000000000007276}}"),("adminPassword","$2a$10$FPezj0mwz2hlPdwfNK7qF.a3bTharGnIqQqcjlMoMNbrhmtG4cQjy"),("jwtSecretKey","ed47f7d5c6a47b58ae72de4f818be963501e921121afbc2ea94813d2a8c78b48"),("lp1.mode","now") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 5 <nil>
[site  ] DEBUG 2025/04/27 19:45:37 ----
[lp-1  ] DEBUG 2025/04/27 19:45:37 charge power: 0W
[grid1 ] TRACE 2025/04/27 19:45:37 modbus: send 01 03 02 71 00 01 d5 a9
[grid1 ] TRACE 2025/04/27 19:45:37 modbus: recv 01 03 02 00 18 b8 4e
[site  ] DEBUG 2025/04/27 19:45:37 grid power: 24W`

albertusQuelle, 2025-04-27:
Mit der aktuellen nightly klappts. :-)
Closed und danke für eure zeit

andig, 2025-04-27:
Da hat sich nichts geändert…