evcc-io/evcc #20866: Shelle PM Mini Gen3

petroideal, 2025-04-25:
### Describe the bug

With version 203.01, the Shelly was still recognised and I was able to integrate the meter data as a balcony power station Since .02 and 03 it no longer works. 

All Shelly Config tested 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
As it is no longer possible, I had to remove it.
```

### Log details

```text
As it is no longer possible, I had to remove it.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

VolkerK62, 2025-04-26:
https://github.com/evcc-io/evcc/pull/20855
check with latest nightly version

thierolm, 2025-04-27:
We need a trace log.

petroideal, 2025-04-27:
[image]

danny2630, 2025-04-27:
is it possible that the selection has changed somehow? Or does it not matter? Because the drop down menu only mentions 1PM. 
We have a Shelly Plus PM Mini Gen3. Until recently it worked without any problems

petroideal, 2025-04-27:
It doesn't matter which shelly type I choose, everything has already been tested.

[site ] ERROR 2025/04/27 21:22:23 2025-04-27 21:22:18.899786217 +0200 CEST m=+2.031832979 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:18.912236249 +0200 CEST m=+2.044283018 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:18.940760878 +0200 CEST m=+2.072807640 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:18.982430726 +0200 CEST m=+2.114477491 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:19.05097002 +0200 CEST m=+2.183016791 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:19.109434919 +0200 CEST m=+2.241481687 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:19.188439048 +0200 CEST m=+2.320485812 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:19.513465051 +0200 CEST m=+2.645511814 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:19.800343944 +0200 CEST m=+2.932390709 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:20.290132837 +0200 CEST m=+3.422179603 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:21.433456447 +0200 CEST m=+4.565503212 !! 0ms unknown shelly model: S3PM 2025-04-27 21:22:23.068942335 +0200 CEST m=+6.200989100 !! 0ms unknown shelly model: S3PM
[site ] ERROR 2025/04/27 21:22:23 pv 2 power: unknown shelly model: S3PM

petroideal, 2025-04-27:
In Home Assitens itself, I always receive the data in parallel.

thierolm, 2025-04-28:
Wer kann bitte mal #20919 test.

PR #20919 (2025-04-28): Shelly: add missing PM1 condition
Fixes #20866

## Summary by Sourcery

Bug Fixes:
- Fix missing power, energy, current, and voltage readings for Shelly Gen2 devices identified primarily by a `PM1` endpoint rather than a `Switch` endpoint.
closed 2025-04-28 by andig