evcc-io/evcc #20867: cinfig-UI: Standard-Fahrzeug nicht änderbar

VolkerK62, 2025-04-26:
### Describe the bug

Ich hatte bei der Erstellung des Loadpoint "automatischen Erkennung" ausgewählt. Das hat funktioniert.
Vor ein paar Tagen (evcc Version unklar) habe ich (um etwas zu testen) mein Fahrzeug als Standard ausgewählt. Das hat auch funktioniert.
Problem ist, ich komme jetzt nicht mehr zurück auf "automatische Erkennung".
Ich kann es ändern und speichern, aber nach einem Neustart ist weiterhin das Fahrzeug ausgewählt.
Auch eine Änderung auf ein anderes Fahrzeug ist nicht möglich.
 

### Steps to reproduce

.

### Configuration details

```yaml
volker@raspberrypi:~ $ evcc config
[main  ] INFO 2025/04/26 06:12:12 evcc 0.203.3
[main  ] INFO 2025/04/26 06:12:12 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/26 06:12:12 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/04/26 06:12:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `settings` 154 <nil>
[mqtt  ] INFO 2025/04/26 06:12:12 connecting evcc-232636033 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/04/26 06:12:12 tcp://localhost:1883 connected
[db    ] TRACE 2025/04/26 06:12:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` WHERE `configs`.`class` = 1 4 <nil>
charger
---
db:4 {Type:template Title: Icon: Product:} map[host:localhost port:5200 template:daheimladen-mb]
db:6 {Type:template Title: Icon: Product:} map[ain:***** icon:bike integrateddevice:true password:***** standbypower:10 template:fritzdect uri:http://192.168.178.1 user:*****]
db:8 {Type:template Title: Icon: Product:} map[channel:0 heating:true host:192.168.178.29 icon:heatpump integrateddevice:true standbypower:-1 template:shelly]
db:10 {Type:template Title: Icon: Product:} map[channel:0 heating:true host:192.168.178.21 icon:heatpump integrateddevice:true standbypower:10 template:shelly]

[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` WHERE `configs`.`class` = 2 2 <nil>
meter
---
db:16 {Type:template Title: Icon: Product:} map[host:192.168.178.22 template:shelly-pro-3em usage:grid]
db:18 {Type:template Title:SolarLog Icon: Product:} map[host:192.168.178.23 port:502 template:solarlog usage:pv]

[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` WHERE `configs`.`class` = 3 2 <nil>
vehicle
---
db:2 {Type:template Title: Icon: Product:} map[cache:15m capacity:73 icon:car language:de mode:pv password:***** template:hyundai title:(T)Raumschiff user:*****]
db:17 {Type:template Title: Icon: Product:} map[capacity:73 icon:bus template:offline title:ohne_Bluelink]

[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` WHERE `configs`.`class` = 4 0 <nil>
[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` WHERE `configs`.`class` = 5 4 <nil>
loadpoint
---
db:5 {Type: Title: Icon: Product:} map[charger:db:4 circuit:load defaultMode:off limitEnergy:0 limitSoc:0 maxCurrent:20 meter: minCurrent:6 mode:pv phasesConfigured:1 planEnergy:0 planTime:0001-01-01T00:00:00Z soc:map[estimate:true poll:map[interval:2.16e+13 mode:charging]] thresholds:map[disable:map[delay:1.2e+12 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Carport(L3) vehicle:db:2]
db:7 {Type: Title: Icon: Product:} map[charger:db:6 circuit:main defaultMode: maxCurrent:1 meter: minCurrent:1 mode:off phasesConfigured:1 priority:1 soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Garage vehicle:Vbike]
db:9 {Type: Title: Icon: Product:} map[charger:db:8 circuit:WPSG defaultMode: limitSoc:0 maxCurrent:2 meter: minCurrent:2 mode:off phasesConfigured:1 priority:3 soc:map[estimate:false poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:3e+11 threshold:0] enable:map[delay:1.2e+12 threshold:0]] title:WP-SG+ vehicle:WasserSG]
db:11 {Type: Title: Icon: Product:} map[charger:db:10 circuit: defaultMode:now limitSoc:99 maxCurrent:2 meter: minCurrent:2 mode:now phasesConfigured:1 priority:5 soc:map[estimate:false poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:WP(L2)-Info- vehicle:WasserWP]

[db    ] TRACE 2025/04/26 06:12:12 SELECT * FROM `configs` WHERE `configs`.`class` = 6 0 <nil>
```

### Log details

```text
keine Ahnung, welches Log ich hier einbringen soll. Bitte um entsprechende Hinweise.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.3

andig, 2025-04-26:
@VolkerK62 kannst du mal Browser->Entwicklerkonsole->Netzwerk schauen welche Request/Responses dabei über die Leitung gehen?

VolkerK62, 2025-04-26:
Mach ich heute Abend

VolkerK62, 2025-04-26:
> kannst du mal Browser->Entwicklerkonsole->Netzwerk schauen welche Request/Responses dabei über die Leitung gehen?

Das ist Neuland für mich.
Falls das nciht das Richtige ist, brauch ich ne Wegbeschreibung.

[192.168.178.25.har.txt](https://github.com/user-attachments/files/19924404/192.168.178.25.har.txt)

VolkerK62, 2025-04-27:
Ich habe die Änderung in verschiedenen Browsern und Rechnern probiert. Auch über die Android APP. 
Ich bekomme das Auto nicht mehr aus der Config.

naltatis, 2025-04-27:
Das ist super. Danke!

naltatis, 2025-04-29:
Hm, ich hab gerade versucht es nachzustellen, bekomme das aber nicht hin. Bei mir funktioniert es wie erwartet. In deinem HAR sieht man aber, dass die gewünschte Einstellung (`vehicle: ""`) korrekt an die API übertragen wird, die API auch mit 200 bestätigt, beim lesen danach aber immer noch der vorherige Wert (`vehicle: "db:2"`) zurückgegeben wird.

 PUT /api/config/loadpoints/5
[image]

GET /api/config/loadpoints
[image]

@VolkerK62 magst du mal ein db=trace Log zum Zeitpunkt des Speicherns posten?

@andig hast du eine Idee was hier schiefgehen kann?

andig, 2025-04-29:
Vllt hilft ein trace log der Api Aufrufe plus debug des Loadpoints?

VolkerK62, 2025-04-29:
log level geändert. evcc neu gestartet. Fahrzeug Eintrag auf "automatisch" geändert. Speichern. Neustart.

[db.log](https://github.com/user-attachments/files/19961490/db.log)

VolkerK62, 2025-04-30:
>Hm, ich hab gerade versucht es nachzustellen, bekomme das aber nicht hin. Bei mir funktioniert es wie erwartet.

Ich hab es deshalb mal auf meinem Testsystem (WSL) probiert. Gleiches Verhalten.

naltatis, 2025-05-02:
@VolkerK62 im Datenbanklogging sieht man, dass er weiterhin db:2 als Fahrzeug sieht. Heißt es muss irgendwo auf dem Weg "verloren" gehen. Ich hab im master gerade etwas logging hinzugefügt. Mangst du mit dem nächsten Nightly nochmal testen und sowohl ein Log als auch ein HAR aus dem Browser posten?

Nur um sicherzugehen: anderen Änderungen in dem Dialog (Titel ändern und so) funktionieren bei dir?

VolkerK62, 2025-05-02:
>Nur um sicherzugehen: anderen Änderungen in dem Dialog (Titel ändern und so) funktionieren bei dir?

Hab es gerade mal ausprobiert. Es lässt sich alles ändern (Titel, Priorität, Lademodus, elektr. Anschluss), nur Vehicle nicht.

Ich habe den Fahrzeug-Eintrag am Loadpoint jetzt direkt in der Datenbank gelöscht.
Danach kann ich es im UI problemlos hinzufügen. Nur nicht wieder löschen oder ändern. Es ist wie festgetackert.

Schmaviator, 2025-05-02:
Ich habe dasselbe Problem. siehe #20900

VolkerK62, 2025-05-03:
@naltatis 
>Mangst du mit dem nächsten Nightly nochmal testen und sowohl ein Log als auch ein HAR aus dem Browser posten?

die letzte 4 Nightlys sind gecrasht.

andig, 2025-05-03:
GORM oder Compiler bug?

> ../../../go/pkg/mod/gorm.io/gorm@v1.26.0/internal/stmt_store/stmt_store.go:99:10: cannot use defaultMaxSize (untyped int constant 9223372036854775807) as int value in assignment (overflows)

andig, 2025-05-03:
Build läuft

VolkerK62, 2025-05-04:
Da meine bestehenden Loadpoint schon älter sind, habe ich jetzt für den Test einen Neuen angelegt.
Gleiches Verhalten

[db.log](https://github.com/user-attachments/files/20027123/db.log)

[192.168.178.25.har.txt](https://github.com/user-attachments/files/20027125/192.168.178.25.har.txt)

andig, 2025-05-04:
@VolkerK62 kann ich auch nicht nachvollziehen. Es braucht steps to reproduce. So gehts nicht:

- mit leerer db/config starten: `gor main.go --database foo1.db -c foo1.yaml --disable-auth`
- loadpoint und 2 fahrzeuge anlegen, neustart
- loadpoint fzg a fest zuweisen, neustart
- fzg am Loadpoint kann auf b geändert werden

-> kein Problem

VolkerK62, 2025-05-04:
Keine Ahnung, was bei mir dann anders ist.
Hab es gerade eben genau so gemacht. 
yaml + db gelöscht. evcc gestartet und alles im UI konfiguriert.
Fahrzeugänderung nicht möglich.

Da ihr es nicht reproduzieren könnt, ist eine Problemfindung natürlich mehr als schwierig. Deshalb mach ich hier zu.

andig, 2025-05-04:
Irgendeinen Unterschied muss es ja geben- und den müssten wir finden.

Schmaviator, 2025-05-04:
Schaut euch mal meinen Beitrag hier mit an #20900. Hab alles deinstalliert und bin einen Schritt weitergekommen obwohl es noch nicht zu 100% funktioniert.

chris-e-codes, 2025-05-08:
bei mir tritt es auch auf. Hast du @VolkerK62 zufällig auch RFID Identifier hinterlegt (ja ich habe welche)? Die anderen Tester nicht?

VolkerK62, 2025-05-08:
nein, ohne RFID Identifier.

closed 2025-05-23 by andig