evcc-io/evcc #20869: Tibber-Pulse, Rate limit exceeded on Tibber Websocket API

jpvolkmann, 2025-04-26:
### Describe the bug

Dedicated issue based on comment https://github.com/evcc-io/evcc/issues/20502#issuecomment-2831477665

I got a mail from tibber support, "Rate limit exceeded on Tibber Websocket API". After checking with Tibber support it seems that the Tibber integration in EVCC is connecting the Subscription API at the Tibber backend to often (Limit is 20 / hour).

Checking the EVCC generated logs (TRACE level) I can see a re-connect to Tibber approx. every 60s

```
{"power":61,"powerProduction":0,"lastMeterConsumption":10431.7269,"lastMeterProduction":3081.5861,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/26 07:30:37 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/04/26 07:30:42 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/26 07:30:42 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/26 07:30:42 {"id":"8ac74630-99c9-4799-b4f0-801b4fa4e0b2","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"e1ce067c-c74b-4ba3-b2b1-d05510c8361b"}}} client
[pulse ] TRACE 2025/04/26 07:30:45 {"id":"8ac74630-99c9-4799-b4f0-801b4fa4e0b2","type":"next","payload":{"data":{"liveMeasurement":{"power":40,"powerProduction":0,"lastMeterConsumption":10431.7272,"lastMeterProduction":3081.5861,"currentL1":null,"currentL2":null,"currentL3":null}}}} server

...

[pulse ] TRACE 2025/04/26 07:31:25 {"id":"8ac74630-99c9-4799-b4f0-801b4fa4e0b2","type":"next","payload":{"data":{"liveMeasurement":{"power":68,"powerProduction":0,"lastMeterConsumption":10431.7279,"lastMeterProduction":3081.5861,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/26 07:31:35 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/04/26 07:31:40 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/26 07:31:40 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/26 07:31:40 {"id":"10db274d-6f97-4c69-8407-7afe35e31f1c","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"e1ce067c-c74b-4ba3-b2b1-d05510c8361b"}}} client
```

I'm running EVCC as a add-on to Home Assistant on Home Assistant OS.

The Tibber pulse is used for meter and tariff (Token removed):

```
meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: "***"
  timeout: 120s
  name: tibber_grid

...

tariffs:
  grid:
    type: template
    template: tibber
    token: "..." # Access Token
```

Yesterday I started learning go and downloaded the latest source. I realized that EVCC is using the Query to get basic information 

https://github.com/evcc-io/evcc/blob/832012ef150ead8a5b535bbd28c075d29c5cfd33/meter/tibber/client.go#L44

and using a subscription to continuously update the values.

https://github.com/evcc-io/evcc/blob/832012ef150ead8a5b535bbd28c075d29c5cfd33/meter/tibber-pulse.go#L99

Still learning go and try to further debug it, looking especially at the context. 

For test purposes I increased the timeout to 120s to check if this has an impact on the re-connect, but re-connect time stays approx. at 60s.

### Steps to reproduce

[evcc_log_pulse.txt](https://github.com/user-attachments/files/19920823/evcc_log_pulse.txt)

1.
2.
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: "***"
  timeout: 120s
  name: tibber_grid

...

tariffs:
  grid:
    type: template
    template: tibber
    token: "..." # Access Token
```

### Log details

```text
{"power":61,"powerProduction":0,"lastMeterConsumption":10431.7269,"lastMeterProduction":3081.5861,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/26 07:30:37 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/04/26 07:30:42 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/26 07:30:42 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/26 07:30:42 {"id":"8ac74630-99c9-4799-b4f0-801b4fa4e0b2","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"e1ce067c-c74b-4ba3-b2b1-d05510c8361b"}}} client
[pulse ] TRACE 2025/04/26 07:30:45 {"id":"8ac74630-99c9-4799-b4f0-801b4fa4e0b2","type":"next","payload":{"data":{"liveMeasurement":{"power":40,"powerProduction":0,"lastMeterConsumption":10431.7272,"lastMeterProduction":3081.5861,"currentL1":null,"currentL2":null,"currentL3":null}}}} server

...

[pulse ] TRACE 2025/04/26 07:31:25 {"id":"8ac74630-99c9-4799-b4f0-801b4fa4e0b2","type":"next","payload":{"data":{"liveMeasurement":{"power":68,"powerProduction":0,"lastMeterConsumption":10431.7279,"lastMeterProduction":3081.5861,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/26 07:31:35 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/04/26 07:31:40 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/26 07:31:40 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/26 07:31:40 {"id":"10db274d-6f97-4c69-8407-7afe35e31f1c","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"e1ce067c-c74b-4ba3-b2b1-d05510c8361b"}}} client
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.3

jpvolkmann, 2025-04-26:
Short update after analysing it in Debugger:
- The re-connect is caused by go-graphql-client:
https://github.com/hasura/go-graphql-client/blob/f20a101ec4c9ffd65fbc6f413461d1287300db64/subscription.go#L1126
triggered by:
https://github.com/hasura/go-graphql-client/blob/f20a101ec4c9ffd65fbc6f413461d1287300db64/subscription.go#L459
and
https://github.com/hasura/go-graphql-client/blob/f20a101ec4c9ffd65fbc6f413461d1287300db64/subscription.go#L427

- Received message from Tibber json-API is go-graphql-client.OperationMessage {ID: "", Type: "", Payload: encoding/json.RawMessage len: 0, cap: 0, nil}

@toini any idea about the message from the Tibber-API? I'll further debug later.

 Not directly related to EVCC.

mreschka, 2025-04-26:
Hi guys,

same problem here - I got the same mail from tibber support after ugrading to 203.0 or 1 (dont't know exactly) some days ago. Version 203.3 does not solve the problem but thanks to the Fix in [https://github.com/evcc-io/evcc/pull/20841](url) the user agent now is more descriptive.

With 203.3 I get the same log entries:

`[pulse ] TRACE 2025/04/26 11:11:16 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/04/26 11:11:21 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/26 11:11:22 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/26 11:11:22 {"id":"***","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client`

The grafana guys had a similar discussion in [https://github.com/grafana/loki/issues/6182](url)

In the thread the only try for here would be to increase the ingester_client.remote_timeout to a little higher value?

Thanks,

Markus

jpvolkmann, 2025-04-26:
I think a fix is needed either in Tibber-API or in go-graphql-client to avoid a re-connect in case of empty payload. Just wait for more comments...

andig, 2025-04-26:
@jpvolkmann Klasse Analyse, vielen Dank!

> Received message from Tibber json-API is go-graphql-client.OperationMessage {ID: "", Type: "", Payload: encoding/json.RawMessage len: 0, cap: 0, nil}

Are saying that this is the root cause of the reconnect? If yes- where does this happen in the hasura client? I couldn't spot it in the links above.

If not I still don't understand:
- are there reconnects that should not be happing (i.e. because there is no server issue)
- or are the reconnects warranted but due to some other error happen multiple times instead of once?

andig, 2025-04-26:
@mreschka the `context deadline exceeded` looks like a red herring to me since it only happens when the reconnect is already under way. If there are multiple reconnects (which we suspect as issue) we should focus on the root cause instead.

jpvolkmann, 2025-04-26:
When 

`if err := conn.ReadJSON(&message); err != nil {`  (https://github.com/hasura/go-graphql-client/blob/f20a101ec4c9ffd65fbc6f413461d1287300db64/subscription.go#L427 )

receives the message 

`{ID: "", Type: "", Payload: encoding/json.RawMessage len: 0, cap: 0, nil}`

it will return an error that is handled in 

```
				switch websocket.StatusCode(closeStatus) {
				case websocket.StatusBadGateway, websocket.StatusNoStatusRcvd, websocket.StatusServiceRestart, websocket.StatusTryAgainLater, websocket.StatusMessageTooBig, websocket.StatusInvalidFramePayloadData:
					sc.Log(err, "server", GQL_CONNECTION_ERROR)
					sc.client.errorChan <- errRestartSubscriptionClient

```

The `errRestartSubscriptionClient` causes the restart of the client. 

I'll try to capture some Wireshark logs to see the complete communication.

andig, 2025-04-26:
> it will return an error that is handled in

Ah, I didn't suspect this since that's perfectly valid json. I think we should update https://github.com/hasura/go-graphql-client/issues/164 latest when you can confirm via wireshark.

jpvolkmann, 2025-04-26:
Just realized that the connection is encrypted and not directly readable in Wireshark, don't want to setup MITM scenario...
How can I easily check the (https://github.com/hasura/go-graphql-client/issues/164) ?

andig, 2025-04-26:
Esit locally and create your own build?

jpvolkmann, 2025-04-26:
Just read it too fast, thought a new version was already available. As mentioned before just learned go yesterday evening...

Now I'm debugging into the Websocket package, maybe the JSON message is just a default message that is reported in case of error in the lower layers. Need some more time and will update later

jpvolkmann, 2025-04-26:
Checked connection establishment with Wireshark. (see attached logs).
It seems that EVCC is disconnecting the TCP connection, [FIN, ACK] first send by EVCC.

Error message in lower layers of websocket/read.go is 

*net.OpError {Op: "read", Net: "tcp", Source: net.Addr(*net.TCPAddr) *{IP: net.IP len: 4, cap: 4, [172,17,0,3], Port: 48704, Zone: ""}, Addr: net.Addr(*net.TCPAddr) *{IP: net.IP len: 4, cap: 4, [34,247,241,203], Port: 443, Zone: ""}, Err: error(internal/poll.errNetClosing) {}}

Tibber API seems to change between 34.x.x.x and 52.x.x.x adresses. 172.17.0.3 is the EVCC adress in the docker container.

[tibber_wireshark_log.txt](https://github.com/user-attachments/files/19922541/tibber_wireshark_log.txt)

MayI have to setup my Linux machine again and check without docker...

toini, 2025-04-26:
That empty message is extremely suspicious. Tibber backend isn't really capable of emitting empty payload without even a timestamp. l believe majority of clients (including evcc) would choke on that. Very interesting to see where it originates from if you can set a breakpoint on it and check the stack?

jpvolkmann, 2025-04-26:
> That empty message is extremely suspicious. Tibber backend isn't really capable of emitting empty payload without even a timestamp. l believe majority of clients (including evcc) would choke on [...]

As mentioned before, empty message might be a default return of the lower layer in case of an error... Still learning go and debugging

toini, 2025-04-26:
> maybe the JSON message is just a default message that is reported in case of error in the lower layers.

This sounds plausible.

jpvolkmann, 2025-04-27:
Further update:

I identified two different reasons for the error:

1. Disconnect requested by websocket:
- In the received Websocket header (not the TCP header) the FIN flag is set and the go websocket implementation is closing the connection as expected.
- The first received byte is 0x88 which corresponds to FIN set with opcode "opClose"
- For websocket header see https://noio-ws.readthedocs.io/en/latest/overview_of_websockets.html#the-websocket-connection-lifespan-and-frame-structure
- This correlates to the set FIN flag set in the TCP packet from EVCC to Tibber in the Wireshark log before as reaction to the Websocket close request

2. NetClosing error message
- Error when reading the next byte in https://github.com/golang/go/blob/f9ce1dddc264cb30e68bfedbabf159b32bb6a719/src/bufio/bufio.go#L113
- error(*net.OpError) *{Op: "read", Net: "tcp", Source: net.Addr(*net.TCPAddr) *{IP: net.IP len: 4, cap: 4, [172,17,0,3], Port: 50546, Zone: ""}, Addr: net.Addr(*net.TCPAddr) *{IP: net.IP len: 4, cap: 4, [52,210,58,1], Port: 443, Zone: ""}, Err: error(internal/poll.errNetClosing) {}}

Unfortunately I can not decode the message sent by Tibber API in Wireshark because it's using TLS .

Any suggestions for further debugging? I've no further ideas...

toini, 2025-04-27:
Were you abe to set a breakpoint and see the stack when the empty message is being logged? I don't think it's originating from the network i.e. perhaps it makes sense to debug the application and not the networking with wireshark?

jpvolkmann, 2025-04-27:
I guess it's much easier than expected before... I just ran the example from https://github.com/hasura/go-graphql-client/blob/master/example/tibber/client.go with DEMO token and my credentials.

I finally realised that the pushing rate for the demo is approx. 1 update / s while my real-time date is updated every 3 seconds, sometimes with a gap of 15s. (See attached logs)

It seems that EVCC is using a timeout of 10s which will cause the failure. Let me locate the timeout setting in EVCC and test.

jpvolkmann, 2025-04-27:
Missing logs:

[tibber_client_log.txt](https://github.com/user-attachments/files/19928423/tibber_client_log.txt)

jpvolkmann, 2025-04-27:
Lines 1..175 DEMO token

Lines 178... Real data, gap e.g. between 2025/04/27 09:16:03 and 2025/04/27 09:16:19 

[tibber_client_log.txt](https://github.com/user-attachments/files/19928442/tibber_client_log.txt)

andig, 2025-04-27:
> I identified two different reasons for the error:

Aren't we focusing on the wrong issue here? Can we reflect which path we're actually investigating?

1) If the graphql client uses empty messages internally to signal "closed connection" then that may be as it is. What we're interested in is why or how this would trigger multiple reconnects.

2) Or are we investigating if the client wrongly treats a connection closed which leads to a proper but undesired reconnect?

> Lines 1..175 DEMO token

What demo?

jpvolkmann, 2025-04-27:
Finally figured out that the websocket waiting in EVCC is timing out, just testing a fix...

To separate the possible failures I started with the Tibber subscription example from hasura/go-graphql-client and used the provided DEMO token for the Tibber API and my personal credentials in the example. Both where working fine in the example.

Only difference was that the update rate with real data was slower (approx. one update every 3 seconds with gap of 15 s instead of one update every second with demo data)

Finally it's just a simple timeout setting in EVCC. Let me check my modification and create a pull request

toini, 2025-04-27:
10s is too short of a value for a no-data timeout. The update frequency depends on the actual meter hardware and in some cases may be up to 30s (I can confirm this with the hardware people). To make evcc work properly with all the meters there are out there the timeout should be much higher. Home assistant uses 90s https://github.com/Danielhiversen/pyTibber/blob/7aa240e8a99fa0316e97b88f65946a9aaf5ead39/tibber/websocket_transport.py#L27

toini, 2025-04-27:
> Finally it's just a simple timeout setting in EVCC. Let me check my modification and create a pull request

In the PR could you make the re-subscription be logged as a warning? That's something that should not happen during normal operation so it would be very good to have it clearly flagged in the logs.

jpvolkmann, 2025-04-27:
Just created pull request for increased timeout. 

https://github.com/evcc-io/evcc/pull/20895

For the logging I need more time and it might not be time-critical...

jpvolkmann, 2025-04-27:
Re-connect is actually happening in the go-graphql-client:

https://github.com/hasura/go-graphql-client/blob/4bb0498c485cc0d81b6d530ff8070c1eb1106e98/subscription.go#L1188

To log it into EVCC an error handler could be provided.

As mentioned before, just learned go on Friday evening and need definitely more time to fully understand EVCC.

andig, 2025-04-27:
@toini merged, new nightly in 30min.

NICOLFRA, 2025-04-27:
I found the reconnect after 10sec without data in my log, too.
It happened ever 2 to 3 minutes.
I will test the nightly.

toini, 2025-04-27:
Unfortunately the evcc version version number isnt populated in the module when it's passed to the user agent here https://github.com/evcc-io/evcc/blob/dab056190e0658828c58365e389936f5b7feb645/meter/tibber-pulse.go#L29 (it seems to have a timestamp and a hash but version is always set to v0.0.0?) so I cant easily verify if the problem goes away for people running the very latest version.

Number of affected users should ofc decrease.

NICOLFRA, 2025-04-27:
@toini I am running he nightly since 15:58 and there was no reconnection for me since then;  unlike before.
From my point of view the changes work.

andig, 2025-04-27:
> Unfortunately the evcc version version number isnt populated in the module

Any idea why that is? Something with go build? We should be able though to use server.Version instrad which we stamp on build.

toini, 2025-04-27:
> Any idea why that is? Something with go build? We should be able though to use server.Version instrad which we stamp on build.

Im assuming the build number is not fully passed to the go build yes. I didn't look into it further. I tried to read the version number from server.Version, too but the compiler said no as it lead to some kind of a circular reference.

andig, 2025-04-27:
Yes, absolutely.

andig, 2025-04-27:
It seems we can close here. If anything pops up please comment below.

closed 2025-04-27 by andig

gramss, 2025-05-18:
Just wanted to leave a small mark here:

Tibber is now detecting or guessing the use of old `evcc` instances.

Content of a mail that I received right now:

 > "It appears you are running evcc, that has a known issue. Please make sure you are running the version 0.203.4 or later."

Pretty cool @toini . Lead me straight to upgrading my `evcc` instance.

I initially thought it has something to do with my new intermediate vDSL that does not behave nicely with my Powerline to my wallbox and the tibber pulse (via PLC->Wifi).. (and hoped my new fiber would come before I get blocked.. 😄  )