evcc-io/evcc #20874: expose Meter/PV Data over MQTT

JonasGhost, 2025-04-26:
**Is your feature request related to a problem? Please describe.**
The problem is a not working PV integration in Home Assistant that works in EVCC. Meter data from the Grid and PV is available in EVCC but not in Home Assistant. The meter data is visible in the EVCC UI, but unfortunately, not exposed over MQTT.

**Describe the solution you'd like**
I'd like to have the data of all meters in the API and MQTT for further use in other applications. 

**Describe alternatives you've considered**
Another solution I've considered, activating the Modbus proxy plugin and implementing the Modbus TCP requests in Home Assistant by hand to read the values from the PV Systems. 

**Additional context**
In this case, it is a Huawei PV System, which, for whatever reason, does not work with the Huawei Solar Home Assistant addon. In the log run, it does not make sense that every home automation project implements its own connectors to every external system. From this point of view, it makes sense to implement IO/connectors in a special system that generalises them and exposes them to other platforms over API or MQTT.

SolarPower2024, 2025-04-26:
Why don't you use the evcc integration in Home Assistant? There you have all values from evcc visible in HA, including PV forecast, PV power, grid meter, etc

VolkerK62, 2025-04-26:
All these data are published. Did you configure MQTT in evcc?

JonasGhost, 2025-04-26:
thx for your response. I did configure MQTT and the evcc integration in Home Assistant. 
I did configure the PV and Grid Meter, but no Wallbox.

Sorry for opening the issue, I found the issue in my configuration. Without a Wallbox EVCC did not expose the data. This was resolved by adding the Demo Wallbox.

closed 2025-04-26 by JonasGhost