evcc-io/evcc #20875: UX: NaN on sleeping tab in Browser

RenatusRo, 2025-04-26:
### Describe the bug

really minor issue:

Ich habe einen 24/7 tab im Desktop chrome mit evcc.
Wenn ich zu dem zurueckkomme, steht PV production auf NaN und bleibt auch so
nach dem Oeffnen und Schliessen des forecast  links. 
Alle anderen (W-) Zaehlen werden aber laufend geupdated.
Wenn ich die charger aufklappe, springen noch mehr Zahlen auf NaN.
Mit Shift-Reload verschwinden die NaNs.

[image]

[image]

### Steps to reproduce

1 .s.o.

### Configuration details

```yaml
does not matter
```

### Log details

```text
does not matter
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

203.3 und vorherige

andig, 2025-04-26:
Was sagt /api/state?

RenatusRo, 2025-04-26:
hab', den shift-reolad schon hinter mir - folgt beim naechsten NaN.

naltatis, 2025-04-26:
[image]
💚😂

naltatis, 2025-04-26:
@RenatusRo lass mal bitte die Browser Devtools Konsole auf und schau ob man da einen Fehler sieht wenn der Effekt eintritt. Hab gerade versucht den NaN Zustand mit invaliden Werten selbst zu produzieren. Leider ohne Erfolg.
Möglich, dass da in der Nummernanimations-Lib [countUp.js](https://github.com/inorganik/CountUp.js)  was schief geht. Das sollte in in der Konsole auftauchen.

RenatusRo, 2025-04-27:
UX state
[image]
console
[image]
console
[image]
/api/state
[nan-6.txt](https://github.com/user-attachments/files/19928367/nan-6.txt)

RenatusRo, 2025-04-27:
und nach back von /api/state ist alles wieder gut

StevieC121176, 2025-04-27:
> ### Describe the bug
> 
> really minor issue:
> 
> Ich habe einen 24/7 tab im Desktop chrome mit evcc. 

Ich habe heute das gleiche bei mir beobachtet, allerdings mit Safari 18.4 . Nach einem reload war alles wieder normal.

naltatis, 2025-04-27:
@RenatusRo schade, keine Hinweise in der Konsole. Der Hinweis mit dem Scrollhandler ist unrelated.

userwithoutpassword, 2025-04-29:
Kann es mit Firefox auch bestätigen

RenatusRo, 2025-04-30:
Jetzt kocht der Traktor: meistens steht da 100%, aber ab und zu 100°C.
ich dachte erst, dass das mit dem Temperatur-Update vom Boiler  passiert, aber so einfach isses nicht.
[image]

RenatusRo, 2025-04-30:
ich hatte den Eindruck, dass uns /api/state
hier https://github.com/evcc-io/evcc/issues/20875#issuecomment-2833259318
nicht entscheidend weitergeholfen hat?

und das log jetzt ueberrascht auch nicht:  soc hat immer %:

```
[lp-1  ] DEBUG 2025/04/30 09:59:41 charger status: C
[lp-1  ] DEBUG 2025/04/30 09:59:41 soc estimated: 100.00% (vehicle: 100.00%)
[lp-1  ] DEBUG 2025/04/30 09:59:41 vehicle soc: 100.00%

[lp-1  ] DEBUG 2025/04/30 09:59:41 vehicle range: 152km
[lp-1  ] DEBUG 2025/04/30 09:59:41 pv charge current: 18.1A = 2A + 16.1A (-3692W @ 1p)
[circuit-main] TRACE 2025/04/30 09:59:41 validate power: -1118.8W + (0W -> 4152.2W) <= 40000W ok
[lp-1  ] DEBUG 2025/04/30 09:59:41 set charge current limit: 18.1A
[site  ] DEBUG 2025/04/30 09:59:41 solar forecast: accumulated 116.315kWh, produced 121.966kWh, scale 1.049
[site  ] DEBUG 2025/04/30 09:59:44 ----
[lp-3  ] DEBUG 2025/04/30 09:59:44 charge power: 2001W
[lp-4  ] DEBUG 2025/04/30 09:59:44 charge power: 571W
[lp-4  ] DEBUG 2025/04/30 09:59:44 charge currents: [2.45 0 0]A
[lp-6  ] DEBUG 2025/04/30 09:59:44 charge power: 0W
[lp-1  ] DEBUG 2025/04/30 09:59:44 charge power: 0W
[lp-7  ] DEBUG 2025/04/30 09:59:44 charge power: 0W
[lp-2  ] DEBUG 2025/04/30 09:59:44 charge power: 0W
[lp-5  ] DEBUG 2025/04/30 09:59:44 charge power: 0W
[lp-1  ] DEBUG 2025/04/30 09:59:44 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/04/30 09:59:44 power: -1213W
[circuit-main] DEBUG 2025/04/30 09:59:44 current: 3.81A
[site  ] DEBUG 2025/04/30 09:59:44 grid power: -1213W
[site  ] DEBUG 2025/04/30 09:59:44 pv 1 power: 5916W
[site  ] DEBUG 2025/04/30 09:59:44 grid powers: [613 -523 -1315]W
[site  ] DEBUG 2025/04/30 09:59:44 grid currents: [3.81 -3.09 -5.58]A
[site  ] DEBUG 2025/04/30 09:59:44 site power: -1214W
[lp-2  ] DEBUG 2025/04/30 09:59:44 charge total import: 1.575kWh
[lp-2  ] DEBUG 2025/04/30 09:59:44 charger status: B
[lp-2  ] DEBUG 2025/04/30 09:59:44 pv charge current: min 9A > 5.28A (-1214W @ 1p, battery: false)
[circuit-main] TRACE 2025/04/30 09:59:44 validate power: -1213W + (0W -> 2070W) <= 40000W ok
[site  ] DEBUG 2025/04/30 09:59:44 solar forecast: accumulated 116.321kWh, produced 121.972kWh, scale 1.049
[site  ] DEBUG 2025/04/30 09:59:48 ----
[lp-3  ] DEBUG 2025/04/30 09:59:48 charge power: 2028W
[lp-7  ] DEBUG 2025/04/30 09:59:48 charge power: 0W
[lp-4  ] DEBUG 2025/04/30 09:59:48 charge power: 571W
[lp-4  ] DEBUG 2025/04/30 09:59:48 charge currents: [2.45 0 0]A
[lp-2  ] DEBUG 2025/04/30 09:59:48 charge power: 0W
[lp-6  ] DEBUG 2025/04/30 09:59:48 charge power: 0W
[lp-5  ] DEBUG 2025/04/30 09:59:48 charge power: 0W
[lp-1  ] DEBUG 2025/04/30 09:59:48 charge power: 0W
[lp-1  ] DEBUG 2025/04/30 09:59:48 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/04/30 09:59:48 power: -1310.2W
[circuit-main] DEBUG 2025/04/30 09:59:48 current: 3.78A
[main  ] DEBUG 2025/04/30 09:59:48 lp Boiler_ESP at prio 50 gets additional 571W from fietskot_Rood at prio 21, total 571W
[site  ] DEBUG 2025/04/30 09:59:48 grid power: -1310W
[site  ] DEBUG 2025/04/30 09:59:48 pv 1 power: 6005W
[site  ] DEBUG 2025/04/30 09:59:48 grid powers: [597 -534 -1371]W
[site  ] DEBUG 2025/04/30 09:59:48 grid currents: [3.78 -3.17 -5.83]A
[site  ] DEBUG 2025/04/30 09:59:48 site power: -1882W (including 571W prioritized power)

[lp-3  ] DEBUG 2025/04/30 09:59:48 charger status: C
[lp-3  ] DEBUG 2025/04/30 09:59:48 vehicle soc: 41.50%

```

andig, 2025-04-30:
> ich hatte den Eindruck, dass uns /api/state nicht entscheidend weitergeholfen hat?

Ein Screenshot tut es jedenfalls auch nciht. In Summe besteht der Verdacht, dass irgendwie im UI ein NaN entsteht, warum ist unklar.

RenatusRo, 2025-04-30:
Im letzten screenshot ging es nicht primair um die NaN (das steht da auch), sondern dass  der SoC vom 
vehicle Zoe intermettierend mit 100°C anstelle 100%  visualisiert wird.
Gut, das ist bezueglich NaN off topic, aber vielleicht auch nur ein weiteres Symptom desselben Problems im frontend?

andig, 2025-04-30:
Kann sein. Hast Du in Log und /api/state geschaut?

naltatis, 2025-04-30:
Nochmal für mich zusammengefasst. Wir habe 5 Stellen an denen das bislang auftaucht:
- In
- Produktion Vorhersage
- Produktion Leistung  (nur bei gesamter Leistung, nicht bei Unterpunkten)
- Out
- Ladepunkte (nur bei gesamter Leistung, nicht bei Unterpunkten)

Auf den Screenshots (@andig ich find die bei diesem Thema in der Tat hilfreich), sieht man, dass nicht immer alle Stellen betroffen sind, richtig?

andig, 2025-04-30:
> ich find die bei diesem Thema in der Tat hilfreich

Die zeigen, _dass_ es ein Problem gibt. Leider nicht woher es kommen könnte. /api/state hingegen hilft um Ursachen auszuschließen.

naltatis, 2025-04-30:
Ja, das hilft das Fehlerbild besser zu verstehen. Die beschriebenen Effekte sind ja durchaus unterschiedlich. Siehe vorheriger Kommentar.

RenatusRo, 2025-04-30:
> Nochmal für mich zusammengefasst. Wir habe 5 Stellen an denen das bislang auftaucht:
>
> * In
> * Produktion Vorhersage
> * Produktion Leistung  (nur bei gesamter Leistung, nicht bei Unterpunkten)
> * Out
> * [...]

Ja, alles korrekt zusammengefasst.
Erstmalig ist das mit dem solar production forcast aufgetaucht. 
Und dann auch nur dort fuer die remaining kWh today. 
Dass andere Stellen betroffen sind, habe ich vorige Woche  zum ersten Mal gesehen (203.3).
(Die 100°C  SoC fuer ein Auto hatten heute Premiere.)

Beim naechsten Mal gibt's den vollen Datensatz, das dauert aber > 12h.

naltatis, 2025-05-03:
Hier gibts einen möglichen Fix. Das adressiert nur die Anzeige. Meine Vermutung ist, dass hier irgend etwas auf Datenebenen schief läuft. Aber beobachtet bitte mal ob sich das Verhalten mit den nächsten Nightly verändert.

RenatusRo, 2025-05-05:
Ich hatte seit Mittwoch einen cronjob laufen, der alle 5 min
`curl "http://10.7.30.174:7070/api/state" >> state.txt`
macht (auf dem evcc device, also nicht auf dem browser host.)
Seitdem kein NaN mehr gesehen. 
Schroedingers Katze.

RenatusRo, 2025-05-05:
here we go:
(noch ohne den format fix, zwecks Reproduzierbarkeit)

**08:45**
state:
[state_0845.json](https://github.com/user-attachments/files/20049430/state_0845.json)

Rund 10:30 sah alles immmer noch normal aus.

**13:13**
[image]
state:

[state_1313.json](https://github.com/user-attachments/files/20049448/state_1313.json)

**13:14, jetzt auch die charger summe**

[image]
state:

[state_1314.json](https://github.com/user-attachments/files/20049450/state_1314.json)

**13:25, weiter, trotz anderer zahl charger**

[image]
state:

[state_1325.json](https://github.com/user-attachments/files/20049454/state_1325.json)

**13:26, weiter, trotz anderer zahl charger**
[image]
state:

[state_1326.json](https://github.com/user-attachments/files/20049461/state_1326.json)

**17:07, jetzt auch noch solar forecast kaputt**

[image]
state:
[state_1707.json](https://github.com/user-attachments/files/20049463/state_1707.json)

log:
[evcc-250505-2.zip](https://github.com/user-attachments/files/20049468/evcc-250505-2.zip)

closed 2025-05-06 by andig

RenatusRo, 2025-05-06:
cool. 
Thx!

RenatusRo, 2025-05-08:
@naltatis ja - hatte das nach dem PR merge kompiliert.  Wollte aber noch auf ein 2tes Mal warten, bevor ich reopen.