evcc-io/evcc #20878: Skoda Citigo iev

HerrAlbers, 2025-04-26:
### Describe the bug

Der SOC des Citigo wird schon seit Februar nicht mehr abgerufen, in OpenWB als Fahrzeug neu angelegt, Token erneuert, Sprache bei Skoda verändert, immer die gleiche Fehlermeldung, s.u. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
?
```

### Log details

```text
2025-04-26 12:36:20,874 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev11} - Vehicle Instance <class 'modules.vehicles.evcc.config.EVCCVehicleSoc'>
2025-04-26 12:36:20,877 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev11} - Calculated SoC-State CalculatedSocState(imported_start=None, manual_soc=None, soc_start=76.0)
2025-04-26 12:36:20,904 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev11} - Vehicle Update Data VehicleUpdateData(plug_state=False, charge_state=False, imported=None, battery_capacity=58000, efficiency=95, soc_from_cp=None, timestamp_soc_from_cp=None, soc_timestamp=1740369729.475205)
2025-04-26 12:36:20,910 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev11} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=600, request_interval_not_charging=7200, request_only_plugged=False)
2025-04-26 12:36:20,915 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev11} - Update Komponente ['EVCC']
2025-04-26 12:36:20,917 - {modules.vehicles.evcc.api:66} - {DEBUG:fetch soc_ev11} - Fetching EVCC SOC
2025-04-26 12:36:21,548 - {modules.vehicles.evcc.api:53} - {DEBUG:fetch soc_ev11} - Vehicle client received: 5691
2025-04-26 12:36:21,553 - {modules.vehicles.evcc.api:27} - {DEBUG:fetch soc_ev11} - saving  vehicle_id: 5691
2025-04-26 12:36:21,602 - {modules.common.fault_state:49} - {ERROR:fetch soc_ev11} - EVCC: FaultState FaultStateLevel.ERROR, FaultStr <class 'UnboundLocalError'> ("local variable 'vehicle_to_fetch' referenced before assignment",), Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 114, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 81, in fetch_soc
    log.debug("Fetching SoC for vehicle id: " + str(vehicle_to_fetch))  # fetch SoC
UnboundLocalError: local variable 'vehicle_to_fetch' referenced before assignment

2025-04-26 12:36:21,784 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev11} - Vehicle Instance <class 'modules.vehicles.evcc.config.EVCCVehicleSoc'>
2025-04-26 12:36:21,788 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev11} - Calculated SoC-State CalculatedSocState(imported_start=None, manual_soc=None, soc_start=76.0)
2025-04-26 12:36:21,792 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev11} - Vehicle Update Data VehicleUpdateData(plug_state=True, charge_state=False, imported=11112655.27, battery_capacity=58000, efficiency=95, soc_from_cp=None, timestamp_soc_from_cp=None, soc_timestamp=1740369729.475205)
2025-04-26 12:36:21,797 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev11} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=600, request_interval_not_charging=7200, request_only_plugged=False)
2025-04-26 12:36:21,800 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev11} - Update Komponente ['EVCC']
2025-04-26 12:36:21,803 - {modules.vehicles.evcc.api:66} - {DEBUG:fetch soc_ev11} - Fetching EVCC SOC
2025-04-26 12:36:21,812 - {modules.vehicles.evcc.api:79} - {DEBUG:fetch soc_ev11} - Vehicle id found in config: 5691
2025-04-26 12:36:21,816 - {modules.vehicles.evcc.api:81} - {DEBUG:fetch soc_ev11} - Fetching SoC for vehicle id: 5691
2025-04-26 12:36:21,956 - {modules.common.fault_state:49} - {ERROR:fetch soc_ev11} - EVCC: FaultState FaultStateLevel.ERROR, FaultStr <class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "invalid vehicle type: skoda"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"invalid vehicle type: skoda", grpc_status:2, created_time:"2025-04-26T12:36:21.938447045+02:00"}"
>,), Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 86, in fetch_soc
    response = stub.SoC(
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1160, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1003, in _end_unary_response_blocking
    raise _InactiveRpcError(state)  # pytype: disable=not-instantiable
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "invalid vehicle type: skoda"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"invalid vehicle type: skoda", grpc_status:2, created_time:"2025-04-26T12:36:21.938447045+02:00"}"
>
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-26:
Das ist ein Fehler im Python Code, nicht in evcc 🥳

HerrAlbers, 2025-04-26:
... bedeutet dass es bei den Leuten von OpenWB angepasst werden muss?

andig, 2025-04-27:
Für den Python Teil ja. Die Ursache sehe ich aber hier:

> details = "invalid vehicle type: skoda"

Bitte mal `skoda` auf `enyaq` ändern- das alte API gibts nicht mehr.

andig, 2025-04-27:
Depends on: https://github.com/snaptec/openWB/issues/2868

closed 2025-04-27 by andig