evcc-io/evcc #20910: Since update 0.203.0 my Solax inerters  show 0W power

Rocke2511, 2025-04-27:
### Describe the bug

1. Update to 0.203.0 or later results in showing  0W power for my two Solax inverters.
2. Downgrade to 0.202.1. solves the issue.

### Steps to reproduce

1. Update to 0.203.0 or later results in showing  0W power for my two Solax inverters.
2. Downgrade to 0.202.1. solves the issue.

### Configuration details

```yaml
Config as following:

- name: Solax-Garage
  type: template
  template: solax-inverter-cloud 
  usage: pv  
  tokenid: 'XXX'  
  serial: ***
  
- name: Solax-Gartenhaus
  type: template
  template: solax-inverter-cloud 
  usage: pv  
  tokenid: 'XXX'  
  serial: ***
```

### Log details

```text
n/a
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-27:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

Rocke2511, 2025-04-27:
It does not prouduce any error. It shows that the connection with the inverters is fine, but with 0W power.

Rocke2511, 2025-04-27:
[image]

Rocke2511, 2025-04-27:
When i downgrade to 0.202.1it is sowing the correct value.

andig, 2025-04-29:
There we no changes since 202.1. In any case, we need 

    evcc meter --log trace

of old and new version. Without logs there's nothing to do.

closed 2025-04-29 by andig