evcc-io/evcc #20911: Since update 0.203.0 my Solax inerters show 0W power

Rocke2511, 2025-04-27:
### Describe the bug

Since update 0.203.0 my Solax inerters show 0W power

### Steps to reproduce

1. Update to 0.203.0 or later results in showing 0W power for my two Solax inverters.

2. Downgrade to 0.202.1. solves the issue.

### Configuration details

```yaml
Config as following:

- name: Solax-Garage
  type: template
  template: solax-inverter-cloud 
  usage: pv  
  tokenid: '20250427165731090648433'  
  serial: SRNUPDEXSJ
  
- name: Solax-Gartenhaus
  type: template
  template: solax-inverter-cloud 
  usage: pv  
  tokenid: '20250427165731090648433'  
  serial: SWWXHZZDMB
```

### Log details

```text
Trace from V0.202.1
[Solax-Gartenhaus] TRACE 2025/04/27 20:41:15 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20250427165731090648433&sn=SWWXHZZDMB
[Solax-Garage] TRACE 2025/04/27 20:41:15 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20250427165731090648433&sn=SRNUPDEXSJ
[Solax-Garage] TRACE 2025/04/27 20:41:15 {"success":true,"exception":"Query success!","result":{"inverterSN":"XM3A33I7484016","sn":"SRNUPDEXSJ","acpower":0.0,"yieldtoday":10.4,"yieldtotal":3315.4,"feedinpower":0.0,"feedinenergy":0.0,"consumeenergy":0.0,"feedinpowerM2":null,"soc":null,"peps1":null,"peps2":null,"peps3":null,"inverterType":"4","inverterStatus":"102","uploadTime":"2025-04-27 20:40:25","batPower":null,"powerdc1":0.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null,"batStatus":null,"utcDateTime":"2025-04-27T12:40:25.438Z"},"code":0}
[Solax-Gartenhaus] TRACE 2025/04/27 20:41:15 {"success":true,"exception":"Query success!","result":{"inverterSN":"XMT072G6001156","sn":"SWWXHZZDMB","acpower":0.0,"yieldtoday":1.4,"yieldtotal":1120.8,"feedinpower":0.0,"feedinenergy":0.0,"consumeenergy":0.0,"feedinpowerM2":null,"soc":null,"peps1":null,"peps2":null,"peps3":null,"inverterType":"4","inverterStatus":"100","uploadTime":"2025-04-27 20:05:48","batPower":null,"powerdc1":0.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null,"batStatus":null,"utcDateTime":"2025-04-27T12:05:48.984Z"},"code":0}
[Solax-Garage] TRACE 2025/04/27 20:43:50 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20250427165731090648433&sn=SRNUPDEXSJ
[Solax-Gartenhaus] TRACE 2025/04/27 20:43:50 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20250427165731090648433&sn=SWWXHZZDMB
[Solax-Gartenhaus] TRACE 2025/04/27 20:43:50 {"success":true,"exception":"Query success!","result":{"inverterSN":"XMT072G6001156","sn":"SWWXHZZDMB","acpower":0.0,"yieldtoday":1.4,"yieldtotal":1120.8,"feedinpower":0.0,"feedinenergy":0.0,"consumeenergy":0.0,"feedinpowerM2":null,"soc":null,"peps1":null,"peps2":null,"peps3":null,"inverterType":"4","inverterStatus":"100","uploadTime":"2025-04-27 20:05:48","batPower":null,"powerdc1":0.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null,"batStatus":null,"utcDateTime":"2025-04-27T12:05:48.984Z"},"code":0}
[Solax-Garage] TRACE 2025/04/27 20:43:50 {"success":true,"exception":"Query success!","result":{"inverterSN":"XM3A33I7484016","sn":"SRNUPDEXSJ","acpower":0.0,"yieldtoday":10.4,"yieldtotal":3315.4,"feedinpower":0.0,"feedinenergy":0.0,"consumeenergy":0.0,"feedinpowerM2":null,"soc":null,"peps1":null,"peps2":null,"peps3":null,"inverterType":"4","inverterStatus":"102","uploadTime":"2025-04-27 20:40:25","batPower":null,"powerdc1":0.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null,"batStatus":null,"utcDateTime":"2025-04-27T12:40:25.438Z"},"code":0}

Trace from V0.203.3
[Solax-Garage] TRACE 2025/04/27 20:46:09 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=%2720250427165731090648433%27&sn=SRNUPDEXSJ
[Solax-Gartenhaus] TRACE 2025/04/27 20:46:09 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=%2720250427165731090648433%27&sn=SWWXHZZDMB
[Solax-Garage] TRACE 2025/04/27 20:46:09 {"exception":"token invalid!","code":103,"tokenId":"'20250427165731090648433'","success":false}
[Solax-Gartenhaus] TRACE 2025/04/27 20:46:09 {"exception":"token invalid!","code":103,"tokenId":"'20250427165731090648433'","success":false}
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.203.3

mfuchs1984, 2025-04-27:
`tokenId=20250427165731090648433` alt vs.
`tokenId=%2720250427165731090648433%27`

Da geht was bei der Verarbeitung des Strings aus der yaml schief.

Geht's ohne die quotes ''? Könnte von der Art des fehlers her mit https://github.com/evcc-io/evcc/pull/20376 zusammenhängen.

Rocke2511, 2025-04-27:
Das war auch meine erste Idee, aber wenn ich die ' weglasse sieht es ähnlich aus:

[Solax-Gartenhaus] TRACE 2025/04/27 22:00:22 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.025042716573109e%2B22&sn=SWWXHZZDMB
[Solax-Garage] TRACE 2025/04/27 22:00:22 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.025042716573109e%2B22&sn=SRNUPDEXSJ
[Solax-Gartenhaus] TRACE 2025/04/27 22:00:22 {"exception":"token invalid!","code":103,"tokenId":"2.025042716573109e+22","success":false}
[Solax-Garage] TRACE 2025/04/27 22:00:22 {"exception":"token invalid!","code":103,"tokenId":"2.025042716573109e+22","success":false}

mfuchs1984, 2025-04-27:
Oh, dann wird es in eine Zahl mit exponent verwandelt :D

andig, 2025-04-29:
Closed in b6f5293d9693b7a2e5a971eb84bb1c0913bfafd5

slagon15, 2025-05-06:
Hi!
I think it is still broken in 0.203.5.
My trace still shows the "%27":
[battery3] TRACE 2025/05/06 23:03:41 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=%27xxxxxx12%27&sn=xxxxxx

andig, 2025-05-07:
It works using your config above (you've exposed your credentials):

```
[main  ] INFO 2025/05/07 20:25:19 using config file: solax.yaml
[Solax-Garage] TRACE 2025/05/07 20:25:19 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20250427165731090648433&sn=SRNUPDEXSJ
[Solax-Garage] TRACE 2025/05/07 20:25:20 {"success":true,"exception":"Query success!","result":{"inverterSN":"XM3A33I7484016","sn":"SRNUPDEXSJ","acpower":0.0,"yieldtoday":8.7,"yieldtotal":3397.9,"feedinpower":0.0,"feedinenergy":0.0,"consumeenergy":0.0,"feedinpowerM2":null,"soc":null,"peps1":null,"peps2":null,"peps3":null,"inverterType":"4","inverterStatus":"102","uploadTime":"2025-05-07 20:25:05","batPower":null,"powerdc1":8.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null,"batStatus":null,"utcDateTime":"2025-05-07T12:25:05.936Z"},"code":0}
Power: 0W
```

slagon15, 2025-05-07:
@andig Thanks for your quick response.
After a lot of trys i finally found the problem:
Only the solax-inverter-cloud.yaml is fixed and working.
Could you please also update solax-hybrid-cloud.yaml (and there for all 3 usage types)
That would be nice!

andig, 2025-05-07:
> Only the solax-inverter-cloud.yaml is fixed and working.

That's what you've shown above.

> Could you please also update solax-hybrid-cloud.yaml (and there for all 3 usage types)

Done.

Champ40, 2025-06-09:
Hallo andig,

die Solax-Hybrid-Cloud.yaml scheint immer noch defekt zu sein. Gebe ich den Token und Seriennummer bei Solax PV Wechselrichter (Cloud) ein ist die Prüfung erfolgreich (1.tes Bild). Als Solax Hybrid Wechselrichter (Cloud) wird weiterhin der Syntax-Fehler (2.tes Bild) bei der Prüfung angezeigt. Ich kann daher die PV-Anlage nicht ins evcc einbinden. Kannst du das bitte noch Mal prüfen?

Vielen Dank.

[image]

[image]

andig, 2025-06-09:
Es braucht bitte ein

    evcc meter --log trace

andig, 2025-06-10:
Im Zweifel neue Installation auf einem PC

Champ40, 2025-06-22:
Hallo andig,

anbei das benötigte Trace

site  ] INFO 2025/06/22 17:15:46 site config:
[site  ] INFO 2025/06/22 17:15:46   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/06/22 17:15:46     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/06/22 17:15:46     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/06/22 17:15:46     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/06/22 17:15:46   vehicles:
[site  ] INFO 2025/06/22 17:15:46     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/06/22 17:15:46   tariffs:
[site  ] INFO 2025/06/22 17:15:46     grid:      ✗
[site  ] INFO 2025/06/22 17:15:46     feed-in:   ✗
[site  ] INFO 2025/06/22 17:15:46     co2:       ✗
[site  ] INFO 2025/06/22 17:15:46     solar:     ✓
[lp-1  ] INFO 2025/06/22 17:15:46 loadpoint 1:
[lp-1  ] INFO 2025/06/22 17:15:46   mode:        off
[lp-1  ] INFO 2025/06/22 17:15:46   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/06/22 17:15:46   meters:      charge ✓
[lp-1  ] INFO 2025/06/22 17:15:46     charge:    power ✓ energy ✓ currents ✓
[main  ] TRACE 2025/06/22 17:15:46 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[db    ] TRACE 2025/06/22 17:15:46 SELECT * FROM `configs` WHERE `configs`.`id` = 5 AND `configs`.`class` = 5 ORDER BY `configs`.`id` LIMIT 1 1 <nil>
[db    ] TRACE 2025/06/22 17:15:46 UPDATE `configs` SET `class`=5,`type`="",`title`="",`icon`="",`product`="",`value`="{""charger"":""db:4"",""circuit"":"""",""defaultMode"":""minpv"",""meter"":"""",""phasesConfigured"":0,""soc"":{""poll"":{""mode"":""charging"",""interval"":3600000000000},""estimate"":true},""thresholds"":{""enable"":{""delay"":60000000000,""threshold"":0},""disable"":{""delay"":180000000000,""threshold"":0}},""title"":""Garage"",""vehicle"":""""}" WHERE `id` = 5 1 <nil>
[lp-1  ] DEBUG 2025/06/22 17:15:46 phase timer inactive
[lp-1  ] DEBUG 2025/06/22 17:15:46 pv timer inactive
[em2go ] TRACE 2025/06/22 17:15:46 modbus: send 00 05 00 00 00 06 ff 03 00 5f 00 01
[main  ] TRACE 2025/06/22 17:15:46 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/226090048","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/226090048/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/226090048/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.204.3","id":226090048,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"node_id":"RE_kwDODX4bUs4NedxA","tag_name":"0.204.3","target_commitish":"master","name":"0.204.3","draft":false,"prerelease":false,"created_at":"2025-06-18T07:25:05Z","published_at":"2025-06-18T08:31:58Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/265019948","id":265019948,"node_id":"RA_kwDODX4bUs4Py-Is","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":754,"digest":"sha256:7b659c95b7350a49f384ad3dc63a3fea06dacc8e8c687069747411ca83c4a100","download_count":5,"created_at":"2025-06-18T08:45:59Z","updated_at":"2025-06-18T08:45:59Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.204.3/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/265019940","id":265019940,"node_id":"RA_kwDODX4bUs4Py-Ik","name":"evcc_0.204.3_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gi [... cut ...]
[db    ] TRACE 2025/06/22 17:16:45 INSERT INTO `settings` (`key`,`value`) VALUES ("plant","1121d3bcea6f04dfdcb4cb07939b6e0214ec1f8f31be7c8342e3e33916394864"),("adminPassword","$2a$10$uBFq1/jvK/5NgF7RWWySgO6ORBQjGL5CGpPDZ6WK6g/gamQ9eQ.kG"),("jwtSecretKey","5e7b3c73002d329068427392d3df596505aa49b9f700dbd7058b690a946c6092"),("interval","30000000000"),("pvMeters","db:1"),("batteryMeters","db:2"),("gridMeter","db:3"),("solarAccYield","{""db:1"":{""accumulated"":0}}") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 8 <nil>
[httpd ] TRACE 2025/06/22 17:16:54 GET /api/system/log
[httpd ] TRACE 2025/06/22 17:16:54 GET /api/system/log

Champ40, 2025-06-23:
Hallo andig,

anbei das benötigte Trace

[em2go ] TRACE 2025/06/22 17:15:46 modbus: send 00 05 00 00 00 06 ff 03 00 5f 00 01
[em2go ] TRACE 2025/06/22 17:15:46 modbus: recv 00 05 00 00 00 05 ff 03 02 00 02
[em2go ] TRACE 2025/06/22 17:15:47 modbus: send 00 06 00 00 00 06 ff 03 00 0c 00 02
[em2go ] TRACE 2025/06/22 17:15:47 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 00
[em2go ] TRACE 2025/06/22 17:15:47 modbus: send 00 07 00 00 00 06 ff 03 00 06 00 01
[em2go ] TRACE 2025/06/22 17:15:47 modbus: recv 00 07 00 00 00 05 ff 03 02 00 00
[em2go ] TRACE 2025/06/22 17:15:47 modbus: send 00 08 00 00 00 06 ff 03 00 08 00 01
[em2go ] TRACE 2025/06/22 17:15:47 modbus: recv 00 08 00 00 00 05 ff 03 02 00 00
[em2go ] TRACE 2025/06/22 17:15:47 modbus: send 00 09 00 00 00 06 ff 03 00 0a 00 01
[em2go ] TRACE 2025/06/22 17:15:47 modbus: recv 00 09 00 00 00 05 ff 03 02 00 00
**[db:3 ] TRACE 2025/06/22 17:15:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025*******************&sn=SR**********
**[db:2 ] TRACE 2025/06/22 17:15:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025*******************&sn=SR************
**[db:1 ] TRACE 2025/06/22 17:15:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025*******************&sn=SR************
**[db:2 ] TRACE 2025/06/22 17:15:47 {"exception":"token invalid!","code":103,"tokenId":"2025*******************","success":false}**
**[db:3 ] TRACE 2025/06/22 17:15:47 {"exception":"token invalid!","code":103,"tokenId":"2025*******************","success":false}**
**[db:1 ] TRACE 2025/06/22 17:15:47 {"exception":"token invalid!","code":103,"tokenId":"2025*******************","success":false}****
[site ] ERROR 2025/06/22 17:15:54 battery 1 power: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:54 battery 1 soc: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:55
2025-06-22 17:15:47.282006776 +0200 CEST m=+2.355685903 !! 166ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.470997607 +0200 CEST m=+2.544676753 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.506572777 +0200 CEST m=+2.580251905 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.558129528 +0200 CEST m=+2.631808655 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.604590146 +0200 CEST m=+2.678269273 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.6786282 +0200 CEST m=+2.752307346 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.794052798 +0200 CEST m=+2.867731926 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:48.048277887 +0200 CEST m=+3.121957014 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:48.382123065 +0200 CEST m=+3.455802192 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:49.082067851 +0200 CEST m=+4.155746979 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:49.525850619 +0200 CEST m=+4.599529765 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:50.377851261 +0200 CEST m=+5.451530388 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:51.803450189 +0200 CEST m=+6.877129335 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:55.057299292 +0200 CEST m=+10.130978419 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:55 pv 1 power: add[0]: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:55 grid power: strconv.ParseFloat: parsing "": invalid syntax
[site ] DEBUG 2025/06/22 17:16:25 ----
[em2go ] TRACE 2025/06/22 17:16:25 modbus: send 00 0a 00 00 00 06 ff 03 00 0c 00 02
[em2go ] TRACE 2025/06/22 17:16:25 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 00
[em2go ] TRACE 2025/06/22 17:16:25 modbus: send 00 0b 00 00 00 06 ff 03 00 06 00 01
[em2go ] TRACE 2025/06/22 17:16:25 modbus: recv 00 0b 00 00 00 05 ff 03 02 00 00
[em2go ] TRACE 2025/06/22 17:16:25 modbus: send 00 0c 00 00 00 06 ff 03 00 08 00 01
[em2go ] TRACE 2025/06/22 17:16:25 modbus: recv 00 0c 00 00 00 05 ff 03 02 00 00
[em2go ] TRACE 2025/06/22 17:16:25 modbus: send 00 0d 00 00 00 06 ff 03 00 0a 00 01
[em2go ] TRACE 2025/06/22 17:16:25 modbus: recv 00 0d 00 00 00 05 ff 03 02 00 00
[site ] ERROR 2025/06/22 17:16:34 pv 1 power: add[0]: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:16:34 grid power: strconv.ParseFloat: parsing "": invalid syntax

Champ40, 2025-06-23:
Hallo andig,

anbei das benötigte Trace

[db:3 ] TRACE 2025/06/22 17:15:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025*******************&sn=SR********
[db:2 ] TRACE 2025/06/22 17:15:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025*******************&sn=SR**********
[db:1 ] TRACE 2025/06/22 17:15:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025*******************&sn=SR**********
[db:2 ] TRACE 2025/06/22 17:15:47 {"exception":"token invalid!","code":103,"tokenId":"2025*****************","success":false}**
[db:3 ] TRACE 2025/06/22 17:15:47 {"exception":"token invalid!","code":103,"tokenId":"2025*****************","success":false}**
[db:1 ] TRACE 2025/06/22 17:15:47 {"exception":"token invalid!","code":103,"tokenId":"2025*****************","success":false}**
[site ] ERROR 2025/06/22 17:15:54 battery 1 power: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:54 battery 1 soc: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:55
2025-06-22 17:15:47.282006776 +0200 CEST m=+2.355685903 !! 166ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.470997607 +0200 CEST m=+2.544676753 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.506572777 +0200 CEST m=+2.580251905 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.558129528 +0200 CEST m=+2.631808655 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.604590146 +0200 CEST m=+2.678269273 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.6786282 +0200 CEST m=+2.752307346 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:47.794052798 +0200 CEST m=+2.867731926 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:48.048277887 +0200 CEST m=+3.121957014 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:48.382123065 +0200 CEST m=+3.455802192 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:49.082067851 +0200 CEST m=+4.155746979 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:49.525850619 +0200 CEST m=+4.599529765 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:50.377851261 +0200 CEST m=+5.451530388 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:51.803450189 +0200 CEST m=+6.877129335 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
2025-06-22 17:15:55.057299292 +0200 CEST m=+10.130978419 !! 0ms add[0]: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:55 pv 1 power: add[0]: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:15:55 grid power: strconv.ParseFloat: parsing "": invalid syntax
[site ] DEBUG 2025/06/22 17:16:25 ----
[site ] ERROR 2025/06/22 17:16:34 pv 1 power: add[0]: strconv.ParseFloat: parsing "": invalid syntax
[site ] ERROR 2025/06/22 17:16:34 grid power: strconv.ParseFloat: parsing "": invalid syntax

Champ40, 2025-06-23:
Guten Morgen andig,

danke für den Hinweis...wenn man halt von den Sachen keine Ahnung hat :/

anbei noch ein früherer Trace

[db:3  ] TRACE 2025/06/22 16:56:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025************************&sn=SR************
[db:3  ] TRACE 2025/06/22 16:56:47 {"exception":"token invalid!","code":103,"tokenId":**"**2025**************************"**,"success":false}

Ich denke es geht einfach nur um die **""**

Champ40, 2025-06-23:
Hallo andig,wie in den Bildern zuvor gezeigt, habe ich aber keine "" gesetzt. Die werden durchs Programm gesetzt. Danke für deine Mühe.

andig, 2025-06-23:
Dein Token ist falsch und wird dann "richtig" übergeben:

> [db:3 ] TRACE 2025/06/22 16:56:47 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2025************************&sn=SR************

Champ40, 2025-06-23:
Hallo andig,

egal wie ich es in der UI-Config drehe...ob mit "" oder ohne oder mit ' es gibt einfach keine Verbindung zur Solax Cloud. Die Token-ID ist wie gesagt zu 100% korrekt.

Anbei ein aktuelles Trace mit einmal Solax-Hybridwechselrichter (Cloud) (db:11) und mit Solax-PV-Wechselrichter (Cloud) (db:10).

[image]

Anbei die Übersicht aus der Solax Cloud:

[image]

 Eventuell liegt es an der API Adresse?

Wie gesagt vor Version 0.203 liefen die Einstellungen ohne Probleme. Irgendwas muss verändert worden sein.

mfuchs1984, 2025-06-24:
> Wie gesagt vor Version 0.203 liefen die Einstellungen ohne Probleme. 

Dann wäre der nächste Schritt, mit einer alten Version ein trace log zu machen.

closed 2025-06-24 by andig