evcc-io/evcc #20916: Preislimit für's Laden nach Ladevorgang zurücksetzen

TheFreaker86, 2025-04-27:
**Is your feature request related to a problem? Please describe.**
Szenario: man war wegen schlechten Wetters gezwungen das Auto aus dem Stromnetz zu laden und hat dafür ein Preislimit gesetzt. In den darauffolgenden Tagen wird das Wetter besser, und fast immer deshalb der Strompreis aus dem Netz wegen des großen PV-Stromüberschusses niedriger. Nun schließt man das Auto an die Wallbox an um den eigenen PV-Überschuss in den Akku vom Auto zu packen. Doch nun tut EVCC genau das was es soll: im PV-Modus wird das Auto nun nicht nur vom Überschuss sondern auch aus dem Netz geladen weil wegen des jetzt sonnigen Wetters das zuvor eingestellte Preislimit unterschritten und nicht wieder zurückgesetzt wurde. Dabei ist das bestimmt nicht immer erwünscht.

**Describe the solution you'd like**
Hier wäre ein Toggle Switch hilfreich der wenn aktiviert das Preislimit auf "keins" zurücksetzt wenn das Auto den gewünschten SoC erreicht hat oder von der Wallbox getrennt wurde. 

**Describe alternatives you've considered**
Alternative? Naja, selbst dran denken das Preislimit wieder zurückzusetzen. Doch das ist meiner Meinung nach an der Idee von EVCC vorbei, da es ja meines Verständnisses nach bei EVCC um maximale Kosteneinsparungen, maximale Ladeautomatisierung und maximale Ökologie der Elektromobilität geht.

**Additional context**
Add any other context or screenshots about the feature request here.

StefanSchoof, 2025-04-28:
Warum nutzt du für den Zweck nicht den Planner?
https://docs.evcc.io/docs/features/plans

TheFreaker86, 2025-04-28:
@naltatis Mein Usecase ist eher dass ich den Akku vom Auto als Erweiterung des Hausakkus verwende damit ich den gespeicherten Strom auf die eine oder andere Art selbst verbrauchen kann statt ihn für die wenigen Cent zu verkaufen. Deshalb ist das m.M.n. keine einmalige planmäßige Ladung da ich in der Regel nicht vorhabe zu einem bestimmten Zeitpunkt mit einem bestimmten Mindest-SoC loszufahren. Deshalb sehe ich das auch nicht als ein Feature Request für die Ladeplanung sondern das Ausnutzen von Gelegenheiten des Überschusses. Da funkt dann wie oben beschrieben das Mindestpreislimit dazwischen. EVCC tut ja so genau das was es soll (an der Stelle noch einmal danke für die tolle Arbeit!), das führt halt nur unter den oben beschriebenen Umständen zu einem Ergebnis das nicht im Sinne der Ökologie und Ökonomie ist.