evcc-io/evcc #20918: 203.4 Planner: 3P

detheese, 2025-04-28:
Mit 203.4 bekommen wir das Feature, dass man die Ladung zur Abfahrt hin planen kann. Superklasse!
Ich hatte das neulich schon mal angesprochen. Wäre hier ein 3Phasen-Schalter mega aufwändig?

Anwendung: **Tagsüber mit 1-phasig 5-20A** laden an „kleineren“ PV-Anlage damit der Phasenwechsel keine Unterbrechungen bringt.

Beim Planer wird dann „einfach“ (ich weiß nicht ob das so einfach ist) **3Phasen und 11kW** vorgehen und danach wieder in 1-Phasig geschaltet.

(Ich weiß, dass kann mal mit HA machen ... aber als Feature für Minimalanwender wäre das doch nen wirklich sinnvolles Feature.

[image]

andig, 2025-04-28:
Ziel des Planners ist „billigst“ zu laden. 1p ist da eher abträglich bzw. richtiger wäre ja, den Planner das auf Basis PV Forecast abschätzen zu lassen und selbst 1p3p Umschaltung zu betreiben.

detheese, 2025-04-28:
@andig Ja Du hast recht. Aber die Abschätzung wäre wahrscheinlich viel zu aufwändig: GANZ EINFACH: Das wäre nur für die **Nacht** und nur wenn **wenn der LADEPUNKT auf 1-Phasig** steht. (Damit er nachts in des günstigen Stunden das Auto voll bekommt.

Ich habe hier so ein paar Anwendungsfälle im meinem Umfeld.

andig, 2025-04-28:
> Das wäre nur für die Nacht und nur wenn wenn der LADEPUNKT auf 1-Phasig steht.

Das macht es m.E. noch unsinniger- denn dann STEIGEN die Kosten anstatt die günstigsten Slots zu nutzen.

> Ich habe hier so ein paar Anwendungsfälle im meinem Umfeld.

Welche wären das denn?

naltatis, 2025-04-28:
Das eigentliche Problem scheint doch zu sein, dass Phasenmodus "Auto" über Tag aus irgend einem Grund nicht oder nicht gut bei dir funktioniert, oder?

Was ich mir vorstellen kann ist, dass wir die Nutzer-Phaseneinstellung (auto, 1p, 3p) perspektivisch an den Modus binden. Hier gibts mit https://github.com/evcc-io/evcc/discussions/3530 schon eine relativ altes aber immer noch relevantes Thema. Da könnte man dann den Planer mitdenken (bspw. Plan-Slot verhält sich immer wie "Schnell").

andig, 2025-04-28:
Erstmal sollten wir raus finden wozu das ganze

andig, 2025-04-28:
> tagsüber: ladepunkt steht wegen der 3,7kW-Schwelle und dem damit verbundenen dauernden Wechsel zwischen 1- und 3-phasig fest auf 1-phasig und 20A.

Die Häufigkeit des Wechsels kannst du ja einstellen (enable/disable timeouts). Ebenso könntest Du (extern) morgens/abends umschalten. Auch könntest Du mit Hilfe von residualpower verhinden, dass genug Überschuss für 3p "entsteht".

Ich sehe noch nicht warum das unbedingt in den Planner zu integrieren wäre, insbesondere da dieser ja ohnehin bei Automatik auf 3p schaltet.

detheese, 2025-04-28:
@andig 

Ja diese Parameter sind sicher auch funktional ... aber die Ost/West-Analge meiner Schwester bewegt sich immer von 10-16 Uhr in diesem Schaltbereich. Da bringt es Ihr nichts, wenn ich mit ENABLE/DISABLE was probiere.

Des andere fall ist auch eine OST/WEST, die ihre „besten“ Stunden im 3,7-4 kW-Bereich hat. Quasi öfters hier schaltet.

War ja nur so eine Anregung für die „kleine“ Leistungsklasse.

detlefheese, 2025-04-29:
@Schimi1983 … könnte es auch sein. 
Aktuell stelle ich auf 1-phasig 20A … nur dann kommt in der Nacht nicht genug um die Karre „voll“ zu kriegen.

Schimi1983, 2025-04-29:
bei meiner variante, könnte man auf "automatisch wechseln" lassen und dann würde er Nachts wenn es schnell gehen soll von selbst auf 3p wechseln...

naltatis, 2025-04-29:
@andig sehr gerne. Den Issue dazu gibts schon: https://github.com/evcc-io/evcc/issues/14661

closed 2025-05-03 by andig