evcc-io/evcc #20920: Definition of time tariffs should use consistently english names for days

andromedavpn, 2025-04-28:
### Describe the bug

Not really a bug, rather an inconsistency.

For the definition of days in 

> tariffs:
  grid:
    type: fixed
    price: 0.294 # EUR/kWh (default)
    zones:
      - days: Mo-Fr
        hours: 2-5
        price: 0.2 # EUR/kWh
      - days: Sa,So
        price: 0.15 # EUR/kWh

one has to use germand abbrevaitions Sa, So and not the english one Sa, Su.

### Steps to reproduce

1. Create time tarif with
  grid: # price using energy from the grid
  type: fixed
  price: 0.2565 # EUR/kWh
  zones: 
    - days: Mo-Su
      hours: 2-5 # actually 2:15 - 5:00
      price: 0.21 
2. Entry generates error as Su is not a valid day
3.
...

### Configuration details

```yaml
grid: # price using energy from the grid
  type: fixed
  price: 0.2565 # EUR/kWh
  zones: 
    - days: Mo-Su
      hours: 2-5 # actually 2:15 - 5:00
      price: 0.21
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/04/28 10:48:12 evcc 0.203.4
[main  ] INFO 2025/04/28 10:48:12 using config file: /config/evcc.yaml
[lp-2  ] WARN 2025/04/28 10:48:16 poll mode '{always 5m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[main  ] ERROR 2025/04/28 10:48:16 creating tariff grid failed: cannot create tariff type 'fixed': invalid day: su
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2025-04-28:
> one has to use germand abbrevaitions Sa, So and not the english one Sa, Su.

You can use the three char English abbrevaitions:

https://github.com/evcc-io/evcc/blob/5eff3d995a3db329e3610006b6efd32dc593d379/evcc.dist.yaml#L148-L153

andig, 2025-04-28:
Are these less common?

StefanSchoof, 2025-04-28:
A search produce some pages that have only the three letter ones and no mention of the two letter variant.
 - https://www.pristineword.com/abb-days-of-the-week/
 -  https://www.yourdictionary.com/articles/abbreviations-days-months

StefanSchoof, 2025-04-28:
The English wiktionary has for Monday also only the three letter variant: https://en.wiktionary.org/wiki/Monday#Synonyms

andig, 2025-04-28:
Thank you, then lets close since we have a working English option.

StefanSchoof, 2025-04-30:
I just saw that the English docs are using the German short names: 
 https://docs.evcc.io/en/docs/tariffs#time-based-electricity-price

I can create a PR this fix this, but not before Friday.