evcc-io/evcc #20930: Time-based charging plans

asdfklgash, 2025-04-28:
**Describe the solution you'd like**
With my current setup with SolarEdge I've setup multiple time-based charging plans (Mo-Fr, 6.30-7.30 i.e. / different shorter/longer time slots) for "discharging" my battery in the morning to nearly null on sunny days. So I decide in the morning which plan fits best but have to set the max current on the car as the SolarEdge app is not able to set it on the Keba charger. So I would like to be able to set a charging time plan to load between the defined time with the available current from the battery (here: max. 5 kW) and perhaps also to add already available overage from the PV (starts with around 1 kW at 7.00 and grows to 2 kW at 7.20) to discharge battery at maximum to be "free" for the sunny day.

**Describe alternatives you've considered**
I've seen some discussion threads or perhaps also an feature request for an "discharge battery in the morning" feature, that would be the best case but also more complex.

Lako-Unbekannt, 2025-04-29:
Hi,
so basically batterie boost but automatically startet at a certain time?

asdfklgash, 2025-04-29:
Yes. For today, I tried it with the 1p-charger and played around yesterday with the planned charge level to get the desired start time.
But I could get more energy used using the 3p-charger with 6A starting and then going up to fully use the 5 kW from battery and 2.5 kW from PV at around 7.20 :-)

I think it makes sense to have the ability to plan also charging start time with power (6-x A) instead of just planning with charge level to get.
Will try to take a look at the code, but I'm not familiar with Go yet.

andig, 2025-04-29:
It's unclear why this specific logic would belong into evcc instead of home automation?

asdfklgash, 2025-04-29:
Why the need of an additional home automation when you can have it all in one?
Why there is then the possibility to create a plan for specific charge level?

I thought of setting evcc up for my parents and I don't want to make so many updates on different systems when I'm remote.

andig, 2025-04-29:
Seems this would be obe aspect of #17424

andig, 2025-04-30:
Ahh, sorry.

andig, 2025-05-03:
@naltatis ?

naltatis, 2025-05-03:
For me it's still unclear whats desired here. Title says time-based charging, but when I understood correctly the plans goal is not to reach a specific energy level or soc of the vehicle and rather optimize to pull energy out of the home battery for it to have a good starting soc for the next day. So more of a "overall optimization" intention.

closed 2025-05-15 by github-actions[bot]