evcc-io/evcc #20939: Shift Open-Meteo forecast data by 30 minutes

MarkusGH, 2025-04-29:
Looking at the the solar predition data for my location, especially sunrise and sunset, I assume that the Open-Meteo API returns average radiation values for the following hour. It looks like the calculations in evcc assume, that the returned values are the current values for the gibten time, resulting in an offset of 30 minutes.

Example:

For tomorrow I get 487 W @ 6:00 local time.
The sun rises at 6:04, so the correct reading would be 487 W @ 6:30 local time which matches the actual data.

andig, 2025-04-30:
Ich sehe zwei Probleme:

1. es gab schonmal eine Zeitanpassung in https://github.com/evcc-io/evcc/pull/19507. Mir ist nicht klar, ob sich diese Änderungen jetzt widersprechen. /cc @thecem 
2. evcc erwartet (noch) Stundenslots zur ganzen Stunde. D.h. statt den Timestamp zu ändern müssten die Werte entsprechend interpoliert werden.

thecem, 2025-04-30:
Hintergrund: Zeitstempel bei Open-Meteo
Open-Meteo liefert Wetter- und Strahlungsprognosen mit einer stündlichen Auflösung. Die Werte sind dabei immer der abgelaufenen Stunde zugeordnet. Das bedeutet: Wird zwischen 8 und 9 Uhr beispielsweise ein Ertrag von 3 kWh erwartet, wird dieser Wert mit dem Zeitstempel 09:00 gemeldet. Der Wert beschreibt also die Energiemenge, die im Zeitraum 08:00 bis 09:00 erwartet wird.

Problemstellung bei evcc
Das System evcc erwartet die Prognosewerte jedoch am Anfang der Stunde, in der die Leistung erbracht wird. Das heißt: Der Wert für den Zeitraum 08:00–09:00 Uhr muss bei evcc mit dem Zeitstempel 08:00 hinterlegt werden.

Zeitverschiebung (Nicht Zeitzonenverschiebung!)
Um die Werte korrekt zuzuordnen, müssen die von Open-Meteo gelieferten Zeitstempel um eine Stunde (3600 Sekunden) nach vorne verschoben werden. So wird der Wert, der von Open-Meteo für 09:00 geliefert wird (Ertrag von 08:00 bis 09:00), in evcc als Wert für 08:00 eingetragen.

thecem, 2025-04-30:
Sollte evcc in 1/4h Werte rechnen können, wird das template angepasst.

andig, 2025-04-30:
> Open-Meteo gibt stündliche Werte immer für die abgelaufene Stunde an.

Das bedeutet, dass es sich um die Energiemenge für eine durchschnittliche Leistung _zur Mitte_ der Stunde handelt. "Richtig" wäre jetzt, aus Durchschnittswerter für Stunde und Folgestunde den Leistungswert zur _vollen_ Stunde zu berechnen. 

> Sollte evcc in 1/4h Werte rechnen können, wird das template angepasst.

Ja, das müssen wir früher oder später mal angehen. Auch dann würde evcc aber vmtl. (tbc) davon ausgehen, dass wir für jeden 1/4 Slot einen Wert bekommen. Vielleicht ist das aber auch nicht realistisch und wir müssten die Interpolation im Kern einbauen.

thecem, 2025-04-30:
Open-Meteo kann 1h und 15 min Werte.

realcatgirly, 2025-04-30:
Wir haben bereits eine Interpolation, um aus Forecast-Daten minütliche Vorhersagen zu bekommen. Leider kein go playground link, das importieren der module dauert zu lange und timeoutet. 

```go
package main

import (
	"encoding/json"
	"fmt"
	"math"
	"time"

	"gonum.org/v1/gonum/interp"
)

type ForecastData struct {
	Forecast []struct {
		Timestamp    string  `json:"timestamp"`
		PredictedKwh float64 `json:"predicted_kwh"`
	} `json:"forecast"`
}

// InterpolateData performs cubic interpolation on time series data
func InterpolateData(times []time.Time, values []float64, interval time.Duration) ([]time.Time, []float64) {
	// Convert times to float64 for interpolation
	x := make([]float64, len(times))
	for i, t := range times {
		x[i] = float64(t.Unix())
	}

	// Create new time points at specified interval
	start := times[0]
	end := times[len(times)-1]
	newTimes := []time.Time{}
	for t := start; t.Before(end) || t.Equal(end); t = t.Add(interval) {
		newTimes = append(newTimes, t)
	}

	// Convert new times to float64
	newX := make([]float64, len(newTimes))
	for i, t := range newTimes {
		newX[i] = float64(t.Unix())
	}

	// Perform cubic interpolation
	var pc interp.PiecewiseCubic
	dydx := make([]float64, len(values)) // Assuming zero derivatives for simplicity
	pc.FitWithDerivatives(x, values, dydx)
	newValues := make([]float64, len(newX))
	for i, xVal := range newX {
		newValues[i] = pc.Predict(xVal)
	}

	return newTimes, newValues
}

// ClipNegative sets all negative values to 0
func ClipNegative(values []float64) []float64 {
	result := make([]float64, len(values))
	for i, v := range values {
		if v < 0 {
			result[i] = 0
		} else {
			result[i] = v
		}
	}
	return result
}

// ProcessPowerData combines all operations
func ProcessPowerData(times []time.Time, values []float64) ([]time.Time, []float64) {
[... 228 lines cut ...]
	values := make([]float64, len(data.Forecast))

	for i, entry := range data.Forecast {
		t, err := time.Parse(time.RFC3339, entry.Timestamp)
		if err != nil {
			fmt.Printf("Error parsing time: %v\n", err)
			return
		}
		times[i] = t
		values[i] = entry.PredictedKwh
	}

	// Process the data
	newTimes, newValues := ProcessPowerData(times, values)

	// Print the results
	fmt.Println("Time,Value")
	energy := 0.0
	for i := range newTimes {
		//fmt.Printf("%s,%f\n", newTimes[i].Format(time.RFC3339), newValues[i])
		energy += newValues[i] / 60
	}
	fmt.Printf("Energy: %f\n", energy)
}
```

thecem, 2025-04-30:
> Das bedeutet, dass es sich um die Energiemenge für eine durchschnittliche Leistung zur Mitte der Stunde handelt. "Richtig" wäre jetzt, aus Durchschnittswerter für Stunde und Folgestunde den Leistungswert zur vollen Stunde zu berechnen.

Das Template übergibt keine Leistung (kVA) sondern die Energiemenge pro Stunde (kWh).

@realcatgirly Danke!

MarkusGH, 2025-04-30:
> Das mag sein, ist aber nicht was evcc erwartet. Wir brauchen Leistung zum Zeitpunkt.

Wie ist das zu verstehen? Momentanwert der Leistung zum Zeitpunkt t, oder Mittelwert der Leistung im Intervall [t - 30 min , t + 30 min]?

andig, 2025-04-30:
Zum Zeitpunkt t.

thecem, 2025-04-30:
wie auch immer, steht das irgendwo? 

Würde dann das template neu bauen.

global_tilted_irradiance_instant ist dann ohne Zeitkomponente unser Freund.

MarkusGH, 2025-04-30:
> Also die Leistung in kWp (kVA) zur Mitte der Stunde, soll das template liefern?

Nach meinem Verständnis wäre der Momentanwert der Leistung jeweils zur vollen Stunde zu liefern.

andig, 2025-04-30:
Genau- `start` timestamp.

andig, 2025-04-30:
> wie auch immer, steht das irgendwo?

Ich fürchte nicht. Es gab dazu auf Slack eine längere Diskussion, bin mir aber auch grad nicht mehr sicher, auf welchen Tarif sich das bezog (der dann umgebaut wurde). Vielleicht weiß es @VolkerK62 noch?

andig, 2025-04-30:
Ich meine mich zu erinnern- es war forecast-solar bei dem wir festgestellt hatten, dass es sich nur um die Leistung zum Stundenbeginn handeln _kann_. Dabei sind wir dann geblieben. `core.timeseries.energy()` zeigt, wie die Werte interpoliert werden (https://github.com/evcc-io/evcc/pull/19554).

thecem, 2025-04-30:
Werde zwei Vorschläge machen, kann aber 14 Tage dauern.

MarkusGH, 2025-05-12:
@thecem: Der github Bot hat das Issue geschlossen - kannst Du abschätzen, bis wann Du dazu kommen würdest.
@andig: Will man das Issue wieder öffnen? Ich bin dazu nicht berechtigt.

MarkusGH, 2025-05-29:
[@thecem](https://github.com/thecem): Der github Bot hat das Issue geschlossen - kannst Du abschätzen, bis wann Du dazu kommen würdest?
[@andig](https://github.com/andig): Magst Du das Issue wieder öffnen?

tantive, 2025-06-29:
[@thecem](https://github.com/thecem): Der github Bot hat das Issue geschlossen - kannst Du abschätzen, bis wann Du dazu kommen würdest?
[@andig](https://github.com/andig): Magst Du das Issue wieder öffnen?

thecem, 2025-06-30:
Wird Ende der Sommerferien.

tantive, 2025-07-06:
Wäre super, wenn jemand folgenden PR reviewen könnte: https://github.com/evcc-io/evcc/pull/22225
Ich habe das open-meteo template auf den instant irradiance Wert zu Beginn jedes Slots angepasst und die Zeitstempel auf UTC/GMT umgestellt.

andig, 2025-07-06:
LGTM. Da es im Moment nicht passt merge ich mal- korrigieren können wir immer noch. VIELEN DANK!