evcc-io/evcc #20950: Take over charges in mode pv via the rest api

stadlaa, 2025-04-29:
**Is your feature request related to a problem? Please describe.**
I develop a more intelligent energy manager that uses evcc as an interface to many devices. I like the evcc features in mode "fast" but want to overtake over the loadpoints in mode "pv" to do more advanced forecast-based stuff with the cars during solar power charging... Therefore I am at the REST-API trying to take over loadpoints, when the user selects mode: "pv" or "min+pv". Sadly there is no direct endpoint to disable or enable the charger. Therefore I have to do the dirty workaround to set the "enable-" and "disable-threshold" to really high or low values when I want the charger to be enabled or disabled on purpose. Further I am only able to limit the chargepower on purpose by setting maxcurrent, which would work quite well seemingly. The problem with moving around the enable and disable threshold is, that a charge is maintained regardless of these thresholds if the soc of the home battery is greater than buffer soc.

**Describe alternatives you've considered**
An easy fix would be to turn the buffer soc up to 100% so the vehicle would never use the battery and always follow the enable and disable threshold right? No! there is a bug in there where the system still maintains a mincurrent charge when the battery is exactly at 100%... Due tu my thesholds that I set to int32.min for disabling the charger on purpose also during the day when pv-power greater min-current is available, the approached min current by evcc to try initiate a charge from the battery is not sufficient to trigger the battery to actually discharge because there is just to much pv-power available... Therefore the battery stays at a 100% and the car continues charging at minsoc forever... I don't have control over that.

**Describe the solution you'd like**
- The possibility to choose something like: "only with pv-power" for the buffer soc, which behaves equivalent to "101%", because "100%" is still a value that can be reached by the battery and would trigger a short charge, which in my case lasts for ever....

- POST Endpoints for enable and disable behaviour of loadpoints in mode "pv" to enable clean control over the begin and ending of a charge by  third party applications. 

- POST Endpoints for not just "min-" and "maxcurrent" but for the actual charge power in mode pv to be able to take over the system without having to move around "min-" and "maxcurrent"

**Additional context**

[image]

andig, 2025-04-30:
> Sadly there is no direct endpoint to disable or enable the charger.

I don‘t understand. Whats wrong with mode: off?

stadlaa, 2025-04-30:
As I said, I want to use the clean and intuitive UI of evcc to give the user full control over his cars and their charges. I don't wan't to create my own UI. I only want to overwrite the behaviour of evcc when the user explicitly requests solar powered charging with mode "pv". If I were to change the mode during the pv-powered-charge to restrict charging, the user might get confused as he has no indicator that the system is trying to charge with pv-power at the time...

To allow third party software to overwrite charges only when it is in mode pv, some endpoints would be great!!
I am currently achieving what I want by setting the thresholds high or low enough, but I run into the issue mentioned above....

andig, 2025-04-30:
There are no plans to provide this.

closed 2025-04-30 by andig

stadlaa, 2025-04-30:
I'm sorry but I can't set the Maxcurrent and MinCurrent so low... I want to fully stop charging at times...
 
I understand if the endpoints are to much trouble... but at least the additional option in the UI to set the bufferSoc to a value other than 0 to 100?

andig, 2025-04-30:
I don't see a non-confusing and compelling case that benefits other users for doing so.

andig, 2025-05-01:
That sounds plausible. Happy to take a PR.

stadlaa, 2025-05-01:
Hmm... have checked the code, nothing suspicious found, should work as expected:

`// if battery is above bufferSoc allow using it for charging
			batteryBuffered = site.bufferSoc > 0 && site.batterySoc > site.bufferSoc
			batteryStart = site.bufferStartSoc > 0 && site.batterySoc > site.bufferStartSoc`

I don't know why my issue even was able to happen... Maybe because I didn't put the bufferStartSoc on "only with enough surplus" or in the above code "0" altough I think to remember that I did...

I will analyze further next time....

Best regards