evcc-io/evcc #20951: 203.4 missing method Phases1p3p

Klausmd5, 2025-04-29:
### Describe the bug

I tried to configure a new Shelly Pro3 (without meter) via the mobile app as new Loadpoint "Pool" as I was planning to use the new loadpoint to power the pool pump a few hours longer, when enough energy is available.

After trying to save, there was an error with saving which caused evcc to create multiple chargers as the log down suggests. After that, evcc automatically rebooted/ the frontend disconnected from the backend. Now there is no way to restart evcc. I've tried overriding the evcc.db with the evcc.yaml, but still no result.

And as i did try to add a feature to evcc on another branch, my setup is currently self compiled. The error occured on the commit #36330ac, after reviewing the commits, I tried if updating to #6aa6e27 would solve the issue, which didn't.

[main  ] INFO 2025/04/30 01:42:53 evcc 0.0.0
[main  ] INFO 2025/04/30 01:42:53 using config file: /home/evcc/evcc.yaml
[db    ] INFO 2025/04/30 01:42:53 using sqlite database: /root/.evcc/evcc.db
wallbox1
--------
Power:          0W
Energy:         4444.9kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 232V 234V
Charge status:  B
Enabled:        false
Identifier:     <none>

db:4
----
Power:          0W
Energy:         4444.9kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 232V 234V
Charge status:  B
Enabled:        false
Identifier:     <none>

db:3
----
Power:          0W
Energy:         4444.9kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 232V 234V
Charge status:  B
Enabled:        false
Identifier:     <none>

db:6
----
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 0V 0V 0V
Power L1..L3:   0W 0W 0W
Charge status:  B
Enabled:        false
Features:       [IntegratedDevice]

### Steps to reproduce

1. Open mobile App, go to Settings
2. add new Loadpoint and Charger, select shelly
3. click save

### Configuration details

```yaml
root@evcc:/home/evcc# ./evcc dump --cfg
[main  ] INFO 2025/04/30 01:37:21 evcc 0.0.0
[main  ] INFO 2025/04/30 01:37:21 using config file: /home/evcc/evcc.yaml
[db    ] INFO 2025/04/30 01:37:21 using sqlite database: /root/.evcc/evcc.db
[MTEC-battery] INFO 2025/04/30 01:37:21 connecting evcc-1161446317 at tcp://192.168.0.118:1883
[MTEC-battery] DEBUG 2025/04/30 01:37:21 tcp://192.168.0.118:1883 connected
[MTEC-battery] DEBUG 2025/04/30 01:37:21 tcp://192.168.0.118:1883 subscribe MTEC/Z112300207530284/now-base/battery
[MTEC-battery] DEBUG 2025/04/30 01:37:21 tcp://192.168.0.118:1883 subscribe MTEC/Z112300207530284/now-base/grid_power
[MTEC-battery] DEBUG 2025/04/30 01:37:21 tcp://192.168.0.118:1883 subscribe MTEC/Z112300207530284/now-base/pv
[circuit-main] DEBUG 2025/04/30 01:37:22 validation of max phase current disabled
[lp-2  ] DEBUG 2025/04/30 01:37:22 set title: Pool
[lp-2  ] DEBUG 2025/04/30 01:37:22 set priority: 0
[lp-2  ] DEBUG 2025/04/30 01:37:22 set smart cost limit: <nil>
[lp-2  ] DEBUG 2025/04/30 01:37:22 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2  ] DEBUG 2025/04/30 01:37:22 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2  ] DEBUG 2025/04/30 01:37:22 set session energy limit: 0
[lp-2  ] DEBUG 2025/04/30 01:37:22 set session soc limit: 0
[lp-2  ] DEBUG 2025/04/30 01:37:22 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc0010f64cc}
[lp-2  ] DEBUG 2025/04/30 01:37:22 set default mode: pv
[lp-2  ] DEBUG 2025/04/30 01:37:22 set phases: 1
panic: interface conversion: *struct { *charger.Shelly; api.PhaseCurrents; api.PhasePowers; api.PhaseVoltages } is not api.PhaseSwitcher: missing method Phases1p3p

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).SetPhasesConfigured(0xc000109688, 0x1)
        github.com/evcc-io/evcc/core/loadpoint_api.go:258 +0x2f0
github.com/evcc-io/evcc/core/loadpoint.DynamicConfig.Apply({{0xc000e2176c, 0x4}, {0xc000e21698, 0x2}, 0x0, 0x1, 0x0, 0x0, 0x0, 0x0, ...}, ...)
        github.com/evcc-io/evcc/core/loadpoint/config.go:73 +0x1e3
github.com/evcc-io/evcc/cmd.configureLoadpoints({{{0xc00022eca8, 0x4}, {0xc00022ecb0, 0xa}, 0x1b9e}, {0xc00022eceb, 0x5}, {0xc00022ee20, 0xb}, {0xc000923040, ...}, ...})
        github.com/evcc-io/evcc/cmd/setup.go:1039 +0x8ca
github.com/evcc-io/evcc/cmd.configureSiteAndLoadpoints(0x5d7a4a0)
        github.com/evcc-io/evcc/cmd/setup.go:917 +0x118
github.com/evcc-io/evcc/cmd.runDump(0x5d78020, {0x2d34db0?, 0x4?, 0x2d34d94?})
        github.com/evcc-io/evcc/cmd/dump.go:57 +0x7a
github.com/spf13/cobra.(*Command).execute(0x5d78020, {0xc000a9cf90, 0x1, 0x1})
        github.com/spf13/cobra@v1.9.1/command.go:1019 +0xa7b
github.com/spf13/cobra.(*Command).ExecuteC(0x5d785a0)
        github.com/spf13/cobra@v1.9.1/command.go:1148 +0x40c
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
        github.com/evcc-io/evcc/cmd/root.go:120 +0x1a
main.main()
        github.com/evcc-io/evcc/main.go:44 +0x2d
```

### Log details

```text
root@evcc:/home/evcc# ./evcc
[main  ] INFO 2025/04/30 01:32:52 evcc 0.0.0
[main  ] INFO 2025/04/30 01:32:52 using config file: /home/evcc/evcc.yaml
[db    ] INFO 2025/04/30 01:32:52 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2025/04/30 01:32:52 listening at :7070
[MTEC-battery] INFO 2025/04/30 01:32:52 connecting evcc-278026599 at tcp://192.168.0.118:1883
[MTEC-battery] DEBUG 2025/04/30 01:32:52 tcp://192.168.0.118:1883 connected
[MTEC-battery] DEBUG 2025/04/30 01:32:52 tcp://192.168.0.118:1883 subscribe MTEC/Z112300207530284/now-base/battery
[MTEC-battery] DEBUG 2025/04/30 01:32:52 tcp://192.168.0.118:1883 subscribe MTEC/Z112300207530284/now-base/grid_power
[MTEC-battery] DEBUG 2025/04/30 01:32:52 tcp://192.168.0.118:1883 subscribe MTEC/Z112300207530284/now-base/pv
[circuit-main] DEBUG 2025/04/30 01:32:53 validation of max phase current disabled
[lp-2  ] DEBUG 2025/04/30 01:32:53 set title: Pool
[lp-2  ] DEBUG 2025/04/30 01:32:53 set priority: 0
[lp-2  ] DEBUG 2025/04/30 01:32:53 set smart cost limit: <nil>
[lp-2  ] DEBUG 2025/04/30 01:32:53 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2  ] DEBUG 2025/04/30 01:32:53 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2  ] DEBUG 2025/04/30 01:32:53 set session energy limit: 0
[lp-2  ] DEBUG 2025/04/30 01:32:53 set session soc limit: 0
[lp-2  ] DEBUG 2025/04/30 01:32:53 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000e3a32c}
[lp-2  ] DEBUG 2025/04/30 01:32:53 set default mode: pv
[lp-2  ] DEBUG 2025/04/30 01:32:53 set phases: 1
panic: interface conversion: *struct { *charger.Shelly; api.PhaseCurrents; api.PhasePowers; api.PhaseVoltages } is not api.PhaseSwitcher: missing method Phases1p3p

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).SetPhasesConfigured(0xc000e04d88, 0x1)
        github.com/evcc-io/evcc/core/loadpoint_api.go:258 +0x2f0
github.com/evcc-io/evcc/core/loadpoint.DynamicConfig.Apply({{0xc000d316ac, 0x4}, {0xc000d315d8, 0x2}, 0x0, 0x1, 0x0, 0x0, 0x0, 0x0, ...}, ...)
        github.com/evcc-io/evcc/core/loadpoint/config.go:73 +0x1e3
github.com/evcc-io/evcc/cmd.configureLoadpoints({{{0xc000b50e78, 0x4}, {0xc000b50e80, 0xa}, 0x1b9e}, {0xc00044a49b, 0x5}, {0xc00044a590, 0xb}, {0xc00093f180, ...}, ...})
        github.com/evcc-io/evcc/cmd/setup.go:1039 +0x8ca
github.com/evcc-io/evcc/cmd.configureSiteAndLoadpoints(0x5d7a4a0)
        github.com/evcc-io/evcc/cmd/setup.go:917 +0x118
github.com/evcc-io/evcc/cmd.runRoot(0x5d785a0, {0x2d412e4?, 0x7?, 0x2d34d94?})
        github.com/evcc-io/evcc/cmd/root.go:197 +0x97f
github.com/spf13/cobra.(*Command).execute(0x5d785a0, {0xc00004e0b0, 0x0, 0x0})
        github.com/spf13/cobra@v1.9.1/command.go:1019 +0xa7b
github.com/spf13/cobra.(*Command).ExecuteC(0x5d785a0)
        github.com/spf13/cobra@v1.9.1/command.go:1148 +0x40c
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
        github.com/evcc-io/evcc/cmd/root.go:120 +0x1a
main.main()
        github.com/evcc-io/evcc/main.go:44 +0x2d
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

203.4

Klausmd5, 2025-04-30:
Update - Was able to remove the db settings entrys using datagrip and then placing the db back in. Works for now.

andig, 2025-04-30:
@Klausmd5 bei den Commit Hashes bitte das `#` weg lassen sonst werden die als Issue/PR interpretiert. Bei https://github.com/evcc-io/evcc/commit/36330ac2ccbc3b17b53d9d64f4505fe92990c541 kann ich nichts relevantes finden. Leider ist auch der Stacktrace nicht aussagekräftig. Könntest Du bitte nochmal einen vom aktuellen Nightly produzieren oder den fehlerhaften Commit direkt benennen? Ich finde leider keine Ursache.

thierolm, 2025-04-30:
Please provide the output of `evcc -l trace charger`

andig, 2025-05-03:
ping @Klausmd5

andig, 2025-05-04:
Mangels feedback geschlossen. Schade :/

closed 2025-05-04 by andig