evcc-io/evcc #20959: UI: themes / custom colors

madmat17, 2025-04-30:
This feature request is not to be considered as must-have, rather than a nice-to-have...

I'm a really big fan of evcc's UI (user interfaces are like jokes: they're bad, if they need explanation...). It is easy to navigate, intuitive and self-explaining (layout as well as language wise) 

One thing I'd be missing though is the option to tweak the colors to my very own and personal liking.
Why? The idea is having a bit more customizable UI so that would better integrate into existing dashboards like in HomeAssistant, ioBroker/VIS or whatever dashboard that support iframes.
That would allow evcc to integrate more seamlessly into existing environments.

The naive me thought that this could be accomplished by allow the user to set CSS color codes (either HEX, or named) for predefined <div> classes by either...
- allow the settings within the UI / SQLite in the backend,
- add the respective parameters in the `evcc.yaml` file (I assume this is not the way to go as everything seems to move to the DB-based configuration)
- parse evcc's root directory for an `app.css` file on every start, and replace `assets/css/app.css`. Tradeoff would be, that "bad" changes in the CSS file could completely distort the UI, and bigger changes in the app.css layout from one evcc version to another could have the same effect, or even render the UI useless.

Taking the last point above into consideration, the best approach still seems to only expose certain color-settings and run a script/routine on each evcc start, which is generating the `assets/css/app.css` whilst replacing default color codes by any custom values.

naltatis, 2025-04-30:
Hi @madmat17, I see the point. Our CSS is structured in a way that makes simple changes like changing colors pretty easy through CSS properties. However I'm undecided if this is a door we should open. I wouldn't see this as an end-user feature in config UI. As you've mentioned injecting CSS has the potential to break things for you and even if it's working correctly we cannot guarantee any backward compatibility. However the effort of introducing this is not very hight and other projects like Pretalx or [Home Assistant](https://developers.home-assistant.io/docs/frontend/custom-ui/registering-resources) offer similar capabilities.

We could introduce this feature via a new flag (`--inject-css=/local/evcc.css`). Since I'd see this as a pure tech/dev-feature (see implications above) I'd assume that this mechanism is approachable enough. On startup evcc would read the contents of this file and inject it into a style tag after our own CSS.

madmat17, 2025-04-30:
Hi @naltatis, I'm stunned how swiftly you picked up my "issue"! 😯

I get the point that this is a niche request  - but would still quite neat.
The idea of injecting a custom CSS sounds interesting. To avoid backward compatibility problems (in case you decide to implement the CSS injection at all): Would it make sense to probe version numbers in `assets/css/app.css` and `some/custom/path/evcc.css` against each other? An the CSS should only be injected, if the versions match. Version number could be the first line of the CSS like `/* v0.203.x */`, whereas the reference number only changes with those evcc versions where default CSS changes as well).
That would "force" folks to adapt to new reference CSS structure/updates and would avoid UI issues after updates.

An-Wi, 2025-05-19:
What about the colors in the "Charging sessions" view? Are the bar colors also defined via CSS? 
Specifically, I'm interested in the colors in the "Vehicles" report.

naltatis, 2025-05-19:
Partly. The colors for cost, grid, export, ... are shared and pulled from CSS. The generic color palette for load points and vehicles is hard coded and will not be changeable via right now.
https://github.com/evcc-io/evcc/blob/master/assets/js/colors.ts

madmat17, 2025-05-19:
I totally dismissed the JS part of the UI.
@naltatis : How about using JS to parse a custom color CSS file (which might, or might not exist) and use applicable values? Check, if e.g. `--evcc-background` is found in the file `/install/dir/custom.css` in a valid format to be applied and use `assets/css/app.css`, if no (valid) custom CSS has been defined.

naltatis, 2025-05-19:
Look here https://github.com/evcc-io/evcc/blob/master/assets/js/colors.ts#L66-L77

crazyfx1, 2025-05-19:
This looks great so far. 
As far as I can see it's not possible to use this in the home-assistant addon?
In (my) ideal world, when using the home-assistant addon, there could be a custom.css which uses the configured colors from home-assistant.

closed 2025-05-20 by naltatis