evcc-io/evcc #20980: Chargmode with dynamic price

enboer, 2025-05-01:
I understood that there is a project running to adjust the charging plan (Epic: Planner enhancements).  On of the issue is to not use Solar charge because of high tariff it is cheaper to return it to the net and charge later during the night.

But is it maybe a option to have a setting to disable solar charging if current price is above a certain level. Like there is now already a setting that if price is eg below 12ct it switches to fast charging. Is that something that can be easily implemented?

I created a screenshot how this could look like.

[image]

[image]

andig, 2025-05-01:
Unfortunately, its unclear what the dutch suggestion means :/

closed 2025-05-01 by andig

enboer, 2025-05-01:
[image]

[image]

enboer, 2025-05-01:
:-)