evcc-io/evcc #20982: Bug: EVCC web interface displays theoretical max charging power instead of actual value

Macoperator, 2025-05-01:
### Describe the bug

## 🐞 Bug: EVCC web interface displays theoretical max charging power instead of actual value

### Description

Since updating to EVCC version `0.203.4`, the web interface displays a charging power of **9.0 kW** during charging sessions, even though the vehicle is actually charging at a much lower rate (e.g., **4.1 kW**). The displayed value appears to reflect my *last configured maximum current limit* rather than the *actual charging power* being delivered.

This is misleading, especially when EVCC is actively regulating the charging current based on PV surplus. In my case, EVCC correctly limits charging to the available PV surplus (e.g., 6A), but the UI still shows 9 kW as the charging power.

### Expected behavior

EVCC has **full real-time access to the home energy management system (E3DC in my case)** and is aware of:
- PV production
- household consumption
- battery state and power
- available PV surplus

EVCC correctly uses this data to compute the charging current and sends it to the charger. For example, when it limits the charger to 6A, the actual charging power is around 4.1 kW — **not** 9 kW.

Therefore, the UI should reflect the *actual charging power* based on either:
- the effective `charge current` EVCC sets after PV surplus calculation, or
- the calculated power flow at the site (PV + grid + battery + consumption), especially if no separate charger meter is configured.

### Screenshots

*(See attachments – e.g. screenshot showing 9 kW display while only ~4 kW of PV power is available, and grid import is zero.)*

### Suggestion

If **no dedicated charging meter** is configured, the UI should:
- either display the power based on the dynamically set charging current (e.g., 6A × 3 phases × 230 V),
- or derive the actual value from site power flow measurements.

Showing a theoretical maximum (based on the charger’s configured limit) is misleading in dynamic PV scenarios.

### System details

- Raspberry Pi Zero W
- EVCC 0.203.4
- Communication via RS485 with E3DC
- No dedicated meter at the wallbox
- Charging mode: "Min+PV"

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 4791ec8ca82ead4f9303a75c82c5ec81b45f4abfdbc84d3425d23b2b42ca3dde

interval: 30s # control cycle interval
[... 156 lines cut ...]
[site ] DEBUG 2025/05/01 13:44:25 battery 1 power: -1326W
time="2025-05-01T13:44:25+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site ] DEBUG 2025/05/01 13:44:25 battery 1 soc: 95%
[site ] DEBUG 2025/05/01 13:44:25 site power: -3060W
[lp-1 ] DEBUG 2025/05/01 13:44:26 charger status: C
[lp-1 ] DEBUG 2025/05/01 13:44:27 pv charge current: 17.4A = 13A + 4.43A (-3060W @ 3p)
[site ] DEBUG 2025/05/01 13:44:55 ----
[lp-1 ] DEBUG 2025/05/01 13:44:55 charge power: 8970W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4

andig, 2025-05-01:
> Therefore, the UI should reflect the actual charging power based on either:
the effective charge current EVCC sets after PV surplus calculation

It does

andig, 2025-05-01:
Or rather: it should. Can you please show the output of /api/state for this situation?

andig, 2025-05-01:
> charge power: 8970W

You are charging at 9k, so thats correct.

closed 2025-05-01 by andig

Macoperator, 2025-05-01:
No, that's my point: The car was charged with 4kW almost all of the time, as I could clearly see on the Tesla's screen, in my local teslamate installation and also according to my E3DC household power output*. EVCC is set to solar surplus charging, which also works as expected. The only thing that's wrong since the last update is EVCC's dashboard showing a wrong charging current. 
* (see attached screenshot. Charging was between 2 and 5 pm on that chart and the red line, which is overall power consumption, remains under the yellow line, which is the available pv power)

andig, 2025-05-01:
> The car was charged with 4kW almost all of the time

That may be as it will. Evcc has no way of knowing how the car actually charges wirhout a meter. It can only know the offered current.

Macoperator, 2025-05-01:
Hi again – I want to clarify that this is not just a cosmetic issue in the UI. This new behavior introduced in 0.203.4 actively breaks EVCC's own charging statistics.

Case in point:

- On Sunday (with a version before 0.203.4), EVCC showed a charging session with 18.3 kWh. This value matched Tesla’s app and my local Teslamate setup exactly.
- Today, with the same car and the same oversupply charging setup, EVCC reports 23.3 kWh charged — while Tesla and Teslamate both say only 12.0 kWh were added.

That’s an 11.3 kWh overestimation in EVCC — solely due to the fact that EVCC now calculates the charged energy based on the *configured max current*, not the *actual power flow*.

This behavior is not only misleading — it results in completely wrong statistics.

EVCC has access to the site power, battery, PV and grid data — it should be able to infer the real charge power from there, or at the very least **use the actual current** set by the controller (after PV-based throttling).

Please consider re-opening this issue — the current behavior introduces a *statistical regression* and undermines EVCC’s role as a reliable charging controller.

andig, 2025-05-02:
> pv charge current: 17.3A = 13A + 4.25A (-2935W @ 3p)
> charge power: 8970W

13A matches 9kW and the calculation is correct afaikt

> solely due to the fact that EVCC now  calculates the charged energy based on the configured max current, not the actual power flow.

Is this „fact“ supported by any log? Please share it.

> calculates the charged energy based

Lets stay with power as topic. If anything then energy is a side effect.

Macoperator, 2025-05-02:
It turns out the incorrect charge power shown by EVCC wasn’t a software issue after all: my Tesla had been manually limited to 6A in the Tesla app. That value persisted across charges and silently overrode EVCC’s current settings. After setting the slider back to 13A (my manually set max charging power in EVCC), EVCC could fully control the charge current again and the display shows correct power values now. Maybe EVCC could eventually warn when a set current isn’t being accepted?

andig, 2025-05-02:
There is no such thing as unaccepted. The solution is simply having a charge meter.

Macoperator, 2025-05-02:
Sorry for wasting your time.