evcc-io/evcc #20984: Deye Micro Inverter SUNMXXX-G4 not working

ChristianKleinschroth, 2025-05-01:
### Describe the bug

Deye updated (and auto-update) all G3 and G4 micro inverters - so maybe register had changed and AT Connection over Port 48899 is not allowed anymore.

I've did a lot of research but in the end the template for deye micro inverter will not running -
My device is a https://de.deyeinverter.com/product/microinverter-1/SUNM130-160-200G3EUQ0P1-13002000W-Single-Phase-4-MPPT-Micro-Inverter.html with only can be connected via wifi and port 8899 (502 is only working for 3-phase inverter)

First either with modbus tcp or modbus rtu the connection is not working

Modbus over TCP says

energy: read failed: modbus: length in response header '23107' must not be zero or greater than '254'

Then only project that I found that runs out of the box is the  solarman home assitant integration

https://github.com/davidrapan/ha-solarman

Especially you should look at the deye_micro.yaml - I guess the the evcc template for this needs an update

https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/deye_micro.yaml

Unfortunately im not an expert for modbus, but I can say with the ha extension it works like it should with evcc - at least we will have no issue to connect or modify / downgrade the latest device firmware.

Thank you in advance for your great work - hope We can fix the template together 

### Steps to reproduce

1. Find out the local ip of your inverter within the wifi
2. Add PV Device with Deye Micro Inverter
3. Add IP 
4. Change Port from 502 to 8899
5. Set Modbus TCP (or RTU) 
6. Check the Connection
7. energy: read failed: modbus: length in response header '23107' must not be zero or greater than '254'
...

### Configuration details

```yaml
Configuration is done via UI (Experimental)so evcc dump has no configuration
```

### Log details

```text
db:2 ] TRACE 2025/05/01 17:03:53 modbus: send 04 11 00 00 00 06 01 03 00 56 00 02
[db:2 ] TRACE 2025/05/01 17:03:53 modbus: close connection: modbus: length in response header '23107' must not be zero or greater than '254'
[db:2 ] TRACE 2025/05/01 17:03:53 modbus: send 04 12 00 00 00 06 01 03 00 3f 00 02
[db:2 ] TRACE 2025/05/01 17:03:53 modbus: close connection: modbus: length in response header '23107' must not be zero or greater than '254'
[httpd ] TRACE 2025/05/01 17:03:54 GET /api/auth/status
[db:2 ] TRACE 2025/05/01 17:04:53 modbus: closing connection due to idle timeout: 1m0.000379718s
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.203.3

ChristianKleinschroth, 2025-05-01:
also tested with 0.203.4 - no difference

andig, 2025-05-01:
8899 seems to indicate Modbus UDP. Did you configure that?

> Set Modbus TCP

Apparently not.

closed 2025-05-01 by andig

ChristianKleinschroth, 2025-05-01:
Found nothing about UDP in your doc https://docs.evcc.io/docs/devices/meters#micro-inverter
How should I test it ?

Also inverter settings are set to 8899/tcp
Modbus RTU work’s neither

andig, 2025-05-01:
> How should I test it ?

Use yaml instead. 

> Also inverter settings are set to 8899/tcp

Whatever that means. Its obviously not modbus tcp or it would work. It would help to find out what the device actually speaks before trying any wild guesses.

ChristianKleinschroth, 2025-05-04:
I figured out how this device is working - they use modbus only as a transport layer but answers are in ascii

Example in evcc.yaml

`meters:
  - type: custom
    name: bkw
    power: 
      source: modbus
      id: 1
      uri: 192.168.10.200:8899
      timeout: 10s # timeout, without unit in ns
      rtu: true # Modbus RTU over TCP
      register:
          address: 109
          type: holding # coil, holding or input
          decode: int32 # int16|32|64, uint16|32|64, float32|64 and u|int32s + float32s
          bitmask: 0 # Optional: a bitmask that is applied to the read value. Here the mask is 0000000000000010b, ignored if value is 0
      scale: 0.1 # floating point factor applied to result, e.g. for kW to W conversion
    energy:`

The evcc Modbus client sends a request to device ID 1, asking to read 2 holding registers starting at address 109.
[bkw ] TRACE 2025/05/04 13:56:18 modbus: send 01 03 00 6d 00 02 55 d6

The device replies not with raw register values, but with an ASCII string that was likely encoded/stored in those registers.

[bkw ] TRACE 2025/05/04 13:56:18 modbus: recv 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 38 5f 35 2e 30 43 2d 53 0d 0a 0d 0a

The decoded string is:
AT+YZCMPVER=MW3_16U_5408_5.0C-S

This looks like:
A firmware version string
Possibly from a cellular or wireless communication module (e.g., based on "AT+")

Fact is we cannot use modbus for deye micro inverter template - how we proceed here?

andig, 2025-05-07:
> The evcc Modbus client sends a request to device ID 1, asking to read 2 holding registers starting at address 109.

2 registers are 2x16 bits. If the thing answers with any number of bytes then it's not modus and will need it's own implementation. Do they advertise modbus? Then they should fix it. It's absolutely unclear why it seems to be so hard to follow long-established standards...

ChristianKleinschroth, 2025-05-08:
No they don't advertise - they sell or integrate a wifi logger call solarman - and so they name their protocol (that weird one with modbus read and ascii answers) also solarman protocol

I would rather prefrer to do it like the home assitant why, usng solarman integration https://github.com/StephanJoubert/home_assistant_solarman

In the end, the actual template for deye micro inverter will expect a modbus only protocol -thats not gonna happen.
So can we change that?

andig, 2025-05-08:
Change what?

ChristianKleinschroth, 2025-05-08:
changing the template for deye micro inverters to solarman integration

andig, 2025-05-08:
As I've said: this isn't modbus and hence a tempalte won't work. This device will need it's own implementation. Happy to take a PR.