evcc-io/evcc #20986: Auth plugin + Authprovider for unified OAuth2

lehmanju, 2025-05-01:
Anfänge, um die Authentifizierung der verschiedenen APIs zu vereinheitlichen existieren schon mit `Authorizer` und `AuthProvider`. Allerdings handelt es sich dabei im Wesentlichen um Scaffolding und wird noch nirgends wirklich eingesetzt.

Dieser Issue soll konkretisieren, was eigentlich das gewünschte Ziel und eine gute API sind. Am sinnvollsten ist es vermutlich erstmal OAuth2 zu betrachten, da dass am weitesten verbreitet ist.

Ein OAuth2 Plugin muss folgendes können:
- Zwischen den Zuständen _Authorised_ und _Unauthorised_ wechseln können. _Authorised_ bedeutet, dass der Nutzer weitergeleitet wurde und die App verknüpft hat und der Refresh Token gültig ist. _Unauthorised_ bedeutet, dass der Refresh Token ungültig ist oder dass der Nutzer noch gar nicht approved hat.
- Einen authentifizierten `http.RoundTripper` zur Verfügung stellen (siehe `Authorizer`)
- Ein Callback zur Verfügung stellen für den (initialen oder wiederholten) Authentfizierungsprozess (siehe `AuthProvider.LoginHandler`)
- Über die typische `oauth2.Config` konfiguriert werden
- Identität über mehrere Instanzen des Plugins hinweg teilen können

andig, 2025-05-01:
Super Idee. Siehe https://github.com/evcc-io/evcc/pull/20111/files#diff-e197f19bd76c072ea9e90586460f05e3c6d20f22ffe844468b7498a85a41b76e für eine Spielart davon. Ich sehe zwei wesentliche Hürden:

- OAuth2 wird leider selten als Standard implementiert. Üblich sind zusätzliche Header etcetc. Dafür braucht es dann leider doch individuelle Lösungen.
- Es braucht Integration in die Anwendungsfälle. Ohne bleibt das eine sehr theoretische Übung.

Long story short: wenn es einen konkreten Fall gibt würde ich mich sehr über einen PR freuen. Falls nein: wontfix :/

lehmanju, 2025-05-01:
Ja, das mit den zusätzlichen Headern hab ich schon im Kopf. Da die meistens statisch sind, kann man die auch einfach dranhängen per Konfigurationsoption.

Ich würde das ganze mit der Volvo Integration probieren, weil ich die jetzt schon gut kenne. Allerdings habe ich im Code viel ähnliches gesehen, sodass ich sehr zuversichtlich bin, dass sich das vereinheitlichen lässt.

andig, 2025-05-01:
Das wäre Klasse. Dann würde ich aber vorschlagen: Volvo (einfache Version). Dann Volvo schöne Version (mit Login übers GUI, so wie wir es für Mercedes mal hatten). Dann Volvo abstrahiert als OAuth2 Plugin? Bis dahin könnten wir hier zu machen, du hast meinen vollen Support.

lehmanju, 2025-05-01:
Achso, ich bin direkt dabei ein generisches OAuth2 Plugin zu schreiben ;). Deswegen der Issue, damit man die Implementierungsvorgensweise dokumentieren kann. Das mit der GUI kommt natürlich erst später.

Meine Idee ist jetzt erstmal das `Authorizer` interface zu erweitern, dass es die beiden States abdecken und im Zweifel den Loginprozess anstoßen kann. Dazu dann wie mit Viessmann ein generisches OAuth2 Plugin und dann mit Volvo eine erste Implementierung.

Aber die Diskussion kann man natürlich auch im geschlossenen Issue weiterführen.

andig, 2025-05-01:
Cool. Dann lass uns doch die Diskussion in den PR verlagern sobald es einen gibt ;)

andig, 2025-05-04:
...und weiter im PR.

PR #21003 (2025-05-15): Volvo Connected: add oauth using generic plugin
Fixes #20986 
Fixes #20631 

Ein generisches OAuth2 plugin, dass die verschiedenen "Identities" in den `vehicle` APIs vereinheitlichen soll.
closed 2025-05-04 by andig