evcc-io/evcc #20991: Show planner price in forecast

StefanSchoof, 2025-05-02:
**Is your feature request related to a problem? Please describe.**
Ich habe keinen dynamischen Strompreis. Ich möchte aber trotzdem möglich dann laden wenn der Strompreis niedrig und der Anteil an erneuerbaren hoch ist. 

Daher habe ich zusätzlich zu meinen fixed `grid` einen `planner` tariff mit dem Börsenstrompreisen angelegt:

```yaml
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.2829 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
  planner:
    type: template
    template: energy-charts-api
    bzn: DE-LU
  solar:
    - type: template
      template: open-meteo
      lat: 52.31
      lon: 13.41
      dec: 45
      kwp: 10
      az: 0
```

Dies wird im Planner auch angezeigt:

[image]

In der Vorhersage wird der Preis aber nicht angezeigt.

[image]

**Describe the solution you'd like**

Um mir einen möglichst guten Zeitpunkt für das nächste Laden auszusuchen wäre es hilfreich wenn die Planner Preisinfo auch in den Vorhersage Dialog angezeigt wird.

**Describe alternatives you've considered**

Weiterhin die [Strompreisampel ](https://energy-charts.info/charts/consumption_advice/chart.htm?l=de&c=DE&source=price_traffic_light) als zusätliche Info öffnen und die Infos im Kopf zusammenführen.

**Additional context**

<details>

<summary>demo yaml erweitert um planner und solar für Vorhersage</summary>

```yaml
network:
  port: 7070

log: debug

interval: 3s

javascript:
  - vm: shared
    script: |
      state = {
        residualpower: 500,
        pvpower: 5000,
        batterypower: -750,
        batterySoc: 55,
        gridpower: -1000,
        loadpoints: [
          { enabled: true, vehicleSoc: 62, maxcurrent: 6, phases: 1, chargepower: 0 },
          { enabled: false, vehicleSoc: 22, maxcurrent: 0, phases: 3, chargepower: 0 }
        ]
      };
      function logState() {
        console.log("state:", JSON.stringify(state));
      }

meters:
  - name: grid
    type: custom
    power:
      source: js
      vm: shared
      script: |
        state.gridpower = state.loadpoints[0].chargepower + state.loadpoints[1].chargepower + state.residualpower - batterypower - pvpower;
        state.gridpower;
      in:
        - name: pvpower
          type: float
          config:
            source: js
            vm: shared
            script: |
              state.pvpower = 8000+500*Math.random();
              state.pvpower
        - name: batterypower
          type: float
          config:
            source: js
            vm: shared
            script: |
              state.batterypower = state.gridpower > 0 ? 1000 * Math.random() : 0;
              state.batterypower

  - name: pv
    type: custom
    power:
      source: js
      vm: shared
      script: state.pvpower;

  - name: battery
    type: custom
    power:
      source: js
      vm: shared
      script: state.batterypower;
    soc:
      source: js
      vm: shared
      script: |
        if (state.batterypower < 0) state.batterySoc++; else state.batterySoc--;
        if (state.batterySoc < 10) state.batterySoc = 90;
        if (state.batterySoc > 90) state.batterySoc = 10;
        state.batterySoc;
    capacity: 13.4
    batterymode:
      source: js
      vm: shared
      script: |
        1

  - name: meter_charger_1
    type: custom
    power:
      source: js
      vm: shared
      script: state.loadpoints[0].chargepower;

  - name: meter_charger_2
    type: custom
    power:
      source: js
      vm: shared
      script: state.loadpoints[1].chargepower;

chargers:
  - name: charger_1
    type: custom
    enable:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[0];
        lp.enabled = enable;
        enable;
      out:
        - name: enable
          type: bool
          config:
            source: js
            vm: shared
            script: |
              if (enable) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    enabled:
      source: js
      vm: shared
      script: |
        state.loadpoints[0].enabled;
    status:
      source: js
      vm: shared
      script: |
        if (state.loadpoints[0].enabled) "C"; else "B";
    maxcurrent:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[0];
        lp.maxcurrent = maxcurrent;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;

  - name: charger_2
    type: custom
    enable:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[1];
        lp.enabled = enable;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    enabled:
      source: js
      vm: shared
      script: |
        state.loadpoints[1].enabled;
    status:
      source: js
      vm: shared
      script: |
        if (state.loadpoints[1].enabled) "C"; else "B";
    maxcurrent:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[1];
        lp.maxcurrent = maxcurrent;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    tos: true
    phases1p3p:
      source: js
      vm: shared
      script: |
        logState();
        if (phases === 1) lp.phases = 1; else lp.phases = 3;
        lp.phases;

vehicles:
  - name: vehicle_1
    title: blauer e-Golf
    type: custom
    soc:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0];
        if (lp.chargepower > 0) lp.vehicleSoc+=0.1; else lp.vehicleSoc-=0.1;
        if (lp.vehicleSoc < 15) lp.vehicleSoc = 80;
        if (lp.vehicleSoc > 80) lp.vehicleSoc = 15;
        lp.vehicleSoc;
    range:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0]
        var range = (44 * lp.vehicleSoc) / 15;
        range
    capacity: 44
  - name: vehicle_2
    title: weißes Model 3
    type: custom
    soc:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[1];
        if (lp.chargepower > 0) lp.vehicleSoc++; else lp.vehicleSoc--;
        if (lp.vehicleSoc < 15) lp.vehicleSoc = 75;
        if (lp.vehicleSoc > 75) lp.vehicleSoc = 15;
        lp.vehicleSoc;
    range:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[1]
        var range = (80 * lp.vehicleSoc) / 17;
        range
    status:
      source: js
      vm: shared
      script: |
        "B"
    capacity: 80
    limitsoc:
      source: const
      value: 90
  - name: vehicle_3
    type: template
    template: offline
    title: grüner Honda e
    capacity: 8
  - name: vehicle_4
    type: template
    template: offline
    title: schwarzes VanMoof
    icon: bike
    capacity: 0.46
  - name: vehicle_5
    type: template
    template: offline
    title: Wärmepumpe
    icon: waterheater

site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
    battery: battery

loadpoints:
  - title: Carport
    charger: charger_1
    mode: pv
    meter: meter_charger_1
    vehicle: vehicle_1
  - title: Garage
    charger: charger_2
    mode: "off"
    meter: meter_charger_2
    vehicle: vehicle_2

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.2829 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
  planner:
    type: template
    template: energy-charts-api
    bzn: DE-LU
  solar:
    - type: template
      template: open-meteo
      lat: 52.31
      lon: 13.41
      dec: 45
      kwp: 10
      az: 0
  #co2:
  #  type: grünstromindex
  #  zip: 12349

```

</details>

naltatis, 2025-05-02:
Puh, das ist schon ein bisschen nieschig und technisch auch eher kompliziert. Heißt du würdest eine vierte Option im Forecastdialog "Planer" haben wollen? Inhaltlich kann das sowohl co2, preis und technisch sogar Solarprognose sein.

Mir ist dein Anwendungsfall allerdings noch nicht klar. Wenn du den Planer öffnest siehst du ja deine gewünschten Infos. Geht es dir um die gemeinsame Anzeige um PV-Vorhersage und Börsenstrompreis? Wenn wir prognosenbasiertes Planning implementieren würden wir das vmtl. ohnehin mit in der Planungsvisualiserung mit aufnehmen.

StefanSchoof, 2025-05-02:
> Geht es dir um die gemeinsame Anzeige um PV-Vorhersage und Börsenstrompreis

Ja

> Heißt du würdest eine vierte Option im Forecastdialog "Planer" haben wollen? Inhaltlich kann das sowohl co2, preis und technisch sogar Solarprognose sein.

Nein, eigentlich würde ich nur gerne das die Preise die evcc über den `planner` tariff abruft auch in dem Vorhersage Dialog sichtbar sind.

> technisch auch eher kompliziert

Hatte gehofft das es eine Kleinigkeit ist. Aber wenn es nicht ohne ist, ist es wahrscheinlich wirklich zu nieschig.

naltatis, 2025-05-02:
> Nein, eigentlich würde ich nur gerne das die Preise die evcc über den planner tariff abruft auch in dem Vorhersage Dialog sichtbar sind.

Der Plannertarif ist halt sehr flexibel. Wir könnten in die Visualisierung ne Logik reinbauen, dass es, wenn es keinen (dynamischen?) Grid-Preis gibt und der Type des Planer Tarifs aber ein dynamischer Preis ist für die Preise einfach den Planner Tarif anzeigen. Wäre technisch einfach und würde auf deinen Anwendungsfall passen. Ist aber glaube ich nicht sehr intuitiv.

Daher würd ich den hier jetzt zumachen, das Thema Solar-Forecast + Planner-Tarif aber für die spätere Überarbeitung der Planner-Ansicht mitnehmen.

closed 2025-05-02 by naltatis

StefanSchoof, 2025-05-02:
Danke.