evcc-io/evcc #21011: Octopus Electricity tariff data is being ignored

realist625, 2025-05-03:
### Describe the bug

The price from octopus energy is being ignored - I can see it pulling id down, but it never forms part of the logic, thus all price driven functionality does not work

### Steps to reproduce

1.Configure Grid Pricing in the GUI, using either API key or Product code
2.restart - see the price being pulled from the octopus api server - never takes effect in the GUI
3. ...quietly cry as the car will no longer charge on its own - without manual intervention
...

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: *****

interval: 5s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: eastron-sdm220_230
  id: 1
  device: /dev/serial0
  baudrate: 9600
  comset: 8N1
  usage: grid
  modbus: rs485serial
  name: grid1
- type: template
  template: solaredge-inverter
  id: 1
  host: 10.184.63.81
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv1

- type: template
  template: victron-energy
  usage: battery
  host: 10.184.63.123
  port: 502
  capacity: 9
  name: battery1

chargers:
- type: template
  template: keba
  host: *****
  rfid: *******
  name: BMW_Wallbox
[... 482 lines cut ...]
May 03 00:58:11 evcc.***.*** evcc[92308]: [mqtt  ] TRACE 2025/05/03 00:58:11 send evcc/site/homePower: '1070.4'
May 03 00:58:11 evcc.***.*** evcc[92308]: [mqtt  ] TRACE 2025/05/03 00:58:11 send evcc/loadpoints/1/smartCostNextStart: ''
May 03 00:58:11 evcc.***.*** evcc[92308]: [mqtt  ] TRACE 2025/05/03 00:58:11 send evcc/loadpoints/1/sessionEnergy: '0'
May 03 00:58:11 evcc.***.*** evcc[92308]: [mqtt  ] TRACE 2025/05/03 00:58:11 send evcc/loadpoints/1/sessionPricePerKWh: ''
May 03 00:58:11 evcc.***.*** evcc[92308]: [mqtt  ] TRACE 2025/05/03 00:58:11 send evcc/loadpoints/1/sessionSolarPercentage: '0'
May 03 00:58:11 evcc.***.*** evcc[92308]: [keba  ] TRACE 2025/05/03 00:58:11 recv from **.**.**.116:7090 {
May 03 00:58:11 evcc.***.*** evcc[92308]: "ID": "2",
May 03 00:58:11 evcc.***.*** evcc[92308]: "State": 1,
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4

realist625, 2025-05-03:
[image]

naltatis, 2025-05-03:
Can you please run `evcc tariff grid` and post the result?

realist625, 2025-05-03:
Here is the output from '
evcc tariff grid'

[main  ] INFO 2025/05/03 20:51:51 evcc 0.203.4
[main  ] INFO 2025/05/03 20:51:51 using config file: /etc/evcc.yaml
[db    ] INFO 2025/05/03 20:51:51 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/05/03 20:51:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/05/03 20:51:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/05/03 20:51:51 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/05/03 20:51:51 SELECT * FROM `settings` 64 <nil>
[mqtt  ] INFO 2025/05/03 20:51:51 connecting evcc-1876420482 at tcp://10.184.63.51:1883
[mqtt  ] DEBUG 2025/05/03 20:51:51 tcp://10.184.63.51:1883 connected
[db    ] TRACE 2025/05/03 20:51:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/05/03 20:51:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/05/03 20:51:51 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[octopus] TRACE 2025/05/03 20:51:51 POST https://api.octopus.energy/v1/graphql/
[octopus] TRACE 2025/05/03 20:51:52 {"query":"mutation ($apiKey:String!){obtainKrakenToken(input: {APIKey: $apiKey}){token}}","variables":{"apiKey":"sk_live_2erxjqqotVzZxMUoApV4RGDA"}}
--
{"data":{"obtainKrakenToken":{"token":"eyJhbGciOiJSUzI1NiIsImlzcyI6Imh0dHBzOi8vYXBpLm9jdG9wdXMuZW5lcmd5L3YxL2dyYXBocWwvIiwiamt1IjoiaHR0cHM6Ly9hdXRoLm9jdG9wdXMuZW5lcmd5Ly53ZWxsLWtub3duL2p3a3MuanNvbiIsImtpZCI6InFRb3hIbzhiTF91Wi1NdE9kQ2labDZacjFaMWsyM1l2Y0taZnU5QldHNkEiLCJ0eXAiOiJKV1QifQ.eyJzdWIiOiJrcmFrZW58YWNjb3VudC11c2VyOjI5MTgzNDkiLCJndHkiOiJBUEktS0VZIiwiZW1haWwiOiJjcmF1YmFsQGdtYWlsLmNvbSIsInRva2VuVXNlIjoiYWNjZXNzIiwiaXNzIjoiaHR0cHM6Ly9hcGkub2N0b3B1cy5lbmVyZ3kvdjEvZ3JhcGhxbC8iLCJpYXQiOjE3NDYzMDE5MTIsImV4cCI6MTc0NjMwNTUxMiwib3JpZ0lhdCI6MTc0NjMwMTkxMn0.hhUzhoi6QYR-0ekVEXJfhIbrUWb0JMLE_PKFaqUvKKqpq1ilOBqW7PIQMAXDkFaCFqyYdAZ7jhRyRjsWCHRFtd5h8DImgKxZ_cpXBEPO-E0m9mr42VQnT_kMBf01xDPzuTCG0Ui9Pyzz8v0r4Y46uFfcMOm7mG3XpB-VzYUI8RXdoPBwI37ia_SxItLedU89KVXfIBDlAKXoCc2_sG-jEr1bfDUU29vh4ASsgBIS9NKyrHOzYJpcjW7NKT58kXnhH7k0ZJXt1mTZHHpY7ddLievXBiYSFWaFk4ZP2geNzZf7d0P83AQJ5rRYYrxo8o1gAqORPxPJW8CqNaAV1XM8Fw"}}}
[octopus] TRACE 2025/05/03 20:51:52 POST https://api.octopus.energy/v1/graphql/
[octopus] TRACE 2025/05/03 20:51:52 {"query":"{viewer{accounts{number}}}"}
--
{"data":{"viewer":{"accounts":[{"number":"A-9D6E4220"}]}}}
[octopus] TRACE 2025/05/03 20:51:52 POST https://api.octopus.energy/v1/graphql/
[octopus] TRACE 2025/05/03 20:51:52 {"query":"query ($accountNumber:String!){account(accountNumber: $accountNumber){electricityAgreements(active: true){id,tariff{... on StandardTariff{id,displayName,fullName,productCode,standingCharge,preVatStandingCharge,tariffCode},... on DayNightTariff{id,displayName,fullName,productCode,standingCharge,preVatStandingCharge,tariffCode},... on ThreeRateTariff{id,displayName,fullName,productCode,standingCharge,preVatStandingCharge,tariffCode},... on HalfHourlyTariff{id,displayName,fullName,productCode,standingCharge,preVatStandingCharge,tariffCode},... on PrepayTariff{id,displayName,fullName,productCode,standingCharge,preVatStandingCharge,tariffCode}},meterPoint{mpan}}}}","variables":{"accountNumber":"A-9D6E4220"}}
--
{"data":{"account":{"electricityAgreements":[{"id":20188822,"tariff":{"id":"175897","displayName":"Octopus Go","fullName":"Octopus Go","productCode":"GO-VAR-22-10-14","standingCharge":47.60616,"preVatStandingCharge":45.3392,"tariffCode":"E-1R-GO-VAR-22-10-14-A"},"meterPoint":{"mpan":"1012766291687"}}]}}}
[octopus] TRACE 2025/05/03 20:51:52 GET https://api.octopus.energy/v1/products/GO-VAR-22-10-14/electricity-tariffs/E-1R-GO-VAR-22-10-14-A/standard-unit-rates/
[octopus] TRACE 2025/05/03 20:51:52 {"count":1881,"next":"https://api.octopus.energy/v1/products/GO-VAR-22-10-14/electricity-tariffs/E-1R-GO-VAR-22-10-14-A/standard-unit-rates/?page=2","previous":null,"results":[{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-05-04T04:30:00Z","valid_to":"2025-05-04T23:30:00Z","payment_method":null},{"value_exc_vat":8.095,"value_inc_vat":8.49975,"valid_from":"2025-05-03T23:30:00Z","valid_to":"2025-05-04T04:30:00Z","payment_method":null},{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-05-03T04:30:00Z","valid_to":"2025-05-03T23:30:00Z","payment_method":null},{"value_exc_vat":8.095,"value_inc_vat":8.49975,"valid_from":"2025-05-02T23:30:00Z","valid_to":"2025-05-03T04:30:00Z","payment_method":null},{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-05-02T04:30:00Z","valid_to":"2025-05-02T23:30:00Z","payment_method":null},{"value_exc_vat":8.095,"value_inc_vat":8.49975,"valid_from":"2025-05-01T23:30:00Z","valid_to":"2025-05-02T04:30:00Z","payment_method":null},{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-05-01T04:30:00Z","valid_to":"2025-05-01T23:30:00Z","payment_method":null},{"value_exc_vat":8.095,"value_inc_vat":8.49975,"valid_from":"2025-04-30T23:30:00Z","valid_to":"2025-05-01T04:30:00Z","payment_method":null},{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-04-30T04:30:00Z","valid_to":"2025-04-30T23:30:00Z","payment_method":null},{"value_exc_vat":8.095,"value_inc_vat":8.49975,"valid_from":"2025-04-29T23:30:00Z","valid_to":"2025-04-3 [... cut ...] lid_from":"2025-04-07T23:30:00Z","valid_to":"2025-04-08T04:30:00Z","payment_method":null},{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-04-07T04:30:00Z","valid_to":"2025-04-07T23:30:00Z","payment_method":null},{"value_exc_vat":8.095,"value_inc_vat":8.49975,"valid_from":"2025-04-06T23:30:00Z","valid_to":"2025-04-07T04:30:00Z","payment_method":null},{"value_exc_vat":27.5796,"value_inc_vat":28.95858,"valid_from":"2025-04-06T04:30:00Z","valid
From    To    Price/Cost (GBP/kWh)

andig, 2025-05-04:
Pls format your post

realist625, 2025-05-04:
This is how it was produced by the output of EVCC command, I did nothing to the output and if I may ask if you want for my outfit, how do you want it formatted.  

Would you see us the alphabet of the query? These are all the tariff data for octopus energy is the output of the query

realist625, 2025-05-05:
Thanks for the pointers on formatting - just want to highlight that I am and end-user - not a developer.   I think it would be great if we could should some respect and understanding to us - we are genuinely trying to to help make a better thing here.   Suggesting that we should do better in how we paste the log information, well, I think it is something that we should be mindful of about.  

To end this - I am super grateful for making such a great thing as ‘evcc’, but this and my other issue are both a source of frustration.

nagisa, 2025-05-05:
@realist625 I suggest that you look into getting your octopus/kraken API token revoked. It ended up getting leaked in your paste. This is something for `evcc` developers to address as well – make sure that the security sensitive information are redacted in the logs. Although @naltatis attempted to modify your paste to redact it a bit, the token is still visible in the edit history (not to mention that it could've already been copied/archived even if the relevant comment is removed.)

Create a new token and update your configuration after you end up revoking your current token(s). If you're not sure how to go about it, reach out to octopus/kraken, they should be able to help.

andig, 2025-05-07:
>  I think it would be great if we could should some respect and understanding to us

@realist625 OT: I don't think this is a matter of (missing) respect. I'm absolutely greatful for our active user base and your and our users contributions are highly invaluable. However, it's also clear that a super small team of maintainers cannot format or reverse engineer thousands of issues. Thank you for your understanding.

Log shows that data is not parsed. /cc @duckfullstop

duckfullstop, 2025-05-07:
Issue confirmed - it's due to an assumption I made that all tariff slots report whether they are Direct Debit or not. PR incoming...

PR #21114 (2025-05-08): tariff/octopus: Fix tariff payment method API presumption
I presumed that Octopus would send a "DIRECT_DEBIT" or "NON_DIRECT_DEBIT" payment_method on responses. Turns out this is not always the case.

This PR treats any `null` entries as valid regardless of the set filter.

If another tariff comes along that starts mixing `null` and a valid payment_method, then this will need a rethink.

Closes #21011
closed 2025-05-08 by andig