evcc-io/evcc #21014: Template "sma-hybrid" ignores third string of SMA "Sunny Boy Smart Energy 5.0"

paTr0ss, 2025-05-03:
### Describe the bug

PV power shown in evcc is lower compared to value shown at the dashboard from SMA.
Reason seems to be that `templates/definition/meter/sma-hybrid.yaml` misses the third string offered by the inverter.

Issue could be solved by also adding the value of modbus register 30967 (DC power input 3 (W)):
```
diff --git a/templates/definition/meter/sma-hybrid.yaml b/templates/definition/meter/sma-hybrid.yaml
index 3443c112..61ac57b4 100644
--- a/templates/definition/meter/sma-hybrid.yaml
+++ b/templates/definition/meter/sma-hybrid.yaml
@@ -88,6 +88,12 @@ render: |
           address: 30961 # SMA Modbus Profile: DcMs.Watt [2]
           type: holding
           decode: int32nan
+      - source: modbus
+        {{- include "modbus" . | indent 6 }}
+        register:
+          address: 30967 # SMA Modbus Profile: DcMs.Watt [3]
+          type: holding
+          decode: int32nan
   energy:
     source: modbus
     {{- include "modbus" . | indent 2 }}
``` 
Please note that the manual also lists a 4th register: 30973 DC power input 4 (W)
I could test above change successfully, but only with my SBSE5.0 inverter.

### Steps to reproduce

1. run evcc in one browser window
2. run SMA dashboard in another bowser window
3. compare PV power

### Configuration details

```yaml
...
meters:
- name: grid_sma
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.109

- name: pv_dach
  type: template
  template: sma-hybrid
  usage: pv
  modbus: tcpip
  id: 3
  host: 192.168.178.108
  port: 502
...
- name: battery_sma
  type: template
  template: sma-hybrid
  usage: battery
  modbus: tcpip
  id: 3
  host: 192.168.178.108
  port: 502
...
```

### Log details

```text
[pv_dach] TRACE 2025/04/27 10:52:19 modbus: send 00 01 00 00 00 06 03 03 78 35 00 02
[pv_dach] TRACE 2025/04/27 10:52:19 modbus: recv 00 01 00 00 00 07 03 03 04 00 00 01 20
[pv_dach] TRACE 2025/04/27 10:52:19 modbus: send 00 02 00 00 00 06 03 03 78 f1 00 02
[pv_dach] TRACE 2025/04/27 10:52:19 modbus: recv 00 02 00 00 00 07 03 03 04 00 00 04 f1
[pv_dach] TRACE 2025/04/27 10:52:19 modbus: send 00 03 00 00 00 06 03 03 77 31 00 04
[pv_dach] TRACE 2025/04/27 10:52:19 modbus: recv 00 03 00 00 00 0b 03 03 08 00 00 00 00 00 1c fa af
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4 (2eb6d31)

andig, 2025-05-03:
Gerne PR

premultiply, 2025-05-04:
SBSE is not supported yet by `sma-hybrid` template.
PR welcome.

closed 2025-05-17 by github-actions[bot]