evcc-io/evcc #21015: Solar production forecast for remainder of UTC dates rather than local time zone.

greatgranddaddy, 2025-05-03:
### Describe the bug

Example:
I am in Australia UTC+10. I see solar production prediction of 3kwh predicted after sunset.
Between sunrise and 10 am local time the estimate will reduce from 3Kwh to 0
After 10 am local time i receive a prediction to 10am the following day.

I suspect the change below to solcast.go may fix it.
FROM: mergeRatesAfter(t.data, data, beginningOfDay())
TO:      mergeRatesAfter(t.data, data, beginningOfDay().Local())

### Steps to reproduce

1. Set local timezone to Melbourne Australia UTC+10
2. configure tariff/solar in yaml 
3. from the GUI select Forecast: remaining solar production today 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: 192.168.20.210 # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 

interval: 30s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.20.240  
  password: 
  refreshToken: 
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.20.240  
  password:   
  refreshToken:   
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.20.240  
  password:   
  refreshToken:   
  name: battery3

chargers:
- type: template
  template: ocpp-autel 
  name: wallbox5

vehicles:
[... 80 lines cut ...]

```text
[solar-template-2] TRACE 2025/05/03 08:03:21 GET https://api.forecast.solar/estimate/-37.99538/145.09006/24/97.5/4.1?time=utc&full=1&resolution=60
[solar-template-0] TRACE 2025/05/03 08:03:21 GET https://api.forecast.solar/estimate/-37.99538/145.09006/24/-172.5/2.87?time=utc&full=1&resolution=60
[solar-template-1] TRACE 2025/05/03 08:03:21 GET https://api.forecast.solar/estimate/-37.99538/145.09006/24/-82.5/4.1?time=utc&full=1&resolution=60
[solar-template-2] TRACE 2025/05/03 08:03:23 {"result":{"watts":{"2025-05-02T14:00:00+00:00":0,"2025-05-02T15:00:00+00:00":0,"2025-05-02T16:00:00+00:00":0,"2025-05-02T17:00:00+00:00":0,"2025-05-02T18:00:00+00:00":0,"2025-05-02T19:00:00+00:00":0,"2025-05-02T20:00:00+00:00":0,"2025-05-02T21:00:00+00:00":32,"2025-05-02T22:00:00+00:00":199,"2025-05-02T23:00:00+00:00":550,"2025-05-03T00:00:00+00:00":1078,"2025-05-03T01:00:00+00:00":1521,"2025-05-03T02:00:00+00:00":1776,"2025-05-03T03:00:00+00:00":2026,"2025-05-03T04:00:00+00:00":1765,"2025-05-03T05:00:00+00:00":964,"2025-05-03T06:00:00+00:00":463,"2025-05-03T07:00:00+00:00":109,"2025-05-03T08:00:00+00:00":0,"2025-05-03T09:00:00+00:00":0,"2025-05-03T10:00:00+00:00":0,"2025-05-03T11:00:00+00:00":0,"2025-05-03T12:00:00+00:00":0,"2025-05-03T13:00:00+00:00":0,"2025-05-03T14:00:00+00:00":0,"2025-05-03T15:00:00+00:00":0,"2025-05-03T16:00:00+00:00":0,"2025-05-03T17:00:00+00:00":0,"2025-05-03T18:00:00+00:00":0,"2025-05-03T19:00:00+00:00":0,"2025-05-03T20:00:00+00:00":0,"2025-05-03T21:00:00+00:00":28,"2025-05-03T22:00:00+00:00":127,"2025-05-03T23:00:00+00:00":218,"2025-05-04T00:00:00+00:00":297,"2025-05-04T01:00:00+00:00":368,"2025-05-04T02:00:00+00:00":411,"2025-05-04T03:00:00+00:00":420,"2025-05-04T04:00:00+00:00":390,"2025-05-04T05:00:00+00:00":322,"2025-05-04T06:00:00+00:00":239,"2025-05-04T07:00:00+00:00":104,"2025-05-04T08:00:00+00:00":0,"2025-05-04T09:00:00+00:00":0,"2025-05-04T10:00:00+00:00":0,"2025-05-04T11:00:00+00:00":0,"2025-05-04T12:00:00+00:00":0,"2025-05-04T13:00:00+00:00":0,"2025-05-04T14:00:00+00:00":0},"watt_hours_period":{"2025-05-02T14:00:00+00:00":0,"2025-05-02T15:00:00+00:00":0,"2025-05-02T16:00:00+00:00":0,"2025-05-02T17:00:00+00:00":0,"2025-05-02T18:00:00+00:00":0,"2025-05-02T19:00:00+00:00":0,"2025-05-02T20:00:00+00:00":0,"2025-05-02T21:00:00+00:00":16,"2025-05-02T22:00:00+00:00":116,"2025-05-02T23:00:00+00:00":375,"2025-05-03T00:00:00+00:00":814,"2025-05-03T01:00:00+00:00":1300,"2025-05-03T02:00:00+00:00":1649,"2025-05-03T03:00:00+00:00":1901,"2025-05-03T04:00:00+00:00":1896,"2025-05-03T05:00:00+00:00":1365,"2025-05-03T06:00:00+00:00":714,"2025-05-03T07:00:00+00:00":286,"2025-05-03T08:00:00+00:00":55,"2025-05-03T09:00:00+00:00":0,"2025-05-03T10:00:00+00:00":0,"2025-05-03T11:00:00+00:00":0,"2025-05-03T12:00:00+00:00":0,"2025-05-03T13:00:00+00:00":0,"2025-05-03T14:00:00+00:00":0,"2025-05-03T15:00:00+00:00":0,"2025-05-03T16:00:00+00:00":0,"2025-05-03T17:00:00+00:00":0,"2025-05-03T18:00:00+00:00":0,"2025-05-03T19:00:00+00:00":0,"2025-05-03T20:00:00+00:00":0,"2025-05-03T21:00:00+00:00":14,"2025-05-03T22:00:00+00:00":78,"2025-05-03T23:00:00+00:00":173,"2025-05-04T00:00:00+00:00":258,"2025-05-04T01:00:00+00:00":333,"2025-05-04T02:00:00+00:00":390,"2025-05-04T03:00:00+00:00":416,"2025-05-04T04:00:00+00:00":405,"2025-05-04T05:00:00+00:00":356,"2025-05-04T06:00:00+00:00":281,"2025-05-04T07:00:00+00:00":172,"2025-05-04T08:00:00+00:00":52,"2025-05-04T09:00:00+00:00":0,"2025-05-04T10:00:00+00:00":0,"2025-05-04T11:00:00+00:00":0,"2025-05-04T12:00:00+00:00":0,"2025-05-04T13:00:00+00:00":0,"2025-05-04T14:00:00+00:00":0},"watt_hours":{"2025-05-02T14:00:00+00:00":0,"2025-05-02T15:00:00+00:00":0,"2025-05-02T16:00:00+00:00":0,"2025-05-02T17:00:00+00:00":0,"2025-05-02T18:00:00+00:00":0,"2025-05-02T19:00:00+00:00":0,"2025-05-02T20:00:00+00:00":0,"2025-05-02T21:00:00+00:00":16,"2025-05-02T22:00:00+00:00":132,"2025-05-02T23:00:00+00:00":507,"2025-05-03T00:00:00+00:00":1321,"2025-05-03T01:00:00+00:00":2621,"2025-05-03T02:00:00+00:00":4270,"2025-05-03T03:00:00+00:00":6171,"2025-05-03T04:00:00+00:00":8067,"2025-05-03T05:00:00+00:00":9432,"2025-05-03T06:00:00+00:00":10146,"2025-05-03T07:00:00+00:00":10432,"2025-05-03T08:00:00+00:00":10487,"2025-05-03T09:00:00+00:00":10487,"2025-05-03T10:00:00+00:00":10487,"2025-05-03T11:00:00+00:00":10487,"2025-05-03T12:00:00+00:00":10487,"2025-05-03T13:00:00+00:00":10487,"2025-05-03T14:00:00+00:00":0,"2025-05-03T15:00:00+00:00":0,"2025-05-03T16:00:00+00:00":0,"2025-05-03T17:00:00+00:00":0,"2025-05-03T18:00:00+00:00":0,"2025-05-03T19:00:00+00:00":0,"2025-05-03T20:00:00+00:00":0,"2025-05-03T21:00:00+00:00":14,"2025-05-03T22:00:00+00:00":92,"2025-05-03T23:00:00+00:00":265,"2025-05-04T00:00:00+00:00":523,"2025-05-04T01:00:00+00:00":856,"2025-05-04T02:00:00+00:00":1246,"2025-05-04T03:00:00+00:00":1662,"2025-05-04T04:00:00+00:00":2067,"2025-05-04T05:00:00+00:00":2423,"2025-05-04T06:00:00+00:00":2704,"2025-05-04T07:00:00+00:00":2876,"2025-05-04T08:00:00+00:00":2928,"2025-05-04T09:00:00+00:00":2928,"2025-05-04T10:00:00+00:00":2928,"2025-05-04T11:00:00+00:00":2928,"2025-05-04T12:00:00+00:00":2928,"2025-05-04T13:00:00+00:00":2928,"2025-05-04T14:00:00+00:00":0},"watt_hours_day":{"2025-05-03":10487,"2025-05-04":2928}},"message":{"code":0,"type":"success","text":"","pid":"Zp36574R","info":{"latitude":-37.9954,"longitude":145.0901,"distance":0,"place":"16 McKay Street, Parkdale VIC 3195, Australia","timezone":"Australia/Melbourne","time":"2025-05-03T18:03:22+10:00","time_utc":"2025-05-03T08:03:22+00:00"},"ratelimit":{"zone":"IP 2406:3400:318:82a0::503","period":3600,"limit":12,"remaining":8}}}
[solar-template-1] TRACE 2025/05/03 08:03:23 {"result":{"watts":{"2025-05-02T14:00:00+00:00":0,"2025-05-02T15:00:00+00:00":0,"2025-05-02T16:00:00+00:00":0,"2025-05-02T17:00:00+00:00":0,"2025-05-02T18:00:00+00:00":0,"2025-05-02T19:00:00+00:00":0,"2025-05-02T20:00:00+00:00":0,"2025-05-02T21:00:00+00:00":298,"2025-05-02T22:00:00+00:00":1076,"2025-05-02T23:00:00+00:00":1662,"2025-05-03T00:00:00+00:00":1978,"2025-05-03T01:00:00+00:00":1946,"2025-05-03T02:00:00+00:00":1645,"2025-05-03T03:00:00+00:00":1377,"2025-05-03T04:00:00+00:00":888,"2025-05-03T05:00:00+00:00":406,"2025-05-03T06:00:00+00:00":239,"2025-05-03T07:00:00+00:00":73,"2025-05-03T08:00:00+00:00":0,"2025-05-03T09:00:00+00:00":0,"2025-05-03T10:00:00+00:00":0,"2025-05-03T11:00:00+00:00":0,"2025-05-03T12:00:00+00:00":0,"2025-05-03T13:00:00+00:00":0,"2025-05-03T14:00:00+00:00":0,"2025-05-03T15:00:00+00:00":0,"2025-05-03T16:00:00+00:00":0,"2025-05-03T17:00:00+00:00":0,"2025-05-03T18:00:00+00:00":0,"2025-05-03T19:00:00+00:00":0,"2025-05-03T20:00:00+00:00":0,"2025-05-03T21:00:00+00:00":40,"2025-05-03T22:00:00+00:00":177,"2025-05-03T23:00:00+00:00":292,"2025-05-04T00:00:00+00:00":364,"2025-05-04T01:00:00+00:00":400,"2025-05-04T02:00:00+00:00":401,"2025-05-04T03:00:00+00:00":368,"2025-05-04T04:00:00+00:00":310,"2025-05-04T05:00:00+00:00":237," [... cut ...] 5-03T08:03:23+00:00"},"ratelimit":{"zone":"IP 2406:3400:318:82a0::503","period":3600,"limit":12,"remaining":9}}}
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4

andig, 2025-05-03:
Please check

    evcc tariff

Is the output correct?

greatgranddaddy, 2025-05-04:
The current value is correct in the sense that the estimate matches the input data. But perhaps not working as intended, in that it estimates to GTM midnight rather than midnight local time.

Apologies, I am new to ago and I don’t have my environment up and running yet to test this change.

andig, 2025-05-04:
> The current value is correct in the sense that the estimate matches the input data

I don't undwerstand. Please https://github.com/evcc-io/evcc/issues/21015#issuecomment-2848569295

andig, 2025-05-06:
@greatgranddaddy could you please show the output of

    evcc tariff

We need to check if the tariff is correct in local timezone or if the mistake is already there.

closed 2025-05-17 by andig