evcc-io/evcc #21026: Solax Modbus tcpip -> read failed: transaction ID doesn't match

Chroni89, 2025-05-03:
### Describe the bug

Sporadisch tritt folgender Fehler auf: 
[site  ] ERROR 2025/05/03 11:09:45 battery 1 soc: read failed: modbus: response transaction id '1' does not match request '167'

Fehler tritt für alle per Modbus abgerufenen Werte auf (Batterie, PV).
Nach Neustart von evcc oder Zeitablauf (>1h) verschwindet der Fehler temporär wieder. 

Einzige bisher funktionierende Lösung: Downgrade auf evcc 0.202.1. Bis zu dieser Version haben die Abfragen per Modbus und der Modbus Proxy einwandfrei über einen Zeitraum von mehreren Wochen funktioniert. 

### Steps to reproduce

1.Upgraden auf die neueste Version von evcc 
2.Starten von evcc
3.Einige Minuten / Stunden warten -> nicht exakt reproduzierbar, wann sich die Kommunikation "verschluckt" 
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  proxy-5021: trace

# unique installation id
plant: 29ca75078a5268b63ffa2400a664ba3d4758b1f3fba75c72497b225d9bb05659

interval: 10s # control cycle interval

# Sponsor token 
sponsortoken: got_one

modbusproxy:
- port: 5021
  uri: 192.168.178.200:502

meters:
- type: template
  template: solax
  id: 1
  host: 192.168.178.200
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solax
  id: 1
  host: 192.168.178.200
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solax
  id: 1
  host: 192.168.178.200
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: hoymiles-opendtu
  usage: pv
[... 127 lines cut ...]
May 02 00:01:01 EVCC evcc[199092]: [site  ] ERROR 2025/05/02 00:01:01
May 02 00:01:01 EVCC evcc[199092]: 2025-05-02 00:00:46.494003484 +0000 UTC m=+51798.336963875 !! 15330ms add[0]: read failed: modbus: response transaction id '>
May 02 00:01:01 EVCC evcc[199092]: [site  ] ERROR 2025/05/02 00:01:01 pv 1 power: add[0]: read failed: modbus: response transaction id '4115' does not match re>
May 02 00:01:01 EVCC evcc[199092]: [site  ] DEBUG 2025/05/02 00:01:01 pv power: 0W
May 02 00:01:06 EVCC evcc[199092]: [site  ] ERROR 2025/05/02 00:01:06 battery 1 soc: read failed: modbus: response transaction id '4116' does not match request>
May 02 00:01:06 EVCC evcc[199092]: [site  ] ERROR 2025/05/02 00:01:06 grid power: read failed: modbus: response transaction id '4113' does not match request '1>
May 02 00:01:06 EVCC evcc[199092]: [site  ] DEBUG 2025/05/02 00:01:06 ----
May 02 00:01:06 EVCC evcc[199092]: [lp-1  ] DEBUG 2025/05/02 00:01:06 charge power: 0W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4

andig, 2025-05-03:
Dein Log ist leider nicht lesbar da die Zeilen abgeschnitten sind

Chroni89, 2025-05-03:
Danke für den Hinweis! Ich habe den Eintrag bearbeitet

premultiply, 2025-05-04:
Ist das eine Direktverbindung oder ist da noch irgendeine aktive Komponente dazwischen (z. B. Modbus-TCP/RTU-Konverter etc.)?

Chroni89, 2025-05-05:
Die Verbindung geht von Wechselrichter per Modbus an einen Waveshare Converter und von da per TCP/IP an EVCC. 
Das ist die IP  host: 192.168.178.200 in der evcc.yaml
(vgl: https://www.photovoltaikforum.com/wissen/entry/86-datenaustausch-mit-home-assistant-über-rs-485-com)
In Evcc habe ich dann den Modbusproxy aktiv, der die Daten für den Homeassistant bereitstellt.

premultiply, 2025-05-05:
Klingt eher nach lokalem Fehler / fehlerhafter Konveterkonfig / paralleler Kommunikation.

andig, 2025-05-08:
Das sieht aus, als würde der Solax Netzwerkverbindungen verwürfeln und auf die falsche Verbindung antworten. Der alte Code hat in dem Fall lange und geduldig gewartet und versucht falsche IDs zu ignorieren. Der neue Code schließt die Verbindung und versucht es erneut. Auch mit neuer Verbindung scheint hier aber Murks raus zu kommen. Leider können wir nicht auf den alten Code zurück da dieser gar nicht damit zu recht kam wenn der Server die Verbindung geschlossen hat. Falls sich das Verhalten hier irgendwie auf Knopfdruck reproduzieren ließe könnte eine Debugsession helfen. Ohne hab ich leider keine gute Idee...

andig, 2025-05-11:
> Falls sich das Verhalten hier irgendwie auf Knopfdruck reproduzieren ließe könnte eine Debugsession helfen.

alexjet, 2025-05-14:
I have very similar issue. I have RS485-USB adapter connected to my Growatt inverter, EVCC's Modbus proxy requests from USB, then Modbus plugin in HA connects to Modbus proxy. 
In HA I have 7 sensors configured that read via Modbus proxy directly from needed registers. It looks like the values of some sensors get messed up, so sometimes I get the value from another sensor, sometimes just an error returned (perhaps because of unexpected data type). I suppose that Modbus proxy's requests to inverter overlap/conflict sometimes, or perhaps conflict with EVCC's own requests to USB.

Pitchen84, 2025-05-14:
Bei mir tauchen auch die ID Fehlermeldungen auf. Ich betreibe einen Sofar HYD15 KTL Wechselrichter an einem Waveshare ETH Modbus Adapter. Zusätzlich habe ich die Solax Integration in Home Assistant am laufen. Diese habe ich für Testzwecke nun deaktiviert. Wenn EVCC alleine auf den Waveshare zugreift, scheint alles in Ordnung zu sein. Dies bestätigt meiner Ansicht nach die Annahme von @andig, dass die Solax Integration hier irgendwie dazwischen Funkt.

premultiply, 2025-05-16:
Modbus ist grundsätzlich nicht für Multi-Master-Betrieb vorgesehen. Bei nativem Modbus TCP kann dies möglich sein. Hier ist es aber durchaus möglich, dass im Gerät nur ein einfacher Protokollumsetzer steckt und intern doch mit Modbus RTU gearbeitet wird.

Grundsätzlich gilt erstmal: Es kann nur einen Master geben.
Im Zweifelsfall immer einen Proxy verwenden der alle Master-Anfragen sauber serialisiert.

Mehrere Möglichkeiten:

- Parallele Zugriffe (mehrere Master) direkt auf das Gerät (ggf. am Proxy vorbei)
- Fehlerhafte Proxy-Konfiguration
- Modbus-Proxy in evcc kaputt

Pitchen84, 2025-05-19:
Ich habe das ganze nun nochmal getestet: ohne Home Assistant ist die Modbus Verbindung stabil und läuft ohne Probleme. 
Habe das ganze nun mit dem EVCC Proxy am laufen.
Ich hoffe ich habe ihn richtig konfiguriert: 

Port: Hier habe ich den Port eingetragen, auf dem EVCC die Modbus Daten weiter geben soll
URI: IP Adresse des Waveshare ETH inkl. Port des Waveshare (ein anderer als der von EVCC)

Die Solax Integration habe ich anschließend mit der IP Adresse der EVCC Instanz und dem oben eingetragenen Port als Modbus Ziel konfiguriert.

Nun habe ich leider wieder die ID Fehler ca. alle 30 Minuten.
Habe ich etwas falsch gemacht?

andig, 2025-05-19:
> https://github.com/evcc-io/evcc/issues/21026#issuecomment-2870062956

If there's anything we could reproduce I could take a look.

Chroni89, 2025-05-22:
Also ich habe jetzt einige Dinge ausprobiert und irgendeine dieser Maßnahmen (oder die Kombination aller) hat dazu geführt, dass der Fehler bei mir nicht mehr auftritt: 

- Update Waveshare Modbus Adapter auf Firmware V1. 523
- Update Solax Inverter Modbus (HomeAssistant Integration) auf Version 2025.04.4 
- Timeout in der Solax Inverter Modbus Konfiguration auf 10 Sekunden erhöht 
- Update EVCC auf Version 0.203.6

Der Fehler ist damit verschwunden und nicht mehr reproduzierbar. 
Vielen Dank für eure Unterstützung!

closed 2025-05-22 by Chroni89

andig, 2025-05-22:
Danke für die Doku der Massnahmen!