evcc-io/evcc #21039: Grid power outdated / Tibber rate limit notification

erdbeertoni, 2025-05-04:
### Describe the bug

Hi all,
since the latest 2 or 3 evcc updates I'm having the issue with error Grid power outdated and also received mails from tibber because of API rate limit ...

The whole thing has been annoying for 2 weeks now, does anyone have a tip?

### Steps to reproduce

Tiiber Mail
We have detected a high number of connection initialization requests to the Tibber Websocket API (wss://websocket-api.tibber.com). You have likely set up a connection with a personal access token from Tibber developer console at developer.tibber.com or via a third-party OAuth2 service.

Your client is currently using the user agent evcc/v0.0.0 hasura/go-graphql-client/v0.13.2 and exceeding the allowed rate of 20 new connections per hour.
It appears you are running evcc, that has known issue. Please make sure you are running the version 0.203.4 or later.

This is a notification only — your connections have not been blocked yet. In the future we will start blocking your connections.

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.10.78
  port: 7070

interval: 90s
sponsortoken: *

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.10.108 # Hostname
    port: 502 # Port
    # timeout: 10s # optional
    
  - name: my_grid
    type: template
    template: tibber-pulse
    usage: grid
    token: *

chargers:
  - name: my_charger
    type: template
    template: go-e-v3
    host: 192.168.10.95 # IP-Adresse oder Hostname 

vehicles:
  - name: my_car
    type: template
    template: audi
    title: Audi eTron # Wird in der Benutzeroberfläche angezeigt # Optional
    user: * # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: * # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: * # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 68 # Akkukapazität in kWh # Optional 
    #minSoc: 30 

site:
  title: Home
  meters:
    pv:
    - my_pv
    grid: my_grid

loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    meter: 
    mode: pv # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: my_car # Referenz auf Standardfahrzeug
    # vehicle: car1 # set default vehicle (disables vehicle detection)
    #resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      poll:
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    #type: fixed
    #price: 0.37 # [currency]/kWh
    type: tibber
    token: *

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.11 # [currency]/kWh

messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
```

### Log details

```text
[site  ] DEBUG 2025/05/04 11:48:12 ----
[lp-1  ] DEBUG 2025/05/04 11:48:12 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 11:48:12 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 11:48:12 pv 1 power: 2150W
[site  ] ERROR 2025/05/04 11:49:12 grid power: outdated
[site  ] DEBUG 2025/05/04 11:49:42 ----
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4

andig, 2025-05-04:
Let's be clear: you are running 203.4? Please share full log from startup.

andig, 2025-05-04:
@GrimmiMeloni since we have increased the read timeout to 90s we should probably increase the grid meter timeout to 2m here? Not sure if that's really usable, but that seems to be what Tibber wants to send.

closed 2025-05-04 by andig