evcc-io/evcc #21050: Bug in calculation?

Hofyyy, 2025-05-04:
### Describe the bug

Hello,

I got the first time today strange values and have no clue what happened.
Chargepower was 5138W ... but it should be 1000W +-

Maybe someone have an Idea what I could check ...

### Steps to reproduce

?

### Configuration details

```yaml
EVCC Docker 0.203.4 -> Victron System
Two Wallboxes
1. Wallbe Pro
2. TWC3
-> Bug happended with TWC3 connected to Zoe
```

### Log details

```text
[lp-1  ] DEBUG 2025/05/04 17:07:51 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 17:07:51 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:07:51 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:07:51 grid power: 2304W
[site  ] DEBUG 2025/05/04 17:07:51 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:07:51 grid currents: [9.2 7.8 7.3]A
[site  ] DEBUG 2025/05/04 17:07:51 pv 1 power: 613W
[site  ] DEBUG 2025/05/04 17:07:51 site power: 2404W
[lp-1  ] DEBUG 2025/05/04 17:07:51 charge total import: 5604.542kWh
[lp-1  ] DEBUG 2025/05/04 17:07:51 charger status: A
[site  ] DEBUG 2025/05/04 17:08:01 ----
[lp-2  ] DEBUG 2025/05/04 17:08:01 charge power: 5138W
[lp-2  ] DEBUG 2025/05/04 17:08:01 charge currents: [7.4 7.4 7.3]A
[lp-1  ] DEBUG 2025/05/04 17:08:01 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 17:08:01 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:08:01 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:08:01 grid power: 2281W
[site  ] DEBUG 2025/05/04 17:08:01 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:08:01 grid currents: [9.3 7.9 7.3]A
[site  ] DEBUG 2025/05/04 17:08:01 pv 1 power: 621W
[site  ] DEBUG 2025/05/04 17:08:01 site power: 2381W
[lp-2  ] DEBUG 2025/05/04 17:08:01 charge voltages: [233 234 230]V
[lp-2  ] DEBUG 2025/05/04 17:08:01 detected connected phases: 3p
[lp-2  ] DEBUG 2025/05/04 17:08:01 detected active phases: 3p
[lp-2  ] DEBUG 2025/05/04 17:08:01 charger status: C
[lp-2  ] ERROR 2025/05/04 17:08:01 vehicle soc: unexpected status: 429 (Too Many Requests)
[lp-2  ] ERROR 2025/05/04 17:08:01 set charge current limit 16A: vehicle not capable of current control
[site  ] DEBUG 2025/05/04 17:08:11 ----
[lp-1  ] DEBUG 2025/05/04 17:08:11 charge power: 0W
[lp-2  ] DEBUG 2025/05/04 17:08:11 charge power: 5133W
[lp-2  ] DEBUG 2025/05/04 17:08:11 charge currents: [7.4 7.4 7.3]A
[lp-1  ] DEBUG 2025/05/04 17:08:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:08:11 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:08:11 grid power: 2304W
[site  ] DEBUG 2025/05/04 17:08:11 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:08:11 grid currents: [9.2 7.8 7.3]A
[site  ] DEBUG 2025/05/04 17:08:11 pv 1 power: 623W
[site  ] DEBUG 2025/05/04 17:08:11 site power: 2404W
[lp-1  ] DEBUG 2025/05/04 17:08:11 charge total import: 5604.542kWh
[lp-1  ] DEBUG 2025/05/04 17:08:11 charger status: A
[site  ] DEBUG 2025/05/04 17:08:21 ----
[lp-2  ] DEBUG 2025/05/04 17:08:21 charge power: 5164W
[lp-2  ] DEBUG 2025/05/04 17:08:21 charge currents: [7.4 7.4 7.4]A
[lp-1  ] DEBUG 2025/05/04 17:08:21 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 17:08:21 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:08:21 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:08:21 grid power: 2257W
[site  ] DEBUG 2025/05/04 17:08:21 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:08:21 grid currents: [9.3 7.8 7.3]A
[site  ] DEBUG 2025/05/04 17:08:21 pv 1 power: 630W
[site  ] DEBUG 2025/05/04 17:08:21 site power: 2357W
[lp-2  ] DEBUG 2025/05/04 17:08:21 charge voltages: [233 234 230]V
[lp-2  ] DEBUG 2025/05/04 17:08:21 detected connected phases: 3p
[lp-2  ] DEBUG 2025/05/04 17:08:21 detected active phases: 3p
[lp-2  ] DEBUG 2025/05/04 17:08:21 charger status: C
[lp-2  ] ERROR 2025/05/04 17:08:21 vehicle soc: unexpected status: 429 (Too Many Requests)
[lp-2  ] ERROR 2025/05/04 17:08:21 set charge current limit 16A: vehicle not capable of current control
[site  ] DEBUG 2025/05/04 17:08:31 ----
[lp-2  ] DEBUG 2025/05/04 17:08:31 charge power: 5168W
[lp-2  ] DEBUG 2025/05/04 17:08:31 charge currents: [7.4 7.4 7.4]A
[lp-1  ] DEBUG 2025/05/04 17:08:31 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 17:08:31 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:08:31 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:08:31 grid power: 2251W
[site  ] DEBUG 2025/05/04 17:08:31 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:08:31 grid currents: [9.2 7.8 7.3]A
[site  ] DEBUG 2025/05/04 17:08:31 pv 1 power: 650W
[site  ] DEBUG 2025/05/04 17:08:31 site power: 2351W
[lp-1  ] DEBUG 2025/05/04 17:08:31 charge total import: 5604.542kWh
[lp-1  ] DEBUG 2025/05/04 17:08:31 charger status: A
[site  ] DEBUG 2025/05/04 17:08:41 ----
[lp-2  ] DEBUG 2025/05/04 17:08:41 charge power: 5170W
[lp-2  ] DEBUG 2025/05/04 17:08:41 charge currents: [7.4 7.4 7.4]A
[lp-1  ] DEBUG 2025/05/04 17:08:41 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 17:08:41 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:08:41 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:08:41 grid power: 2255W
[site  ] DEBUG 2025/05/04 17:08:41 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:08:41 grid currents: [9.2 7.9 7.3]A
[site  ] DEBUG 2025/05/04 17:08:41 pv 1 power: 657W
[site  ] DEBUG 2025/05/04 17:08:41 site power: 2355W
[lp-2  ] DEBUG 2025/05/04 17:08:41 charge voltages: [233 235 231]V
[lp-2  ] DEBUG 2025/05/04 17:08:41 detected connected phases: 3p
[lp-2  ] DEBUG 2025/05/04 17:08:41 detected active phases: 3p
[lp-2  ] DEBUG 2025/05/04 17:08:41 charger status: C
[lp-2  ] ERROR 2025/05/04 17:08:42 vehicle soc: unexpected status: 429 (Too Many Requests)
[lp-2  ] ERROR 2025/05/04 17:08:42 set charge current limit 16A: vehicle not capable of current control
[site  ] DEBUG 2025/05/04 17:08:51 ----
[lp-2  ] DEBUG 2025/05/04 17:08:51 charge power: 5144W
[lp-2  ] DEBUG 2025/05/04 17:08:51 charge currents: [7.4 7.4 7.3]A
[lp-1  ] DEBUG 2025/05/04 17:08:51 charge power: 0W
[lp-1  ] DEBUG 2025/05/04 17:08:51 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/04 17:08:51 battery 1 power: -0W
[site  ] DEBUG 2025/05/04 17:08:51 grid power: 2238W
[site  ] DEBUG 2025/05/04 17:08:51 battery 1 soc: 69%
[site  ] DEBUG 2025/05/04 17:08:51 grid currents: [9.2 7.8 7.2]A
[site  ] DEBUG 2025/05/04 17:08:51 pv 1 power: 660W
[site  ] DEBUG 2025/05/04 17:08:51 site power: 2338W
[lp-1  ] DEBUG 2025/05/04 17:08:51 charge total import: 5604.542kWh
[lp-1  ] DEBUG 2025/05/04 17:08:51 charger status: A
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.4

Hofyyy, 2025-05-04:
[image]
[image]
[image]

Hofyyy, 2025-05-04:
Also wenn ich raten müsste macht mein EM24 komische Dinge, oder ich bin zu doof zum rechnen.
Die Watt Zahlen Scheinen zu stimmen, aber die "A" nicht oder?
Die Frage ist hab ich den Falsch eingestellt, oder benutzt EVCC neuerdings auch die "A" zum rechnen? Fragen über Fragen.

mfuchs1984, 2025-05-04:
Wenn ich in die Templates schauen, komme ich zum Schluss, dass evcc bei Verwendung des em24 als charge Meter die currents verwendet, nicht power.

Hofyyy, 2025-05-04:
Hmm. Ok wenn ich es recht sehe pasen beim EM24 watt nicht zu A. Aber der fehler waren falsche Watt zahleb von der TWC3. ggf. muss ich die mal durchstarten …

Hofyyy, 2025-05-04:
Also mein E24 weicht systematisch mit der Ampere Zahl ab.
L1 Faktor = 2
L2 Faktor = 4
L3 Faktor = 0,66

Versteht kein Mensch ... ich weiss auch nicht ob das schon immer so war.
Jetzt werd ich mal mi Detail Die TW3 prüfen. Warum da falsche Werte sind. #Plötzlich ....